coolsnowwolf/lede #3842: ipv6无法显示虚拟动态接口

mgsqz, 2020-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：同一虚拟机平台，用openwrt官方19.07.1版本拨号以及ipv6测试如下
[image]

[image]

而用commits 87922d5编译的lede（已选择ipv6helper）拨号以及ipv6测试如下
[image]

[image]

dhcpv6设置都是服务器模式+无状态&有状态模式

### (2) 路由器型号和固件版本

A：x86 OpenWrt R20.3.11使用commits 87922d5编译

### (3) 详细日志

A：

coolsnowwolf, 2020-03-16:
特意的，新功能

ckxt, 2020-03-17:
ipv6 我用电脑拨号没问题，用op拨号不行 哎...恩山上找了很多教程试了都不行  现在直接放弃了

ledewrt, 2020-03-17:
大佬禁用了ipv6解析dns服务，所以你网页打不开ipv6.但下载如迅雷可以用ipv6.

tanst, 2020-03-18:
[image]

closed 2020-07-04 by coolsnowwolf