coolsnowwolf/lede #3844: Netflix分离解锁无效

abclai, 2020-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

Netflix解锁失败！而且原本切换到可以看奈飞的线路可以看奈飞 现在也无法观看！目测是因为Netflix域名分离后，Netflix域名名单没有生效/分流服务器没有工作造成酸酸乳plus无法代理！

### (2) 路由器型号和固件版本

A：OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7)

### (3) 详细日志

A：

abclai, 2020-03-16:
好像由于Netflix域名有专门名单，所以在GFW模式下就算是在可看Netflix线路设置成主服务器，Netflix域名加入强制代理名单也依旧无法代理！只有绕过大陆IP并把可看Netflix线路设置成主服务器才能观看！原本只需在GFW模式下

abclai, 2020-03-16:
Netflix.com现在在GFW模式识别为Netflix/cn 说明代理失效！

yangjiey2k, 2020-03-17:
5分钟前更新了最新的固件，测试了一下没问题，可以正常分流，正常解锁

yangjiey2k, 2020-03-17:
我是绕过大陆IP模式

abclai, 2020-03-17:
> 我是绕过大陆IP模式

看了你的反馈试试了，的确是绕过大陆IP模式下，但是GFW模式是是不行啦

coolsnowwolf, 2020-03-17:
已添加 GFW 模式解锁 Netflix 的支持

closed 2020-03-17 by coolsnowwolf

cnhz76, 2020-07-08:
从4.19升级到
**固件版本 | OpenWrt R20.7.1 / LuCI Master (git-20.187.38662-0cb730a)
内核版本 | 5.4.50**
后，实测neiflix分离解锁功能失效，拜托@coolsnowwolf抽空检查下，感谢！