coolsnowwolf/lede #3845: ssr plus，全中转机场的自动切换无效。

lsl9119, 2020-03-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ssr plus 故障自动切换无效，因为使用的是全中转机场，本地端到节点中转端的延迟都是在30ms以下，但是部分节点从中转端到海外端会有掉线的情况，但无法通过ping节点ip体现出来，所以无法连接海外端后主节点还是能ping通，无法正常实现切换。

### (2) 路由器型号和固件版本

A：软路由，cpu n3710  OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7) 

### (3) 详细日志

A：
2020-03-17 11:26:20 ShadowsocksR No Problem.
2020-03-17 11:25:20 ShadowsocksR No Problem.
2020-03-17 11:24:19 ShadowsocksR No Problem.
2020-03-17 11:23:19 ShadowsocksR No Problem.
2020-03-17 11:22:13 switch to default [no5637860493.internetworld.cloud] proxy!
2020-03-17 11:22:12 Shadowsocks/ShadowsocksR 4 Threads Started!
2020-03-17 11:22:10 Main server is avilable.
2020-03-17 11:22:07 Current server is not default Main server, try to switch back.
2020-03-17 11:21:07 ShadowsocksR server switch OK
2020-03-17 11:21:07 Shadowsocks/ShadowsocksR 4 Threads Started!
2020-03-17 11:21:05 Another server is avilable, now switching server.
2020-03-17 11:21:02 Current server error, try to switch another server.
2020-03-17 11:19:44 ShadowsocksR No Problem.
2020-03-17 11:18:44 ShadowsocksR No Problem.
2020-03-17 11:17:43 ShadowsocksR No Problem.
2020-03-17 11:16:43 ShadowsocksR No Problem.
2020-03-17 11:15:43 ShadowsocksR No Problem.
2020-03-17 11:14:43 ShadowsocksR No Problem.
2020-03-17 11:13:42 ShadowsocksR No Problem.
2020-03-17 11:12:42 ShadowsocksR No Problem.
2020-03-17 11:11:42 ShadowsocksR No Problem.
2020-03-17 11:10:41 ShadowsocksR No Problem.
2020-03-17 11:09:41 switch to default [no5637860493.internetworld.cloud] proxy!
2020-03-17 11:09:41 Shadowsocks/ShadowsocksR 4 Threads Started!
2020-03-17 11:09:38 Main server is avilable.
2020-03-17 11:09:36 Current server is not default Main server, try to switch back.
2020-03-17 11:08:36 ShadowsocksR server switch OK
2020-03-17 11:08:36 Shadowsocks/ShadowsocksR 4 Threads Started!
2020-03-17 11:08:31 Another server is avilable, now switching server.
2020-03-17 11:08:30 Current server error, try to switch another server.
2020-03-17 11:07:12 switch to default [no5637860493.internetworld.cloud] proxy!
2020-03-17 11:07:11 Shadowsocks/ShadowsocksR 4 Threads Started!
2020-03-17 11:07:09 Main server is avilable.
2020-03-17 11:07:06 Current server is not default Main server, try to switch back.
2020-03-17 11:06:06 ShadowsocksR server switch OK
2020-03-17 11:06:06 Shadowsocks/ShadowsocksR 4 Threads Started!
2020-03-17 11:06:03 Another server is avilable, now switching server.
2020-03-17 11:06:01 Current server error, try to switch another server.
2020-03-17 11:04:43 ShadowsocksR No Problem.
2020-03-17 11:03:42 ShadowsocksR No Problem.
2020-03-17 11:02:42 ShadowsocksR No Problem.
2020-03-17 11:01:42 ShadowsocksR No Problem.
2020-03-17 11:00:42 ShadowsocksR No Problem.

coolsnowwolf, 2020-03-17:
不是基于 ping 的，是基于这个节点是否能 wget google.com 决定切换

lsl9119, 2020-03-17:
> 
> 
> 不是基于 ping 的，是基于这个节点是否能 wget google.com 决定切换

那基于目前的日志，节点出问题后没有切换到其他节点的现象我该如何排查问题呢？谢谢指点

ZqinKing, 2020-03-17:
> 不是基于 ping 的，是基于这个节点是否能 wget google.com 决定切换

请问一下在强制名单中添加了google.com是否会影响自动切换功能？

garypang13, 2020-03-17:
全中转机场没办法自动切换的,因为在切换到备用线路后,只能通过ping来检测主线路是否可用,结果肯定是能ping通,然后就会切换回主线路,接着wget google.com发现不通又切换成备用线路,这样无限循环.

coolsnowwolf, 2020-03-23:
这个在之后版本会用个全新的方法解决

winddriver, 2020-03-27:
切换机场能否自动根据节点的响应速度来切换，比如同样是wget google.com都能连通的节点A、B、C，优先切换到响应最快的那个节点。

现在我基本上是靠电脑上装的v2rayN检测节点的速度，然后手动去ssr里切换节点，实在是很麻烦。

closed 2020-07-04 by coolsnowwolf