coolsnowwolf/lede #3846: 关于Netflix分离解锁的建议

abclai, 2020-03-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：首先感谢下LEarn做的Netflix分离解锁啦，十分感谢啦！我用的是使用的是酸酸乳的机场，之前没

有使用Netflix分离解锁之前，只能选择Netflix解锁机场啦，当时使用也发现一个问题就是Netflix好像

一些视频都是走UDP的，也就是说之前我看Netflix需要吧机场做为主服务器还需要为客户端开启游

戏模式UDP中继，否则Netflix视频会说使用解锁工具而无法观看！

可否对现在的Netflix分离解锁模式进行加强，就是默认代理Netflix域名的UDP而

不是像普通的只代理TCP啦！

虽然也可以像我一样，Netflix分离解锁+游戏模式UDP中继同时开启来解决啦！

但是对于小白来说好像不清楚还有这个解决方案！

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

coolsnowwolf, 2020-03-17:
没看到有 Netflix 视频流量需要走 UDP 的，你多测试下

aaaisan, 2020-03-17:
我觉得电视盒子好像走UDP，因为如果不选择游戏增强模式，电视盒子会提示代理，将盒子的IP指定走游戏增强模式，就可以正常看，而用电脑相同节点则直接可看

hcym, 2020-03-17:
盒子里YouTube走tcp，这个我试过。手机上app可能有

nameless3721, 2020-03-17:
netflix.com
netflix.net
nflxext.com
nflximg.com
nflximg.net
nflxso.net
nflxvideo.net
netflixdnstest0.com
netflixdnstest1.com
netflixdnstest2.com
netflixdnstest3.com
netflixdnstest4.com
netflixdnstest5.com
netflixdnstest6.com
netflixdnstest7.com
netflixdnstest8.com
netflixdnstest9.com
23.246.0.0/18
37.77.184.0/21
45.57.0.0/17
64.120.128.0/17
66.197.128.0/17
108.175.32.0/20
192.173.64.0/18
198.38.96.0/19
198.45.48.0/20
8.41.4.0/24
69.53.224.0/19
185.2.220.0/22
185.9.188.0/22
207.45.72.0/22
208.75.76.0/22

aaaisan, 2020-03-17:
> 盒子里YouTube走tcp，这个我试过。手机上app可能有

说的是Netflix……

abclai, 2020-03-17:
> > 盒子里YouTube走tcp，这个我试过。手机上app可能有
> 
> 说的是Netflix……而且不同盒子也不一样吧，我的是小米盒子4国际版

是的，我也是小米盒子S也是一样的问题啦！而且IOS也是会出现这个问题啦！感觉应该是有UDP检测或是走UDP啦！要不不可能开启游戏模式就OK

coolsnowwolf, 2020-03-17:
游戏模式 OK 证明有些 IP 段漏网了，目前已经添加自定义分流 IP 模式

coolsnowwolf, 2020-03-17:
[image]

coolsnowwolf, 2020-03-17:
UDP 我暂时没抓到包，等更多人反馈

z826540272, 2020-03-17:
> UDP 我暂时没抓到包，等更多人反馈
研发请求:
希望添加Netflix代理的"前置代理"功能,直连能看美国NF的代理速度太慢了,如果能用套在主代理里面速度就快多了.

z826540272, 2020-03-17:
功能名称可命名为
[使用主服务器作为前置代理] 开关 或 [自定义前置代理服务器] 选项

aaaisan, 2020-03-17:
> > UDP 我暂时没抓到包，等更多人反馈
> > 研发请求:
> > 希望添加Netflix代理的"前置代理"功能,直连能看美国NF的代理速度太慢了,如果能用套在主代理里面速度就快多了.

这个功能貌似需要一个中转机？路由器能解决这个问题？

abclai, 2020-03-17:
> 功能名称可命名为
> [使用主服务器作为前置代理] 开关 或 [自定义前置代理服务器] 选项

你这个自己搞个中转服务器就可以！简单来说：A鸡速度快不能看NF，B鸡速度慢能看NF

你可以在A鸡上设置个中转服务，通过中转到B鸡观看NF！比如A鸡原来用的是111端口

你可以吧222端口设置成转发B鸡的，逗比哪里有攻略啦自己看！你说的叫中转服务，要不就去

购买DNS解锁服务，解锁完A鸡也可以看NF

z826540272, 2020-03-17:
> > 功能名称可命名为
> > [使用主服务器作为前置代理] 开关 或 [自定义前置代理服务器] 选项
> 
> 你这个自己搞个中转服务器就可以！简单来说：A鸡速度快不能看NF，B鸡速度慢能看NF
> 
> 你可以在A鸡上设置个中转服务，通过中转到B鸡观看NF！比如A鸡原来用的是111端口
> 
> 你可以吧222端口设置成转发B鸡的，逗比哪里有攻略啦自己看！你说的叫中转服务，要不就去
> 
> 购买DNS解锁服务，解锁完A鸡也可以看NF

我都用鸡场的没有VPS

z826540272, 2020-03-17:
> > > UDP 我暂时没抓到包，等更多人反馈
> > > 研发请求:
> > > 希望添加Netflix代理的"前置代理"功能,直连能看美国NF的代理速度太慢了,如果能用套在主代理里面速度就快多了.
> 
> 这个功能貌似需要一个中转机？路由器能解决这个问题？

是可以做到的.套一层代理就行

coolsnowwolf, 2020-03-17:
你这个需要自己实现，例如写个开机脚本，用第一个 ssr-redir 连接到 B，监听端口为 3333， 然后写个 iptables 规则，把目的地址为B 的流量转发到本地端口例如 4444

然后启用第二个 ssr-redir 连接到 A，端口在路由器上面为 4444 

然后把 NF 的流量转入路由器的 3333 端口中

这种套娃过于复杂了

aaaisan, 2020-03-17:
> > > > UDP 我暂时没抓到包，等更多人反馈
> > > > 研发请求:
> > > > 希望添加Netflix代理的"前置代理"功能,直连能看美国NF的代理速度太慢了,如果能用套在主代理里面速度就快多了.
> > 
> > 
> > 这个功能貌似需要一个中转机？路由器能解决这个问题？
> 
> 是可以做到的.套一层代理就行

就算可以套，你先从解锁机中取回数据，然后再用套一层代理，貌似速度只会更慢吧？合理的方式应该是找一台连接你和解锁机速度都快的VPS，用它来中转吧？

aaaisan, 2020-03-17:
> 你这个需要自己实现，例如写个开机脚本，用第一个 ssr-redir 连接到 B，监听端口为 3333， 然后写个 iptables 规则，把目的地址为B 的流量转发到本地端口例如 4444
> 
> 然后启用第二个 ssr-redir 连接到 A，端口在路由器上面为 4444
> 
> 然后把 NF 的流量转入路由器的 3333 端口中
> 
> 这种套娃过于复杂了

老大，你更新完没？我今天下午准备编译，刚编上你这边就更新代码，三次，又得重来……

Fyred, 2020-03-17:
电脑chrome浏览器观看，gfw模式，netflix.com加入黑名单，播放一切正常，未发现走UDP。netflix又不是p2p播放，走UDP不可思议。

chengming0, 2020-03-17:
三小时前编译完成的包，trojan协议，hk（Netflix）和us两个节点，测试无法观看香港Netflix；
试过多种方法：主服务器us，分流服务器hk，其他尝试gfw、大陆过滤、全局、是否启用udp、pdnsd、dns2sock、本地端口；
均无法正常看hk的Netflix；

aaaisan, 2020-03-17:
> 三小时前编译完成的包，trojan协议，hk（Netflix）和us两个节点，测试无法观看香港Netflix；
> 试过多种方法：主服务器us，分流服务器hk，其他尝试gfw、大陆过滤、全局、是否启用udp、pdnsd、dns2sock、本地端口；
> 均无法正常看hk的Netflix；

刚又更新了代码，重新编译试试

coolsnowwolf, 2020-03-17:
Netflix及AWS IP合集（当下面的不起作用时）： https://cdn.jsdelivr.net/gh/QiuSimons/Netflix_IP/getflix.txt

Netflix IP合集（适合大多数人）： 
https://cdn.jsdelivr.net/gh/QiuSimons/Netflix_IP/NF_only.txt

可以测下这两个IP段是否好用

chengming0, 2020-03-18:
> > 三小时前编译完成的包，trojan协议，hk（Netflix）和us两个节点，测试无法观看香港Netflix；
> > 试过多种方法：主服务器us，分流服务器hk，其他尝试gfw、大陆过滤、全局、是否启用udp、pdnsd、dns2sock、本地端口；
> > 均无法正常看hk的Netflix；
> 
> 刚又更新了代码，重新编译试试

新固件，加两个节点，大陆ip模式，更新IP列表，验证通过。

greekstreet, 2020-03-18:
越稿越复杂，有看netflix的需求，弄个netflix的机场不就好了。我反正加了下面这些域名，用mibox s国际版看netflix一点问题没有。以后有要求HBO, HULU, Disney Plus的，那不是要弄很多？
amazonaws.com
aws.amazon.com
awsstatic.com
fast.com
netflix.com
netflix.net
nflxext.com
nflximg.net
nflxso.net
nflxvideo.net
netflixdnstest0.com
netflixdnstest1.com
netflixdnstest2.com
netflixdnstest3.com
netflixdnstest4.com
netflixdnstest5.com
netflixdnstest6.com
netflixdnstest7.com
netflixdnstest8.com
netflixdnstest9.com
netvigator.com
1e100.net
imsbiz.com
spcsdns.net

实在要弄，还不如在现有的主服务器、游戏UDP中继服务器的基础上，加一个叫“国外流媒体解锁”的选项，这个选项里面可以选择解锁节点，当然这个节点肯定是要机场支持解锁相应流媒体服务的。然后在访问控制里面加一个“国外流媒体解锁”的tab页，设定需要解锁的域名和IP，域名和IP可以根据自己的观看需要添加Netflix, HBO, Hulu, BBC, Disney+, Amazon Prime, TVB等等。单单限定一个Netflix有点太局限。

个人愚见。

coolsnowwolf, 2020-03-18:
本来就不限定 Netflix，属于二次分流

greekstreet, 2020-03-18:
> 本来就不限定 Netflix，属于二次分流

好的，我现在用的还是3月1日的版本，一直没有更新，等有必要升级再试试这些新功能了。感谢各位的贡献做了这么好的工具。

coolsnowwolf, 2020-03-18:
[image]

[image]

guochan2019, 2020-03-18:
支持L大，虽然不用奈飞

hcym, 2020-03-18:
也没看过奶非，回头试试看开，谢谢大佬

coolsnowwolf, 2020-03-18:
这个其实不只是解锁 Netflix ，实际上是分流指定域名和IP到第二个节点上

第二个节点的流量也能套在第一个节点上，做 NAT 落地机

caolixiang, 2020-03-18:
> 这个其实不只是解锁 Netflix ，实际上是分流指定域名和IP到第二个节点上
> 
> 第二个节点的流量也能套在第一个节点上，做 NAT 落地机

用两个机场做了测试，Netflix分流服务器完全没有走流量，流量消耗还是全部在主服务器上。设备MiboxS。所以我也怀疑走了UDP。

dolphinpaopao, 2020-03-18:
机场统计没有路由器防火墙的准吧，太笼统了。看一下路由器的连接信息更有效一些。

dolphinpaopao, 2020-03-18:
> > 这个其实不只是解锁 Netflix ，实际上是分流指定域名和IP到第二个节点上
> > 第二个节点的流量也能套在第一个节点上，做 NAT 落地机
> 
> 用两个机场做了测试，Netflix分流服务器完全没有走流量，流量消耗还是全部在主服务器上。设备MiboxS。所以我也怀疑走了UDP。

机场统计没有路由器防火墙的准吧，太笼统了。看一下路由器的连接信息更有效一些。

pscj, 2020-03-19:
netflix不走UDP的，只是分流的IP段缺乏。
最新的netflix cdn的Ip可以看 https://ipinfo.io/AS2906

SharerMax, 2020-03-19:
目前的 netflix 分流服务器与主服务器相同时相当于禁用了 netflix 分流服务，因为这时为 netflix 配置的域名规则和 ip 规则并没有启用，导致的问题是解锁 netflix 并没有起到作用（规则缺失）。

1. 建议增加禁用选项，其意义是与之前未提供分流服务一样，不使用 netflix 的相关规则
2. 建议保留分流服务器与主服务器相同，但是将 netflix 的相关规则使用起来（#3875）
3. 建议增加 netflix 的 dns 配置， 并将未 netflix 设置的 dns 加入到 netflix 的 ipset 中，确保设置的为 netflix 设置的 dns 请求也走 netflix分流服务

coolsnowwolf, 2020-03-19:
netflix 分流服务器与主服务器相同就无需再分离 ipset 了

SharerMax, 2020-03-19:
> netflix 分流服务器与主服务器相同就无需再分离 ipset 了

netflix 分流服务器与主服务器相同，意味着netflix 分流服务 是可用的，那此时是否应该将 netflix.list（域名规则） 和 netflixip.list（ ip 规则） 合并到gfwlist ( 当然这可以手动将 netflix 的相关配置复制到对应的强制走代理的配置中，但是既然分离了，那么这部分就完全可以通过合并的手段来处理，而且还方便 netflix 服务器的切换 )

抛开 netflix 分流服务器与主服务器相同这个条件， 总的来说 netflix 分流功能其实是区别于主服务的一个优先度较高的 TCP 分流服务，我觉的可以加一个禁用的选项。

还有就是关于 dns 的问题， netflix 分流走的是 8.8.8.8 的固定 dns，因为没有将 8.8.8.8 加入到 netflix 的 ipset 中 这种情况下 dns 的请求并没有走 netflix 的代理服务（除非 netflix 分流服务器与主服务器相同并且国外 dns 选择的是8.8.8.8 ），拿到的ip可能不是最优的。

PS: 先讨论一下，最近加的功能有点多，不要走向臃肿，保持克制，注意身体。 滑稽（：

JeremyPeng, 2020-03-19:
> 我觉得电视盒子好像走UDP，因为如果不选择游戏增强模式，电视盒子会提示代理，将盒子的IP指定走游戏增强模式，就可以正常看，而用电脑相同节点即使不用游戏增强模式也直接可看

没有用到upd，主要ssr+的游戏模式实质是大陆白名单+UDP，你加了以后能看，主要还是白名单起了作用，或者安卓TV系统的dns劫持强制走8.8.8.8时因为upd转发获得了解析起了作用。实际观看是完全不走udp的。

greekstreet, 2020-03-19:
> > 我觉得电视盒子好像走UDP，因为如果不选择游戏增强模式，电视盒子会提示代理，将盒子的IP指定走游戏增强模式，就可以正常看，而用电脑相同节点即使不用游戏增强模式也直接可看
> 
> 没有用到upd，主要ssr+的游戏模式实质是大陆白名单+UDP，你加了以后能看，主要还是白名单起了作用，或者安卓TV系统的dns劫持强制走8.8.8.8时因为upd转发获得了解析起了作用。实际观看是完全不走udp的。

我用的某个机场V2RAY香港节点，目前使用MIBOX S也是没有问题。有问题的朋友，可能是某些域名漏掉了，电视端的APP检测地区比电脑和手机端严格多了。我之前碰到过的情况是电脑、手机播放一点问题没有，电视端有时候提示代理或者不提示但是速度非常慢。我就用LUCI实时信息->连接那个页面，开启“启动DNS查找”。观察打开电视端app时盒子发出的连接，观察到了如下这个域名。
`imsbiz.com`
我上面的回复中也列出了这个域名，查了一下似乎是香港“电讯盈科”的网址。可能Netflix香港用了他们的服务。有可能其他国家地区的又会用其他国家地区宽带运营商的服务。所以他家如果看其他国家地区的，不妨用这个方法观察一下盒子端的netflix到底连了那些域名，再加入到“强制走代理的名单”就差不多可行。当然了，选择的机场节点肯定要支持netflix才行。
`spcsdns.net`
这个是疑似，不清除是不是。
另外小米盒子老会链接以下这个，查了似乎是谷歌的，不知道根netflix有没有关系，反正加进去了。
`1e100.net`

chengming0, 2020-03-19:
绕过大陆模式，Netflix分流成功；
但：观看期间不能暂停，一旦暂停，恢复时就爆“抱歉，服务中断”；

看下你们有没有此问题；

chengming0, 2020-03-19:
补充一下，以上问题是在电脑上操作；
同时，快进也会爆出来同样问题
服务中断，故障码7375

coolsnowwolf, 2020-03-19:
> 目前的 netflix 分流服务器与主服务器相同时相当于禁用了 netflix 分流服务，因为这时为 netflix 配置的域名规则和 ip 规则并没有启用，导致的问题是解锁 netflix 并没有起到作用（规则缺失）。
> 
> 1. 建议增加禁用选项，其意义是与之前未提供分流服务一样，不使用 netflix 的相关规则
> 2. 建议保留分流服务器与主服务器相同，但是将 netflix 的相关规则使用起来（#3875）
> 3. 建议增加 netflix 的 dns 配置， 并将为 netflix 设置的 dns 加入到 netflix 的 ipset 中，确保为 netflix 设置的 dns 请求也走 netflix分流服务

也有道理

coolsnowwolf, 2020-03-19:
> 补充一下，以上问题是在电脑上操作；
> 同时，快进也会爆出来同样问题
> 服务中断，故障码7375

需要确保没有开启 IPv6

coolsnowwolf, 2020-03-19:
建议将 DNS 的 TTL 时间改大，避免 DNS 重新查询造成的问题

GFWLIST 模式分流更可靠，可以作为测试

[image]

zhengpaiboy, 2020-03-20:
> > 补充一下，以上问题是在电脑上操作；
> > 同时，快进也会爆出来同样问题
> > 服务中断，故障码7375
> 
> 需要确保没有开启 IPv6

谢谢，上午验证通过：
之前只做：在global setting里 取消了IPv6支持；

今天上午在Lan和Wan口里，取消了IPv6管理，看Netflix的时候快进和暂停一切正常了；

a8l1vy, 2020-03-20:
希望下一次排查netflix时候，顺便看看我这个情况。顺便看下。

看电影看电视剧都没什么问题，就算出现解锁不能播放（很少），关一下电视盒子网络开关就好了。或者重启下路由。一般都可以解决了。
但是看动漫就很头疼，片源不同，检测不同。
比如动漫A看了一两集出现解锁问题，那这个动漫再怎么试也是不能放了。同一时间，随便打开另一个电影，还是可以看的。重启也不能看动漫A。
继续随便选一个动漫B，大多数情况也不能看。即使路由器开全局，动漫A和动漫B也是不能看的。
我是不停重启路由器和电视盒子，尝试的。再点一个电视剧或者电影，基本都可以看。

解决办法就是手机开全局，在手机上看动漫A动漫B。手机上用规则也是不能看的。手机用规则可以看电影电视剧。

guancn, 2020-03-23:
> 越稿越复杂，有看 netflix 的需求，弄个 netflix 的机场不就好了。我反正加了下面这些域名，用 mibox s 国际版看 netflix 一点问题没有。以后有要求 HBO, HULU, Disney Plus 的，那不是要弄很多？
> amazonaws.com
> aws.amazon.com
> awsstatic.com
> fast.com
> netflix.com
> netflix.net [...]

我是坚决同意这一点的。特别是家里娃是disney plus的重度用户，目前我的netflix和disney plus和翻墙最快的节点是三个不同的节点，能看netflix看不了disney plus,反之一样。盒子要看不同东西前进路由调整节点太痛苦了，不知道能不能加一个国外流媒体解锁选项。。。

dolphinpaopao, 2020-03-23:
> 我是坚决同意这一点的。特别是家里娃是disney plus的重度用户，目前我的netflix和disney plus和翻墙最快的节点是三个不同的节点，能看netflix看不了disney plus,反之一样。盒子要看不同东西前进路由调整节点太痛苦了，不知道能不能加一个国外流媒体解锁选项。。。

现在的奈菲解锁不就是这个功能吗？域名ip可以自定义添加的啊。

guancn, 2020-03-23:
> 现在的奈菲解锁不就是这个功能吗？域名 ip 可以自定义添加的啊。

不知道disneyplus的ip段。。。。，只添加disneyplus.com的域名显然木有啥效果。。。

feelice, 2020-03-24:
> > > 盒子里YouTube走tcp，这个我试过。手机上app可能有
> > 
> > 
> > 说的是Netflix……而且不同盒子也不一样吧，我的是小米盒子4国际版
> 
> 是的，我也是小米盒子S也是一样的问题啦！而且IOS也是会出现这个问题啦！感觉应该是有UDP检测或是走UDP啦！要不不可能开启游戏模式就OK

请教mi box s在ssr plus怎么设置，不管怎么设置总是提示用代理，哪个设置可以达到隔壁像访问控制一样的效果

txdywy, 2020-03-24:
ip不行吧

feelice, 2020-03-24:
> […](#)

可以的，pc和手机没问题，就盒子不行

coolsnowwolf, 2020-03-25:
这个议题可以关闭了，规则本身已经没有问题

closed 2020-03-25 by coolsnowwolf

0neday, 2020-08-29:
> 你这个需要自己实现，例如写个开机脚本，用第一个 ssr-redir 连接到 B，监听端口为 3333， 然后写个 iptables 规则，把目的地址为B 的流量转发到本地端口例如 4444
> 
> 然后启用第二个 ssr-redir 连接到 A，端口在路由器上面为 4444
> 
> 然后把 NF 的流量转入路由器的 3333 端口中
> 
> 这种套娃过于复杂了

`# netflix
iptables -t nat -A SHADOWSOCKS -d 23.0.0.0/8 -p tcp -j REDIRECT --to-ports 5678
iptables -t nat -A SHADOWSOCKS -d 34.0.0.0/8 -p tcp -j REDIRECT --to-ports 5678
iptables -t nat -A SHADOWSOCKS -d 35.0.0.0/8 -p tcp -j REDIRECT --to-ports 5678
iptables -t nat -A SHADOWSOCKS -d 44.0.0.0/8 -p tcp -j REDIRECT --to-ports 5678
iptables -t nat -A SHADOWSOCKS -d 45.0.0.0/8 -p tcp -j REDIRECT --to-ports 5678
iptables -t nat -A SHADOWSOCKS -d 52.0.0.0/8 -p tcp -j REDIRECT --to-ports 5678
iptables -t nat -A SHADOWSOCKS -d 54.0.0.0/8 -p tcp -j REDIRECT --to-ports 5678

# redirect all 80/443 packets to 1234 port
iptables -t nat -A SHADOWSOCKS -p tcp -j REDIRECT --to-ports 1234

iptables -t nat -I PREROUTING -p tcp -m multiport --dports 80,443 -j SHADOWSOCKS

`

Heiban030, 2021-01-04:
> Netflix及AWS IP合集（当下面的不起作用时）： https://cdn.jsdelivr.net/gh/QiuSimons/Netflix_IP/getflix.txt
> 
> Netflix IP合集（适合大多数人）：
> https://cdn.jsdelivr.net/gh/QiuSimons/Netflix_IP/NF_only.txt
> 
> 可以测下这两个IP段是否好用

找了很多nf域名和ip都不起作用，主线路和分流线路同时工作导致无法使用网飞。
改成Netflix及AWS终于好了，但是这会不会导致其它流媒体平台也通过分流线路打开？
我的主线路和分流线路不是同一个国家的。

0neday, 2021-01-20:
https://gist.github.com/0neday/b571d00ddfebe0a9f1333f8817f582a1
不解释

xhqpp, 2021-02-06:
1.Netflix相关的IP要走代理
2.Netflix相关的域名要走代理
3.Netflix相关的DNS请求要走代理
4.如果开启IPv6 相关的IP 域名 DNS也要走IPv6代理

felitsa, 2021-12-17:
> 1.Netflix相关的IP要走代理 2.Netflix相关的域名要走代理 3.Netflix相关的DNS请求要走代理 4.如果开启IPv6 相关的IP 域名 DNS也要走IPv6代理

请教下3如何保证。我在passwall中已经设置了1和2，电脑和手机可以看netflix，索尼电视仍然看不了，报各种nw-2-5之类的错误。我使用的主节点本身就解锁了奈飞。

919735344, 2022-04-17:
奈飞分流没什么大的问题，但是迪士尼总是出现电脑不能看，手机能看的情况。

befantasy, 2022-11-09:
> 

最近用ssr plus，apple TV Netflix分流一直有问题。找到了这个ISSUE。

自行排查了一圈，最后锁定问题是DNS，apple TV Netflix的DNS查询会走main server，而不是Netflix分流server。
然而我的Main server是韩国（无解锁），Netflix分流server是美国（非自制剧解锁），导致DNS返回的IP可能是韩国的CDN地址。有概率会被检测到使用了代理。
在apple TV上返回的片源也是韩国区域的，观看不同片源的时候，甚至同一个片源，有时候可以看，有时候不能看。。。

这个DNS问题似乎是之前的讨论都忽略的问题。很多人都把他归咎到分流server有问题，但我可以100%确认server可解锁。
有些人说换了server就可以了，我估计是main server和Netflix 分流server恰巧换到了同一个区域，那么通过main server查询DNS返回的CDN节点和Netflix分流节点在同一个区域，则没有问题。

[1 more reply]