coolsnowwolf/lede #3851: 建议大佬更新下frpc和frps，以支持tls连接

wowaqly, 2020-03-17:
建议大佬更新下frpc和frps，以支持tls链接。
frpc中 tls_enable = true，比现在的加密再压缩要安全，还省CPU
frps中开启tls_only = true 更加安全

lwz322, 2020-03-17:
frpc版本0.25.0+, frps版本0.32.0+，前几天的commit就更新到了

luci-app-frps的PR刚刚提交，[为luci-app-frps添加tls-only及extra-setting支持 #3859](https://github.com/coolsnowwolf/lede/pull/3859)

closed 2020-03-18 by coolsnowwolf

wowaqly, 2020-03-18:
我今天刚编译了一个估计，在frpc和frps中均未有tls选项，是不是要自己去ssh改配置文件？

wowaqly, 2020-03-18:
> frpc版本0.25.0+, frps版本0.32.0+，前几天的commit就更新到了
> 
> luci-app-frps的PR刚刚提交，[为luci-app-frps添加tls-only及extra-setting支持 #3859](https://github.com/coolsnowwolf/lede/pull/3859)

我刚才重新编译了一次估计，在frps和frpc中均为出现tls选项。

lwz322, 2020-03-18:
我也重新编译了下，在我的luci-app-frps下是有TLS选项的
[image]
你可以注意下是不是源码没有更新到新版（也有可能是编译中间过程的缓存），版本号是0.0.2-1