coolsnowwolf/lede #3852: SSR PLUS+功能建议

KevinZjYang, 2020-03-17:
有些机场很多节点当时可以使用，然后过一段时间就不行了。希望增加后台自动测试联通性，自动切换节点的功能。
[image]
我最初认为这个可以切换，但是实际并没有效果。
我不知道这个服务器故障节点自动切换的功能是在ping服务器能否访问，还是测试google.com能否访问。
我认为测试google.com更靠谱。因为我发现很多节点，像下图显示这样连接测试是没问题的，其实是访问不了google的。希望能改进一下。我说的都是我的猜测，不对的还请包涵。
[image]

SharerMax, 2020-03-17:
判断依据 测试google.com能否访问

KevinZjYang, 2020-03-17:
@SharerMax 但是在服务器节点界面显示连接测试“ok”，应用这个节点。
然后去状态页面检查连通性，却是失败的。同时浏览器也是访问不了Google的。
是不是说明这个地方其实是有问题的。

coolsnowwolf, 2020-03-17:
一个 是端口连通性，一个是 Google 是否可达，切换根据是后者

KevinZjYang, 2020-03-17:
@coolsnowwolf 明白了，那按道理说这个很好用啊。但是大佬我发现我的不自动切换啊，切换也是开启的。就像我上面发的图这样设置的。

KevinZjYang, 2020-03-17:
@coolsnowwolf  @SharerMax 我刚才发现，有人遇到和我一样的问题。#3845。求大佬解决啊。万分感谢。

garypang13, 2020-03-17:
https://github.com/coolsnowwolf/lede/issues/3845#issuecomment-600068428

coolsnowwolf, 2020-03-18:
并不是根据 ping，也不是端口 ok，而是根据是否可以 wget google首页决定是否切换

garypang13, 2020-03-18:
> 并不是根据 ping，也不是端口 ok，而是根据是否可以 wget google首页决定是否切换

wget google只能检测正在使用的线路是否可用,你要检测主线路是否已恢复正常,不可能先切换到主线路来wget google, 现在ssr+检测主线路是否已恢复正常用的就是tcp,ping.

kktt007, 2020-03-18:
现在每天都抽风,看到大雕每天在更新,我们的希望victory

KevinZjYang, 2020-03-18:
> > 并不是根据 ping，也不是端口 ok，而是根据是否可以 wget google首页决定是否切换
> 
> wget google只能检测正在使用的线路是否可用,你要检测主线路是否已恢复正常,不可能先切换到主线路来wget google, 现在ssr+检测主线路是否已恢复正常用的就是tcp,ping.

我的意思就是检测主线路啊，主线路明明访问不了Google，但是不切换到其他线路去。我只想让它发现主线路不通后，自己换一个。

garypang13, 2020-03-18:
> > > 并不是根据 ping，也不是端口 ok，而是根据是否可以 wget google首页决定是否切换
> > 
> > 
> > wget google只能检测正在使用的线路是否可用,你要检测主线路是否已恢复正常,不可能先切换到主线路来wget google, 现在ssr+检测主线路是否已恢复正常用的就是tcp,ping.
> 
> 1、我的意思就是检测主线路（或者说当前使用的线路），主线路明明访问不了Google，但是不切换到其他线路去。我只想让它发现主线路不通后，自己换一个。可能我描述的不清楚。
> 2、这个服务器节点故障自动切换，不就应该每隔一定时间，检测当前使用的线路是否wget google，不能就换一个。
> 3、我不明白为什么老在纠结主线路的问题，意思是换一个线路后，还得每次去看上一个线路是否能wget google，没必要啊，能用就行，用那个线路都是用。
> 以上只是就这个问题而言的一些想法，希望这个项目越来越好，大雕辛苦了。

你说的第一点不会出现,确定有添加备用线路,且主线路也有开启自动切换.
问题是ssr+在切换到备用线路后,马上又会检测主线路是否已经恢复了正常,在没有切换到主线路的情况下,要检测主线路是否恢复正常就只能ping主线路,如果线路有中转的话结果就会不断在主线路和备用线路之间来回切换.

KevinZjYang, 2020-03-18:
> 你说的第一点不会出现,确定有添加备用线路,且主线路也有开启自动切换.
> 问题是ssr+在切换到备用线路后,马上又会检测主线路是否已经恢复了正常,在没有切换到主线路的情况下,要检测主线路是否恢复正常就只能ping主线路,如果线路有中转的话结果就会不断在主线路和备用线路之间来回切换.

那这么来说，我能不能关掉这个检测主线路的功能，我只要切换的线路正常工作就好。

KevinZjYang, 2020-03-19:
> 那这么来说，我能不能关掉这个检测主线路的功能，我只要切换的线路正常工作就好。

KevinZjYang, 2020-03-19:
2020-03-19 16:53:35 switch to default [itunes.apple.com.ajax.microsoft.com.www.bing.com.cn2.zhileshu.com] proxy!
2020-03-19 16:53:34 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
2020-03-19 16:53:33 Main server is avilable.
2020-03-19 16:53:31 Current server is not default Main server, try to switch back.
2020-03-19 16:52:31 ShadowsocksR server switch OK
2020-03-19 16:52:29 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
2020-03-19 16:52:28 Another server is avilable, now switching server.
2020-03-19 16:52:28 Current server error, try to switch another server.
2020-03-19 16:51:10 switch to default [itunes.apple.com.ajax.microsoft.com.www.bing.com.cn2.zhileshu.com] proxy!
2020-03-19 16:51:08 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
2020-03-19 16:51:07 Main server is avilable.
2020-03-19 16:51:05 Current server is not default Main server, try to switch back.
2020-03-19 16:50:05 ShadowsocksR server switch OK
2020-03-19 16:50:04 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
2020-03-19 16:50:03 Another server is avilable, now switching server.
2020-03-19 16:50:03 Current server error, try to switch another server.
2020-03-19 16:48:43 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
2020-03-19 16:48:41 ShadowsocksR No Problem.

@garypang13 我仔细看了一下日志，确实和你说的一样。@coolsnowwolf 我就想说能不能不去检测主服务器是否可用，切换到可用的，就不管了。是不是主服务器不重要。或者有一个开关，让用户去选择是否去检测主服务器。

garypang13, 2020-03-19:
@KevinZjYang 我的曲线解决办法是将检测主线路是否恢复的等待时间尽量调长,比如半小时

KevinZjYang, 2020-03-19:
> @KevinZjYang 我的曲线解决办法是将检测主线路是否恢复的等待时间尽量调长,比如半小时

这样也存在问题。
1、发现主线路wget google不通，切换线路。
2、30分钟后没检测主服务器可用（其实wget google还是不通）。切换到主服务器。然后在接下来的30分钟内一直都是无法访问的。
3、30分钟又到了，发现主线路wget google不通，切换线路。接下来的30分钟是可用的。
像上面这样循环，一个小时的时间，有半个小时可用，半个小时不可用。

garypang13, 2020-03-19:
> > @KevinZjYang 我的曲线解决办法是将检测主线路是否恢复的等待时间尽量调长,比如半小时
> 
> 这样也存在问题。
> 1、发现主线路wget google不通，切换线路。
> 2、30分钟后检测主服务器可用（其实wget google还是不通）。切换到主服务器。然后在接下来的30分钟内一直都是无法访问的。
> 3、30分钟又到了，发现主线路wget google不通，切换线路。接下来的30分钟是可用的。
> 像上面这样循环，一个小时的时间，有半个小时可用，半个小时不可用。

我只是在检测主服务器是否可用前加了等待时间,其他都没变,切换回主服务器后,会立即wget google,如果发现wget google失败,会马上切换到备用服务器,

KevinZjYang, 2020-03-20:
> 我只是在检测主服务器是否可用前加了等待时间,其他都没变,切换回主服务器后,会立即wget google,如果发现wget google失败,会马上切换到备用服务器,

这是我早上的日志。用事实说话。
```2020-03-20 11:37:48 ShadowsocksR server switch OK
2020-03-20 11:37:47 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
2020-03-20 11:37:45 Another server is avilable, now switching server.
2020-03-20 11:37:45 Current server error, try to switch another server.
2020-03-20 11:07:19 switch to default [itunes.apple.com.ajax.microsoft.com.www.bing.com.zjg.zhileshu.com] proxy!
2020-03-20 11:07:18 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
2020-03-20 11:07:17 Main server is avilable.
2020-03-20 11:07:15 Current server is not default Main server, try to switch back.
2020-03-20 10:37:15 ShadowsocksR server switch OK
2020-03-20 10:37:14 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
2020-03-20 10:37:13 Another server is avilable, now switching server.
2020-03-20 10:37:13 Current server error, try to switch another server.```
首先主服务器是一直wget google失败的。设置的检测时间是30分钟，每次切换会主服务器后并没有马上wget google，而是等到30分钟后才wget google。
这就证明了我上面的说法。设置30分钟检测，一个小时里只有一半时间可用。

garypang13, 2020-03-20:
> 这是我早上的日志。用事实说话。
>
> ```
> 2020-03-20 11:37:48 ShadowsocksR server switch OK
> 2020-03-20 11:37:47 V2Ray 4.22.1 (OpenWrt) Lean (go1.14 linux/amd64) Started!
> 2020-03-20 11:37:45 Another server is avilable, [...]

不能直接改这个检查周期,我是修改/usr/bin/ssr-switch文件
在if test_proxy上一行加上
sleep 3600

KevinZjYang, 2020-03-20:
@garypang13 首先，感谢你告诉我切换代码的位置。在知道了切换的代码文件在/usr/bin/ssr-switch这个路径后，我打开学习了一下。
下面是原代码：
```
start() {
	#不支持kcptun启用时的切换
	[ $(uci_get_by_name $DEFAULT_SERVER kcp_enable) = "1" ] && return 1

	while [ "1" == "1" ]; do #死循环
		sleep $cycle_time
		LOGTIME=$(date "+%Y-%m-%d %H:%M:%S")
		#判断当前代理是否为缺省服务器
		if [ "$CURRENT_SERVER" != "$DEFAULT_SERVER" ]; then
			#echo "not default proxy"
			echo "$(date "+%Y-%m-%d %H:%M:%S") Current server is not default Main server, try to switch back." >>/tmp/ssrplus.log
			#检查缺省服务器是否正常
			if test_proxy $DEFAULT_SERVER; then
				#echo "switch to default proxy"
				echo "$(date "+%Y-%m-%d %H:%M:%S") Main server is avilable." >>/tmp/ssrplus.log
				#缺省服务器正常，切换回来
				CURRENT_SERVER=$DEFAULT_SERVER
				switch_proxy $CURRENT_SERVER
				echo "$(date "+%Y-%m-%d %H:%M:%S") switch to default ["$(uci_get_by_name $CURRENT_SERVER server)"] proxy!" >>/tmp/ssrplus.log
				continue
			else
				echo "$(date "+%Y-%m-%d %H:%M:%S") Main server is NOT avilable.Continue using current server." >>/tmp/ssrplus.log
			fi
		else 
		   echo "$(date "+%Y-%m-%d %H:%M:%S") 当前代理是默认服务器" >>/tmp/ssrplus.log
		fi
		#判断当前代理是否正常
		check_proxy
		current_ret=$?
		if [ "$current_ret" == "1" ]; then
			#当前代理错误，判断有无可用的服务器
			#echo "current error"
			echo "$(date "+%Y-%m-%d %H:%M:%S") Current server error, try to switch another server." >>/tmp/ssrplus.log
			select_proxy
			if [ "$ENABLE_SERVER" != nil ]; then
				#有其他服务器可用，进行切换
				#echo $(uci_get_by_name $new_proxy server)
				echo "$(date "+%Y-%m-%d %H:%M:%S") Another server is avilable, now switching server." >>/tmp/ssrplus.log
				CURRENT_SERVER=$ENABLE_SERVER
				switch_proxy $CURRENT_SERVER
				normal_flag=1
				echo "$(date "+%Y-%m-%d %H:%M:%S") ShadowsocksR server switch OK" >>/tmp/ssrplus.log
			else
				switch_proxy $CURRENT_SERVER
				normal_flag=1
				echo "$(date "+%Y-%m-%d %H:%M:%S") Try restart current server." >>/tmp/ssrplus.log
			fi
		else
			normal_flag=0
			echo "$(date "+%Y-%m-%d %H:%M:%S") ShadowsocksR No Problem." >>/tmp/ssrplus.log
		fi
	done
}
```
学习一下后我终于明白了，问题在哪儿了。我反复发现切换到默认服务器后，并不会立刻去检测默认服务器是否可用的。而是等待我们设置的检测时间后再去检查。
这是因为上述代码，在切换完默认服务器后，continue直接跳出循环了，也就是下面那些检查的代码是不会运行的。
所以解决办法有两个：
1、删掉代码里的continue，但是这会导致在切换到默认服务器后，立刻发现默认服务器不可用，然后马上切到一个可用的服务器。在等待完我们设置的检测时间后，重复上述过程。
2、直接删掉检测默认服务器的代码，这样就只会在我们设置的检测时间里，定时检测当前使用的服务器是否可用。可用不执行任何操作，不可用切换一个可用的服务器。
我的需求第二种方法可以满足。
下面是我修改过的代码。
```
start() {
	#不支持kcptun启用时的切换
	[ $(uci_get_by_name $DEFAULT_SERVER kcp_enable) = "1" ] && return 1

	#while [ "1" == "1" ]; do #死循环
	#	sleep $cycle_time
	#	LOGTIME=$(date "+%Y-%m-%d %H:%M:%S")
	#	#判断当前代理是否为缺省服务器
	#	if [ "$CURRENT_SERVER" != "$DEFAULT_SERVER" ]; then
	#		#echo "not default proxy"
	#		echo "$(date "+%Y-%m-%d %H:%M:%S") Current server is not default Main server, try to switch back." >>/tmp/ssrplus.log
	#		#检查缺省服务器是否正常
	#		if test_proxy $DEFAULT_SERVER; then
	#			#echo "switch to default proxy"
	#			echo "$(date "+%Y-%m-%d %H:%M:%S") Main server is avilable." >>/tmp/ssrplus.log
	#			#缺省服务器正常，切换回来
	#			CURRENT_SERVER=$DEFAULT_SERVER
	#			switch_proxy $CURRENT_SERVER
	#			echo "$(date "+%Y-%m-%d %H:%M:%S") switch to default ["$(uci_get_by_name $CURRENT_SERVER server)"] proxy!" >>/tmp/ssrplus.log
	#			continue
	#		else
	#			echo "$(date "+%Y-%m-%d %H:%M:%S") Main server is NOT avilable.Continue using current server." >>/tmp/ssrplus.log
	#		fi
	#	else 
	#	   echo "$(date "+%Y-%m-%d %H:%M:%S") 当前代理是默认服务器" >>/tmp/ssrplus.log
	#	fi
		#判断当前代理是否正常
		echo "$(date "+%Y-%m-%d %H:%M:%S") Start to determine whether the current server is available" >>/tmp/ssrplus.log
		check_proxy
		current_ret=$?
		if [ "$current_ret" == "1" ]; then
			#当前代理错误，判断有无可用的服务器
			#echo "current error"
			echo "$(date "+%Y-%m-%d %H:%M:%S") Current server error, try to switch another server." >>/tmp/ssrplus.log
			select_proxy
			if [ "$ENABLE_SERVER" != nil ]; then
				#有其他服务器可用，进行切换
				#echo $(uci_get_by_name $new_proxy server)
				echo "$(date "+%Y-%m-%d %H:%M:%S") Another server is avilable, now switching server." >>/tmp/ssrplus.log
				CURRENT_SERVER=$ENABLE_SERVER
				switch_proxy $CURRENT_SERVER
				normal_flag=1
				echo "$(date "+%Y-%m-%d %H:%M:%S") ShadowsocksR server switch OK" >>/tmp/ssrplus.log
			else
				switch_proxy $CURRENT_SERVER
				normal_flag=1
				echo "$(date "+%Y-%m-%d %H:%M:%S") Try restart current server." >>/tmp/ssrplus.log
			fi
		else
			normal_flag=0
			echo "$(date "+%Y-%m-%d %H:%M:%S") ShadowsocksR No Problem." >>/tmp/ssrplus.log
		fi
	done
}

```
上面注释掉了切换默认服务器的代码，并在开始检测服务器时，输出一条日志。
现在这样，我的需求是可以满足了。
最后希望项目越来越好。
@coolsnowwolf

garypang13, 2020-03-20:
按照我的只需要加一行代码不是更简单么,且不影响主线路和备用线路的区分,备用线路只是临时代替下,主线路恢复正常后还能自动切换成主线路.

KevinZjYang, 2020-03-20:
> 按照我的只需要加一行代码不是更简单么,且不影响主线路和备用线路的区分,备用线路只是临时代替下,主线路恢复正常后还能自动切换成主线路.

你的方法有问题啊，你的意思是在检查是否是主服务器时，等待一定的时间，但是不管等待多长时间，还是会切换回去。切换回去后依旧不会马上检测服务器是否可以，要等待我们设置的检测时间。
这没意义啊。

garypang13, 2020-03-20:
等待时间你可以设为1小时,1天,故障一般都是临时的.
你设置的检测时间可以是1秒可以是5秒,
假如你设置的等待时间是1天,检测时间是5秒,那可能发生的最坏的情况是主线路故障了一天导致5秒上不了外网,刚好那五秒你又正在上外网.

KevinZjYang, 2020-03-20:
> 等待时间你可以设为1小时,1天,故障一般都是临时的.
> 你设置的检测时间可以是1秒可以是5秒,
> 假如你设置的等待时间是1天,检测时间是5秒,那可能发生的最坏的情况是主线路故障了一天导致5秒上不了外网,刚好那五秒你又正在上外网.

不讨论了，我们想法不一样。我是尝试解决问题，而你在逃避问题。

coolsnowwolf, 2020-03-20:
我明白你的想法。

所以其实在主程序中，如果是带 switch_server 参数启动的，并不会把当前的 ssr-switch 干掉，也是为了可以这样操作

closed 2020-03-25 by KevinZjYang