coolsnowwolf/lede #3853: ssrplus+的功能方面的提问

BGzErol, 2020-03-17:
L大是否考虑下加入UDP端口代理的自定义，感觉可以添加进去，目前Google方面有quic的原因导致安卓端YouTube看视频直播卡顿严重，而在ssrp中upd代理端口又是和主tcp代理的端口同步的，如果开启UDP代理的话这样以来又有了个BT和P2P的问题，望L大能够考虑考虑加入这个功能

coolsnowwolf, 2020-03-17:
quic udp 端口默认被抛弃了

BGzErol, 2020-03-17:
> quic udp 端口默认被抛弃了

不过我这边使用下来确实是没打开UDP代理看油管非常卡，打开了后就秒开了，这就很奇怪了

BGzErol, 2020-03-21:
> quic udp 端口默认被抛弃了

L大请问一下ssrplus+有什么方法可以自定义UDP端口吗，我在/var/etc/config下找到了UDP配置端口的地方但是设置后重新应用配置就会被该回去，设置为只读也不行，但是我确实需要TCP全端口代理+UDP部分常用端口代理

BraveSail, 2020-03-31:
> > quic udp 端口默认被抛弃了
> 
> L大请问一下ssrplus+有什么方法可以自定义UDP端口吗，我在/var/etc/config下找到了UDP配置端口的地方但是设置后重新应用配置就会被该回去，设置为只读也不行，但是我确实需要TCP全端口代理+UDP部分常用端口代理

https://github.com/coolsnowwolf/lede/blob/08b73fa42d75dedbdb2ac08d9040a9ec859ba9e7/package/lean/luci-app-ssr-plus/root/etc/init.d/shadowsocksr

224行自己改

BGzErol, 2020-03-31:
> > > quic udp 端口默认被抛弃了
> > 
> > 
> > L大请问一下ssrplus+有什么方法可以自定义UDP端口吗，我在/var/etc/config下找到了UDP配置端口的地方但是设置后重新应用配置就会被该回去，设置为只读也不行，但是我确实需要TCP全端口代理+UDP部分常用端口代理
> 
> https://github.com/coolsnowwolf/lede/blob/08b73fa42d75dedbdb2ac08d9040a9ec859ba9e7/package/lean/luci-app-ssr-plus/root/etc/init.d/shadowsocksr
> 
> 224行自己改

这里我看过很多次了，由于不懂代码虽然尝试过一些想法但最终还是没能分离TCP和UDP的代理端口，只能实现自定义代理端口，如果可以的话能够请教一下如果将TCP和UDP的端口分离，达到TCP全端口+UDP代理80,443端口的效果吗

BraveSail, 2020-03-31:
> 这里我看过很多次了，由于不懂代码虽然尝试过一些想法但最终还是没能分离TCP和UDP的代理端口，只能实现自定义代理端口，如果可以的话能够请教一下如果将TCP和UDP的端口分离，达到TCP全端口+UDP代理80,443端口的效果吗

看样子是iptables，我也不怎么懂，我的需求只是添加一个端口

BraveSail, 2020-03-31:
不过我注意到就是这里的规则是比访问控制里面强制走代理IP优先级高的，要不然我也不会为了个ip去改这里

BGzErol, 2020-03-31:
> 看样子是iptables，我也不怎么懂，我的需求只是添加一个端口

这样啊，不过也谢谢了，我再找找别的方法看看能不能实现

closed 2020-07-04 by coolsnowwolf

Gokou-Ruri, 2020-08-10:
UDP和TCP一样只会代理常用端口啊，怎么可能走P2P流量？

BGzErol, 2020-08-11:
> UDP和TCP一样只会代理常用端口啊，怎么可能走P2P流量？

我用的全端口代理