coolsnowwolf/lede #3858: 恳请能够将N1的编译默认支持无线驱动

shinyzgj, 2020-03-17:
鉴于大雕集成划时代意义的SOCKS5 模式，深表感激。
谨代表广大N1党，恳请能够将N1的编译的，无线做一个傻瓜化的默认。虽然不用它的无线，主要方便调试

LeeMenHin, 2020-03-17:
關了吧

hcym, 2020-03-18:
恩三有标准答案，或者用树3编译

shinyzgj, 2020-03-18:
我有攻略，我会搞无线的 ，就是希望有个默认的细分

22613, 2020-03-18:
自己改好后pull过来啊。

dolphinpaopao, 2020-03-18:
开源项目，开源开源，不是一切都要依靠作者的，作者不是神仙。开源就是鼓励自己动手丰衣足食，不会就学习。issue，issue，这也不是issue啊。

hcym, 2020-03-18:
默认就是自己改了，我都是改默认，armv8也可以无线，只有ss+，不然大雕的默认太夸张

shinyzgj, 2020-03-18:
> 自己改好后pull过来啊。
我的攻略是这样的啊
增加WIFI编译
Firmware-
增加
brcmfmac-firmware-43430-sdio
brcmfmac-firmware-43455-sdio
brcmfmac-firmware-usb

Kernel modules-Network Devices
增加
kmod-b44

Kernel modules-Wireless Drivers
增加
kmod-brcmfmac
Enable SDIO bus interface support
Enable USB bus interface support
取消 Enabel PCIE bus ...
选择完可能会自动增加 
kmod-brcmutil
kmod-cfg80211

Network-
增加
Hostapd  (full版本)

我不懂代码 不会PULL

shinyzgj, 2020-03-18:
各位看到的，大佬你们会PULL这个功能吗？

ElectronicElephant, 2020-03-22:
没有必要~

不要随随便便说“代表balabala”之类的话

hcym, 2020-03-22:
意义不大，这个无线就是便携式小ap

shinyzgj, 2020-03-23:
> 意义不大，这个无线就是便携式小ap

不是AP 仅仅调试时候用一下的。这货WIFI很烂的

lizhenglin2002, 2020-04-01:
请问编译N1应该选什么类型设备

closed 2020-07-04 by coolsnowwolf