coolsnowwolf/lede #3860: 最新版adbyby plus+的Adblock Plus 规则数据库不自动更新

jxph1123, 2020-03-18:
(1) 具体问题

最新版adbyby plus+的Adblock Plus 规则数据库不自动更新，勾选了每天凌晨6点更新规则并重启，但不自动更新，记录为0条，手动更新正常。
以前是重启或6点都能自动更新。

(2) 路由器型号和固件版本

MediaTek MT7621 ver:1 eco:3 x 4
OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7)

(3) 详细日志

Wed Mar 18 06:00:00 2020 cron.info crond[8466]: USER root pid 7971 cmd /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1
Wed Mar 18 06:00:26 2020 cron.info crond[8094]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 06:00:35 2020 cron.info crond[8386]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 06:00:39 2020 cron.info crond[8674]: crond (busybox 1.30.0) started, log level 5

lengyuqingqiu, 2020-03-18:
六点更新人太多吧。自己手动更一下。而且ipv6似乎更新不了。得关掉。

jxph1123, 2020-03-18:
以前能正常更，ipv6都是关了的

hplee0120, 2020-03-18:
同样的日志，6点过后点了下手动更新发现有刷新规则，所以应该是6点的时候没自动更新成功

iamggs, 2020-03-19:
根源问题就是固定6点升级，自己手动在计划任务中修改为其他时间都可以正常更新，但更新后会自动把计划任务中的时间又改回6点。只有希望修改程序，把更新时间开放给使用者自定义。

hplee0120, 2020-03-19:
编译最新固件后发现跟你一样的日志，那个hosts规则似乎不自动更新，然后是那个订阅第三方规则，明明是1万的规则数订阅出来提示4w+。几千的规则数刷新出来3w+。然后我有点感觉这个hosts规则似乎不起作用了。之前某些可以屏蔽掉的广告现在出来了。

closed 2020-03-20 by coolsnowwolf

cike2056, 2021-06-29:
系统->计划任务: 0 6 * * * /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1
改成12小时:        0 */12 * * * /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1