coolsnowwolf/lede #3862: 最近系统经常假死机，系统无法访问，且重启后ssr-plus 不能自动运行

AmibaWorld, 2020-03-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最近系统经常假死机，系统无法访问，且重启后ssr-plus 不能自动运行，我已经重新编译去掉了 
dnscrypt-proxy  & dnsforwarder ，但是因为系统远程只能刷upgrade，无法刷openwrt-ramips-mt7621-ubiquiti_edgerouterx-initramfs-kernel，所以系统是不是仍然存在dnscrypt-proxy  & dnsforwarder ，求助有没有办法远程刷 openwrt-ramips-mt7621-ubiquiti_edgerouterx-initramfs-kernel而不是upgrade，谢谢，
### (2) 路由器型号和固件版本

A：ubnt er-x

### (3) 详细日志

A：Wed Mar 18 10:37:57 2020 daemon.info unbound: [6243:0] info: start of service (unbound 1.9.5).
Wed Mar 18 10:38:04 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:38:13 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:38:28 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:38:40 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:38:55 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:39:10 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:39:25 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:39:43 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:39:58 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:40:00 2020 cron.info crond[3806]: USER root pid 7152 cmd /etc/init.d/dnsforwarder health
Wed Mar 18 10:40:19 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:40:34 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:40:58 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:41:13 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:41:40 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:41:55 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:42:25 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:42:40 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:43:13 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:43:28 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:44:04 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:44:19 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:44:58 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:45:00 2020 cron.info crond[3806]: USER root pid 9515 cmd /etc/init.d/dnsforwarder health
Wed Mar 18 10:45:13 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:45:55 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:46:10 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:46:55 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:47:10 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:47:52 2020 cron.info crond[10981]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 10:47:54 2020 daemon.err pdnsd[11213]: Could not bind to udp socket: Address in use
Wed Mar 18 10:47:54 2020 daemon.info pdnsd[11224]: pdnsd-1.2.9b-par starting.
Wed Mar 18 10:47:56 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Mar 18 10:47:56 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Mar 18 10:47:58 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:48:13 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:49:04 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:49:19 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:50:00 2020 cron.info crond[10981]: USER root pid 12614 cmd /etc/init.d/dnsforwarder health
Wed Mar 18 10:50:13 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:50:28 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:51:25 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:51:40 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:52:41 2020 daemon.info dnscrypt-proxy[2463]: dnscrypt-proxy Refetching server certificates
Wed Mar 18 10:52:56 2020 daemon.err dnscrypt-proxy[2463]: dnscrypt-proxy Unable to retrieve server certificates
Wed Mar 18 10:53:49 2020 daemon.warn pdnsd[11224]: Caught signal 15. Exiting.
Wed Mar 18 10:53:50 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Mar 18 10:53:50 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Mar 18 10:53:53 2020 cron.info crond[15130]: crond (busybox 1.30.0) started, log level 5
Wed Mar 18 10:53:55 2020 daemon.err pdnsd[15359]: Could not bind to udp socket: Address in use

coolsnowwolf, 2020-03-18:
卸载 dnscrypt-proxy， 端口冲突了

daemon.err pdnsd[15359]: Could not bind to udp socket: Address in use

closed 2020-03-18 by coolsnowwolf

AmibaWorld, 2020-03-19:
我删除了 dnscrypt-proxy，结果杯具了：系统崩溃，无法正常引导了，获取不到正常ip
无奈只好重新刷原厂固件，提示 dnscrypt-proxy  该目录  dead-directory