coolsnowwolf/lede #3869: 最新版本代码frps服务器选项错误

nicepass, 2020-03-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新代码，FRPS-服务器 选项卡错误，无法设置服务器参数。

### (2) 路由器型号和固件版本

A：X86-64，OpenWrt R20.3.11 / LuCI Master (git-20.062.50551-a6189c5)

### (3) 详细日志

A：
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/services/frps/server'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:38: /usr/lib/lua/luci/model/cbi/frps/server.lua:28: ')' expected (to close '(' at line 27) near 'translatef'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

Awrtiger, 2020-03-18:
看了一下提交记录，少了一个逗号。
ssh 连接路由器，修改`/usr/lib/lua/luci/model/cbi/frps/server.lua` 第27行后面加个英文的逗号","。

lwz322, 2020-03-18:
之前提交的时候不小心漏了个逗号，非常抱歉...已提交PR

nicepass, 2020-03-18:
L大的FRPC客户端目前是没有TLS选项的，是否要等L大更新？

lwz322, 2020-03-18:
看了下这个仓库的luci-app-frpc，确实没有这个功能，之后等熟悉代码的人来更新吧；
如果临时要用的话，可以考虑[kuoruan/luci-app-frpc](https://github.com/kuoruan/luci-app-frpc)；
长期来看，frp会一直添加新功能，还是添加extra setting比较省事（frps已经添加了，frpc的要麻烦一点）

closed 2020-03-19 by nicepass