coolsnowwolf/lede #3880: 小娱C1开启 UPNP 后不定时重启

jdjingdian, 2020-03-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：小娱C1在开启UPnP 与 NAT-PMP 服务之后会经常不间断重启，miniUPNP版本是2.0.20170421-2，因为重启了我也看不到日志，同版本的竞斗云开启upnp没有什么问题，这个问题主要是在更新7621鸡血nat之后开始发现的（mt7621: TCP and UDP traffic is forwarded correctly by HWNAT (1% CPU usage))

我的小娱上装了xxr+、ddns、frpc和frps以及Zerotier，没有编译任何第三方的东西进去，回退版本之后开启upnp不会不定时重启，但是之前版本的hwnat开了效果不明显，就蛮难受的

### (2) 路由器型号和固件版本

A：小娱C1 OpenWrt R20.3.11 / LuCI Master (git-20.062.50551-a6189c5) 内核4.14

### (3) 详细日志

A：Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: set name of thread 5961 to PLUGINSD[ioping
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/system.idlejitter/max.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/system.idlejitter/average.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: thread created with task id 5960
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: set name of thread 5960 to STATSD_COLLECTO
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: STATSD collector thread started with taskid 5960
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: POLLFD: LISTENER: listening on 'udp:[::1]:8125'
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: POLLFD: LISTENER: listening on 'udp:127.0.0.1:8125'
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: POLLFD: LISTENER: listening on 'tcp:[::1]:8125'
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: POLLFD: LISTENER: listening on 'tcp:127.0.0.1:8125'
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_metrics/main.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_metrics/gauges.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: connected to '/usr/lib/netdata/plugins.d/ioping.plugin' running on pid 5964
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_metrics/counters.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_metrics/timers.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_metrics/meters.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_metrics/histograms.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: thread created with task id 5962
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: set name of thread 5962 to PLUGINSD[perf]
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: POLLFD: LISTENER: listening on 'tcp:0.0.0.0:19999'
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: POLLFD: LISTENER: listening on 'tcp:[::]:19999'
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_metrics/sets.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/main.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/gauges.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/counters.db.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/timers.db.
Wed Mar 18 04:04:27 2020 daemon.err netdata[5791]: read failed: end of file
Wed Mar 18 04:04:27 2020 daemon.err netdata[5791]: '/usr/lib/netdata/plugins.d/ioping.plugin' (pid 5964) disconnected after 0 successful data collections (ENDs).
Wed Mar 18 04:04:27 2020 daemon.err netdata[5791]: child pid 5964 exited with code 127.
Wed Mar 18 04:04:27 2020 daemon.err netdata[5791]: '/usr/lib/netdata/plugins.d/ioping.plugin' (pid 5964) exited with error code 127 and haven't collected any data. Disabling it.
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: thread with task id 5961 finished
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: connected to '/usr/lib/netdata/plugins.d/perf.plugin' running on pid 5965
Wed Mar 18 04:04:27 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/meters.db.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/histograms.db.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_useful_metrics/sets.db.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_events/main.db.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: called DISABLE. Disabling it.
Wed Mar 18 04:04:28 2020 daemon.err netdata[5791]: '/usr/lib/netdata/plugins.d/perf.plugin' (pid 5965) disconnected after 0 successful data collections (ENDs).
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_events/gauges.db.
Wed Mar 18 04:04:28 2020 daemon.err netdata[5791]: child pid 5965 exited with code 1.
Wed Mar 18 04:04:28 2020 daemon.err netdata[5791]: '/usr/lib/netdata/plugins.d/perf.plugin' (pid 5965) exited with error code 1 and haven't collected any data. Disabling it.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_events/counters.db.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: thread with task id 5962 finished
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_events/timers.db.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_events/meters.db.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_events/histograms.db.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_events/sets.db.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_events/unknown.db.
Wed Mar 18 04:04:28 2020 daemon.info netdata[5791]: Initializing file /var/cache/netdata/netdata.statsd_events/errors.db.
[... 433 lines cut ...]
Wed Mar 18 22:32:48 2020 user.warn ddns-scripts[1869]: DYNV6: Updating IP at DDNS provider failed - starting retry 109/0
Wed Mar 18 22:37:09 2020 kern.err /usr/bin/ssr-redir[5556]: server recv: Connection reset by peer
Wed Mar 18 22:42:58 2020 user.warn ddns-scripts[1869]: DYNV6: Error in 'expand_ipv6()' - invalid IPv6 found: '2a01:4f9:c010:95b2a00:c380:c0de::53532a01:4f8:1c1c:4c96' expanded: '2a01:04f9:c010:095b2a00:c380:c0de:53532a01:04f8:1c1c:4c96'
Wed Mar 18 22:42:58 2020 user.warn ddns-scripts[1869]: DYNV6: Updating IP at DDNS provider failed - starting retry 110/0
Wed Mar 18 22:48:17 2020 kern.err /usr/bin/ssr-redir[5560]: server recv: Connection reset by peer
Wed Mar 18 22:53:08 2020 user.warn ddns-scripts[1869]: DYNV6: Error in 'expand_ipv6()' - invalid IPv6 found: '2a00:c380:c0de::53532a01:4f8:1c1c:4c962a01:4f9:c010:95b' expanded: '2a00:c380:c0de:53532a01:04f8:1c1c:4c962a01:04f9:c010:095b'
Wed Mar 18 22:53:08 2020 user.warn ddns-scripts[1869]: DYNV6: Updating IP at DDNS provider failed - starting retry 111/0
Wed Mar 18 22:56:07 2020 kern.err /usr/bin/ssr-redir[5560]: remote recv: Connection reset by peer
Wed Mar 18 22:57:23 2020 kern.err /usr/bin/ssr-redir[5556]: remote recv: Connection reset by peer
Wed Mar 18 22:57:51 2020 authpriv.info dropbear[8285]: Child connection from 2409:8a55:c9b:c800:11e0:95e5:908:7dae:61924
Wed Mar 18 22:57:58 2020 authpriv.notice dropbear[8285]: Password auth succeeded for 'root' from 2409:8a55:c9b:c800:11e0:95e5:908:7dae:61924
Wed Mar 18 22:58:21 2020 kern.err /usr/bin/ssr-redir[5558]: remote recv: Connection reset by peer
Wed Mar 18 22:59:24 2020 kern.err /usr/bin/ssr-redir[5558]: server recv: Connection reset by peer
Wed Mar 18 23:03:20 2020 kern.err /usr/bin/ssr-redir[5556]: remote recv: Connection reset by peer
Wed Mar 18 23:03:24 2020 user.warn ddns-scripts[1869]: DYNV6: Error in 'expand_ipv6()' - invalid IPv6 found: '2a01:4f8:1c1c:4c962a01:4f9:c010:95b2a00:c380:c0de::5353' expanded: '2a01:04f8:1c1c:4c962a01:04f9:c010:095b2a00:c380:c0de:5353'
Wed Mar 18 23:03:25 2020 user.warn ddns-scripts[1869]: DYNV6: Updating IP at DDNS provider failed - starting retry 112/0
Wed Mar 18 23:04:51 2020 kern.err /usr/bin/ssr-redir[5560]: remote recv: Connection reset by peer
[... cut ...]
Wed Mar 18 23:13:20 2020 kern.err /usr/bin/ssr-redir[5558]: remote recv: Operation timed out
Wed Mar 18 23:13:33 2020 user.warn ddns-scripts[1869]: DYNV6: Error in 'expand_ipv6()' - invalid IPv6 found: '2a00:c380:c0de::53532a01:4f9:c010:95b2a01:4f8:1c1c:4c96' expanded: '2a00:c380:c0de:53532a01:04f9:c010:095b2a01:04f8:1c1c:4c96'
Wed Mar 18 23:13:33 2020 user.warn ddns-scripts[1869]: DYNV6: Updating IP at DDNS provider failed - starting retry 113/0

以上是关闭upnp正常运行的日志，之前截取过一小段打开upnp后运行的日志，大概是这样的：
daemon.warn miniupnpd[3282]: SSDP packet sender 10.37.129.2:62000 not from a LAN, ignoring

jdjingdian, 2020-03-18:
补上正常运行时的内核日志：

[    0.000000] Linux version 4.14.172 (jarvis@jarvis-Parallels-Virtual-Platform) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r2106-13bc38d3)) #0 SMP Tue Mar 17 10:48:28 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is XiaoYu XY-C5
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 1c000000 @ 00000000 (usable)
[    0.000000]  memory: 04000000 @ 20000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat 805f43e0, node_mem_map 81003000
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from start_kernel+0x9c/0x4d8 with crng_init=0
[    0.000000] percpu: Embedded 14 pages/cpu s26832 r8192 d22320 u57344
[    0.000000] pcpu-alloc: s26832 r8192 d22320 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00076c87
[    0.000000] Readback ErrCtl register=00076c87
[    0.000000] Memory: 510960K/524288K available (4796K kernel code, 247K rwdata, 1052K rodata, 1252K init, 254K bss, 13328K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007822] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073988] pid_max: default: 32768 minimum: 301
[    0.078809] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.085318] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.094531] Hierarchical SRCU implementation.
[    0.099766] smp: Bringing up secondary CPUs ...
[    6.268649] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.268659] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.268670] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    6.268811] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.164531] Synchronize counters for CPU 1: done.
[    5.986230] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    5.986239] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    5.986247] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    5.986325] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.255719] Synchronize counters for CPU 2: done.
[    6.075529] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.075539] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.075547] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    6.075625] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.340895] Synchronize counters for CPU 3: done.
[    0.370753] smp: Brought up 1 node, 4 CPUs
[... 196 lines cut ...]
[   14.981326] kmodloader: done loading kernel modules from /etc/modules.d/*
[   18.722098] xt_FULLCONENAT: RFC3489 Full Cone NAT module
[   18.722098] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
[   22.068002] mtk_soc_eth 1e100000.ethernet: PPE started
[   22.084112] br-lan: port 1(eth0.1) entered blocking state
[   22.095186] br-lan: port 1(eth0.1) entered disabled state
[   22.106661] device eth0.1 entered promiscuous mode
[   22.116277] device eth0 entered promiscuous mode
[   22.129625] br-lan: port 1(eth0.1) entered blocking state
[   22.140454] br-lan: port 1(eth0.1) entered forwarding state
[   22.152227] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   23.056423] pppoe-wan: renamed from ppp0
[   23.104625] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[ 3453.226554] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[... cut ...]
[ 5556.434762] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[ 5559.027550] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[ 5567.014671] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[ 5571.494535] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[61980.926632] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[61984.608399] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[61992.602808] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[61996.810730] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up

bluehj777, 2020-03-19:
同样问题，已经观察3天了，每次都是不定时重启（大概10几个小时左右），用的小娱C3 。深刻的记得同样的MTK 7621芯片的新三D2 在刷的是今年2月份的固件时候没有问题。也是怀疑在更新了mt7621: TCP and UDP traffic is forwarded correctly by HWNAT 后引起的。

bluehj777, 2020-03-19:
题主怎么确定更新7621鸡血nat有改善呢？我这里感觉没更新这个前CPU占用也很低

jdjingdian, 2020-03-19:
我发现关了upnp也会重启，看来是hwnat的问题，我关掉hwnat再试试会不会重启

之前版本的hwnat，测速300Mbps路由器cpu就负载20%，新版本测速时占用就1%@bluhj777

麻烦L大看看hwnat模块是不是有点小问题，看起来不只我一个人会遇到不定时重启的问题@coolsnowwolf

bluehj777, 2020-03-19:
目前我这里不确定这个nat patch 是和miniupnp冲突的问题，我现在rm 掉了647-netfilter-nf_flow_table_hw-fix-incorrect-ethernet-ds.patch 和648-netfilter-nf_flow_table_hw-check-the-status-of-dst_n.patch  然后刚重新编译了一遍最新固件，安装到C3路由器上，看一下明天是否还会自动重启。这种重启通常发生在10多个小时以后。没啥规律，正常时固件的日志和内核日志都是非常正常的

bluehj777, 2020-03-19:
> 之前版本的hwnat，测速300Mbps路由器cpu就负载20%，新版本测速时占用就1%@bluhj777
哦，因为我是100M带宽，所以跑满时CPU基本没啥变化。

closed 2020-03-19 by coolsnowwolf

bluehj777, 2020-03-21:
@jdjingdian 建议测试观察下关闭 "DNS加速选项"和upnp设置里的 "启用 NAT-PMP 功能"，我在使用HTOP监视时发现，迅雷下载满速时dnscache这个进程过于活跃。目前的C3已经正常待机使用18小时，期间做过几次迅雷满负载（100M带宽）下载19G文件的下载。之前没关闭DNS加速时候，路由器在迅雷下载时，满速5，6分钟左右就会造成路由器死机的现象，基本上说只要P2P软件一满速下载有参与，路由器死机的概率就大增。
话说开启这个补丁真的不错，p2p软件满速下载CPU占用一般小于10%，普通多线程软件HTTP满速下载 CPU占用小于2%。

bluehj777, 2020-03-22:
经长时间测试，上面的方法还是不行，使用迅雷长时间满速下载，还是有死机现象，只是频率降低。看来这两个patch还是去掉稳妥