coolsnowwolf/lede #3881: luci-app-ssr打开后无法上网

lm317379829, 2020-03-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：luci-app-ssr打开后无法上网
      mwan3显示接口均在线，但是ssr开启后无法上网，关闭ssr恢复正常。

### (2) 路由器型号和固件版本

A：路由器型号：X86_64
      固件版本：OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7)
      commit：3d3f812

### (3) 详细日志

A：系统日志
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.532041] system 00:0c: [mem 0xfed19000-0xfed19fff] has been reserved
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.538756] system 00:0c: [mem 0xf8000000-0xfbffffff] has been reserved
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.545470] system 00:0c: [mem 0xfed20000-0xfed3ffff] has been reserved
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.552183] system 00:0c: [mem 0xfed90000-0xfed93fff] has been reserved
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.558900] system 00:0c: [mem 0xfed45000-0xfed8ffff] has been reserved
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.565610] system 00:0c: [mem 0xff000000-0xffffffff] has been reserved
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.572330] system 00:0c: [mem 0xfee00000-0xfeefffff] could not be reserved
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.579392] system 00:0c: [mem 0xdfa00000-0xdfa00fff] has been reserved
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.586113] system 00:0c: Plug and Play ACPI device, IDs PNP0c02 (active)
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.586311] system 00:0d: [mem 0x20000000-0x201fffff] has been reserved
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.593027] system 00:0d: [mem 0x40004000-0x40004fff] has been reserved
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.599745] system 00:0d: Plug and Play ACPI device, IDs PNP0c01 (active)
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.599776] pnp: PnP ACPI: found 14 devices
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.609518] clocksource: acpi_pm: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 2085701024 ns
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.618589] pci 0000:00:1c.0: PCI bridge to [bus 01]
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.623656] pci 0000:00:1c.0:   bridge window [io  0xe000-0xefff]
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.629856] pci 0000:00:1c.0:   bridge window [mem 0xf7d00000-0xf7dfffff]
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.636754] pci 0000:00:1c.1: PCI bridge to [bus 02]
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.641820] pci 0000:00:1c.1:   bridge window [io  0xd000-0xdfff]
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.648011] pci 0000:00:1c.1:   bridge window [mem 0xf7c00000-0xf7cfffff]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654916] pci_bus 0000:00: resource 4 [io  0x0000-0x0cf7 window]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654917] pci_bus 0000:00: resource 5 [io  0x0d00-0xffff window]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654919] pci_bus 0000:00: resource 6 [mem 0x000a0000-0x000bffff window]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654921] pci_bus 0000:00: resource 7 [mem 0x000d0000-0x000d3fff window]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654922] pci_bus 0000:00: resource 8 [mem 0x000d4000-0x000d7fff window]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654924] pci_bus 0000:00: resource 9 [mem 0x000d8000-0x000dbfff window]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654926] pci_bus 0000:00: resource 10 [mem 0x000dc000-0x000dffff window]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654927] pci_bus 0000:00: resource 11 [mem 0x000e0000-0x000e3fff window]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654929] pci_bus 0000:00: resource 12 [mem 0x000e4000-0x000e7fff window]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654930] pci_bus 0000:00: resource 13 [mem 0xdfa00000-0xfeafffff window]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654932] pci_bus 0000:01: resource 0 [io  0xe000-0xefff]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654933] pci_bus 0000:01: resource 1 [mem 0xf7d00000-0xf7dfffff]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654935] pci_bus 0000:02: resource 0 [io  0xd000-0xdfff]
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.654937] pci_bus 0000:02: resource 1 [mem 0xf7c00000-0xf7cfffff]
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.655066] NET: Registered protocol family 2
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.659696] tcp_listen_portaddr_hash hash table entries: 2048 (order: 3, 32768 bytes)
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.667720] TCP established hash table entries: 32768 (order: 6, 262144 bytes)
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.675151] TCP bind hash table entries: 32768 (order: 7, 524288 bytes)
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.681949] TCP: Hash tables configured (established 32768 bind 32768)
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.688609] UDP hash table entries: 2048 (order: 4, 65536 bytes)
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.694731] UDP-Lite hash table entries: 2048 (order: 4, 65536 bytes)
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.701318] NET: Registered protocol family 1
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.705786] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.740761] pci 0000:00:1a.0: quirk_usb_early_handoff+0x0/0x690 took 25569 usecs
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.780761] pci 0000:00:1d.0: quirk_usb_early_handoff+0x0/0x690 took 31680 usecs
Thu Mar 19 07:57:25 2020 kern.debug kernel: [    1.788384] PCI: CLS mismatch (64 != 32), using 64 bytes
Thu Mar 19 07:57:25 2020 kern.info kernel: [    1.788433] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[... 1307 lines cut ...]
[   12.790001] e1000e 0000:02:00.0 eth1: changing MTU from 1500 to 1492
[   12.984092] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   12.990108] 8021q: adding VLAN 0 to HW filter on device eth1
[   13.533509] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[   15.395200] e1000e: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   15.403286] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   15.428954] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[   15.440157] br-lan: port 1(eth0) entered blocking state
[   15.445567] br-lan: port 1(eth0) entered forwarding state
[   15.459141] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   15.467251] ax88179_178a 4-4:1.0 eth2: ax88179 - Link status is: 1
[   15.482952] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[   16.982833] IPv6: ADDRCONF(NETDEV_UP): docker0: link is not ready
[   20.760206] pppoe-wan: renamed from ppp0
[   20.801514] pppoe-wantv: renamed from ppp1

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Clackz, 2020-03-19:
最近对SSR改动比较多，用着不稳定……

ricksuzade-maker, 2020-03-19:
确实，CPU占用异常问题应该已经解决了，目前发现的主要问题就在SSR上，一开就没法上网，另外adbyby无法更新规则，点了没反应。

Clackz, 2020-03-19:
> 确实，CPU占用异常问题应该已经解决了，目前发现的主要问题就在SSR上，一开就没法上网，另外adbyby无法更新规则，点了没反应。

adbyby无法更新规则指的是adblock规则吗？可以先关闭adbyby后再尝试更新

lm317379829, 2020-03-19:
> 确实，CPU占用异常问题应该已经解决了，目前发现的主要问题就在SSR上，一开就没法上网，另外adbyby无法更新规则，点了没反应。

只要开了就没法上网吗，我还以为是mwan3冲突了呢，又编译了一个带passwall的固件用着。

closed 2020-03-19 by coolsnowwolf