coolsnowwolf/lede #3883: print string.join(msg) 格式错误编译失败

fryderykhuang, 2020-03-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
有一个完全相同的issue但是关闭了

## 2. 详细叙述

### (1) 具体问题

A：编译时提示   print string.join(msg) 格式错误，网上查了是python2的格式但用了python3去跑导致。
编译环境：ubuntu18.04，其他都按照说明里的进行了配置。

### (2) 路由器型号和固件版本

A： R619AC 128M

### (3) 详细日志

A：
`
Enabling zlinksys_recovery
sed -i "s/Installed-Time: .*/Installed-Time: 1584517387/" /home/fryderykhuang/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/opkg/status
rm -rf /home/fryderykhuang/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/tmp/*
rm -f /home/fryderykhuang/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/opkg/lists/*
rm -f /home/fryderykhuang/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/usr/lib/opkg/info/*.postinst*
rm -f /home/fryderykhuang/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/var/lock/*.lock
rm -rf /home/fryderykhuang/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/boot
rm -rf /home/fryderykhuang/lede/tmp/mklibs-progs /home/fryderykhuang/lede/tmp/mklibs-out
# first find all programs and add them to the mklibs list
find /home/fryderykhuang/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx -type f -perm /100 -exec file -r -N -F '' {} + | awk ' /executable.*dynamically/ { print $1 }' > /home/fryderykhuang/lede/tmp/mklibs-progs
# find all loadable objects that are not regular libraries and add them to the list as well
find /home/fryderykhuang/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx -type f -name \*.so\* -exec file -r -N -F '' {} + | awk ' /shared object/ { print $1 }' > /home/fryderykhuang/lede/tmp/mklibs-libs
mkdir -p /home/fryderykhuang/lede/tmp/mklibs-out
/home/fryderykhuang/lede/staging_dir/host/bin/mklibs -D -d /home/fryderykhuang/lede/tmp/mklibs-out --sysroot /home/fryderykhuang/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx `cat /home/fryderykhuang/lede/tmp/mklibs-libs | sed 's:/*[^/]\+/*$::' | uniq | sed 's:^/home/fryderykhuang/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx:-L :'` --ldlib /lib/ld-musl-armhf.so.1 --target arm-openwrt-linux-muslgnueabi `cat /home/fryderykhuang/lede/tmp/mklibs-progs /home/fryderykhuang/lede/tmp/mklibs-libs` 2>&1
  File "/home/fryderykhuang/lede/staging_dir/host/bin/mklibs", line 60
    **print string.join(msg)**
               ^
SyntaxError: invalid syntax
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/fryderykhuang/lede'
package/Makefile:108: recipe for target '/home/fryderykhuang/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install' failed
make[1]: *** [/home/fryderykhuang/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/fryderykhuang/lede'
Build failed - please re-run with -j1 to see the real error message
/home/fryderykhuang/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1

`

closed 2020-03-19 by coolsnowwolf

1-1-2, 2020-05-28:
我也碰到了同样的问题，报错完全一致，请问楼主解决了吗？

1-1-2, 2020-05-28:
提供一个笨办法，强制在运行相关脚本的时候使用python2。

我也不太清楚造成报错的原因，因为我在没有修改代码的情况下也的确编译成功过（成功率1/8）。
具体是那一条插件或者修改造成了报错，还有待进一步探索。
我在修改后成功通过github actions编译，没有报错。方法如下：

在编译前在【项目目录】下执行
```shell
 str_tmp="sed -i '1s/env python/python2/g' $(STAGING_DIR_HOST)/bin/mklibs
 sed -i "11a\\\t\\$str_tmp" ./include/rootfs.mk
```

原理：

出错部分的过程是通过`./include/rootfs.mk`调用的，在该文件中对报错的PY脚本`mklibs`第一行进行修改，强制使用python2执行。

如有纰漏/误解，望指正；有更好的方案，欢迎讨论。