coolsnowwolf/lede #3890: 请问shadowsocks + kcptun 为什么设置无效？（这个月编译版，内有图片）

geerda123, 2020-03-19:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我编译了最新版包含ss-libev kcptun固件，但是kcptun能运行，ping都能执行，就是无法科学上网。我看了类似issues和网上关于openwrt教程设置均无效，请问具体怎么设置？我用手机的ss可以使用kcptun，服务器和lede都是同版本kcptun。

### (2) 路由器型号和固件版本

A：DomyWifi DW33D
     OpenWrt R20.3.11  
      内核版本 | 4.9.214

### (3) 详细日志

A：
[image]
[image]
[image]
[image]

coolsnowwolf, 2020-03-19:
不在这个目录下。详细看 eSire 教程吧

lphgor, 2020-03-19:
加密方式 poly1305？

geerda123, 2020-03-19:
@coolsnowwolf 在这个目录下。找到原因了，1，我设置的加密少了poly1035；2，最重要的shadowsocks的端口要设置为12948，esir的教程告诉的。希望大佬能在kcptun那里添加这个提示

closed 2020-03-19 by geerda123