coolsnowwolf/lede #3893: 很奇怪，UBNT ERX频繁宕机

realoyrz, 2020-03-19:
路由器：UBNT ERX 固件版本为3.20最新编译的
情况：莫名其妙宕机，体现为ping不通，管理地址无法登录，无法上网，20秒左右自动恢复

我折腾了一天，初步确定可能跟SSR plus有关，特别是加载了100+订阅节点后宕机概率更加高，两小时内复现了4 次，怀疑可能是7621顶不住？？过热？？?

tmpaxx, 2020-03-19:
我也是ERX 我只用一个VPS 也会死机 用3月9号编译的版本就比较稳定

shenyz, 2020-03-20:
有加入网易的app么，有的话去除全部网易的，包括network里面的网易，再编译试试

jdjingdian, 2020-03-20:
您好，我的小娱C1也是7621芯片，在更新3.13版本（mt7621: TCP and UDP traffic is forwarded correctly by HWNAT (1% CPU usage））后不定时重启，现在关了hwnat正常运行，你们要不要看看是不是hwnat的问题？

Aes64X, 2020-03-20:
我的D1也发生了频繁宕机，我换成了备用的R3G，但在大约10分钟前，它也宕机了。。。。
情况表现为：网口等正常闪烁，WIFI突然间信号中断且无法再搜索到，使用网线连接后无法获取IP地址，手动指定IP后无法连接管理页面。断电重启后恢复，然后过不多久它会再次宕机。
由于无法登录管理端，无法看到日志记录。我现在使用备用的备用的竞斗云 - -!

coolsnowwolf, 2020-03-20:
你们这个问题应该是 Flow 更新后出现的， 7621 的交换机启用 HWNAT 后死机了

Aes64X, 2020-03-20:
感谢，将开始新的构建

realoyrz, 2020-03-20:
> 你们这个问题应该是 Flow 更新后出现的， 7621 的交换机启用 HWNAT 后死机了

大雕神速，又可以愉快地编译了

niceguy3-31, 2020-03-21:
> 你们这个问题应该是 Flow 更新后出现的， 7621 的交换机启用 HWNAT 后死机了

感谢，一直编译，一直歇逼，总算知道原因了

closed 2020-03-21 by coolsnowwolf