coolsnowwolf/lede #3894: 歌华链V0.4编译最新版固件无法启动

xjboss, 2020-03-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：启动不了

### (2) 路由器型号和固件版本

A：歌华链 V0.4

### (3) 详细日志

路由器日志：
Boot and Recovery Environment for Embedded Devices
Copyright (C) 2020 HackPascal <hackpascal@gmail.com>
Build date 2020-01-14 [git-b25d507]
Version 1.1 (r1280)

DRAM: 512MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: GeHua GHL-R-001
Clocks: CPU: 880MHz, DDR: 1066MHz, Bus: 293MHz, Ref: 40MHz
Flash: Macronix MX25L25635E (32MB) on mt7621-spi.0
rt2880-eth: MAC address from EEPROM is invalid, using default settings.
rt2880-eth: Using MAC address 00:0c:43:00:00:01
eth0: MediaTek MT7530 Gigabit switch

Network started on eth0, inet addr 192.168.1.1, netmask 255.255.255.0

Press any key to interrupt autoboot ... 0   

Trying to boot firmware from 0x00050000 in flash bank 0 ...
Reading data into memory ...
U-Boot firmware image header detected.
    Image Name:   MIPS OpenWrt Linux-4.14.172
    Data Size:    2021900 Bytes
    Load Address: 80001000
    Entry Point:  80001000
Uncompressing data (LZMA) ... done.
Flushing cache ... done.

Starting kernel at 0x80001000...

[    0.000000] Linux version 4.14.172 (build@UbuntuBuilder) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r2301-2b09def6)) #0 SMP Thu Mar 19 05:44:18 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is GeHua GHL-R-001
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 1c000000 @ 00000000 (usable)
[    0.000000]  memory: 04000000 @ 20000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[... 116 lines cut ...]
[    3.104363] 0x000000040000-0x000000050000 : "factory"
[    3.115422] 0x000000050000-0x000002000000 : "firmware"
[    3.149199] libphy: Fixed MDIO Bus: probed
[    3.225087] mtk_soc_eth 1e100000.ethernet: generated random MAC address 00:00:00:00:00:00
[    3.241539] libphy: mdio: probed
[    4.648908] mtk_soc_eth 1e100000.ethernet: loaded mt7530 driver
[    4.661400] mtk_soc_eth 1e100000.ethernet eth0: mediatek frame engine at 0xbe100000, irq 22
[    4.680460] NET: Registered protocol family 10
[    4.690966] Segment Routing with IPv6
[    4.698389] NET: Registered protocol family 17
[    4.707311] 8021q: 802.1Q VLAN Support v1.8
[    4.718161] hctosys: unable to open rtc device (rtc0)
[    4.729113] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
[    4.744025] Please append a correct "root=" boot option; here are the available partitions:
[    4.760655] 1f00             192 mtdblock0 
[    4.760662]  (driver?)
[    4.773671] 1f01              64 mtdblock1 
[    4.773676]  (driver?)
[    4.786672] 1f02              64 mtdblock2 
[    4.786677]  (driver?)
[    4.799673] 1f03           32448 mtdblock3 
[    4.799678]  (driver?)
[    4.812678] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    4.830794] Rebooting in 1 seconds..
看起来无法找到分区？？？不知道怎么解决

yinzhefuyun, 2020-03-20:
看来忘记打高通补丁了，firmware又无法解开了

AmadeusGhost, 2020-03-20:
你可以试下 #3906 中的gehua_ghl-r-001-e

ify358, 2020-03-20:
2000-3-20 编译了下，也是无限重启

AmadeusGhost, 2020-03-20:
> 2000-3-20 编译了下，也是无限重启

你可以测试下我的pr，最简单的改法如下：
```
--- a/target/linux/ramips/dts/mt7621_gehua_ghl-r-001.dts
+++ b/target/linux/ramips/dts/mt7621_gehua_ghl-r-001.dts
@@ -49,10 +49,9 @@
 	status = "okay";
 
 	flash@0 {
-		compatible = "mx25l25635f", "jedec,spi-nor";
+		compatible = "jedec,spi-nor";
 		reg = <0>;
 		spi-max-frequency = <25000000>;
-		m25p,fast-read;
 
 		partitions {
 			compatible = "fixed-partitions";
```
[image]

xjboss, 2020-03-20:
我试试

xjboss, 2020-03-20:
我这个貌似是MX25L25635F的闪存？？？？芯片上面写着

yinzhefuyun, 2020-03-20:
> 我这个貌似是MX25L25635F的闪存？？？？芯片上面写着

openwrt官方就没有这个spi驱动

AmadeusGhost, 2020-03-21:
> 我这个貌似是MX25L25635F的闪存？？？？芯片上面写着
> 但它竟然能正常的运作 反而F的不行

所以就是e的可以正常工作吗？

xjboss, 2020-03-21:
@AmadeusGhost 是的，但我用的你的PR可能有点问题，有线网连不上，我试试只加补丁。

AmadeusGhost, 2020-03-21:
> @AmadeusGhost 是的，但我用的你的PR可能有点问题，有线网连不上，我试试只加补丁。

那是我的锅2333，重命名了其它地方没改，既然e能用于f，那我就直接统一为e不拆分成2个了。

xjboss, 2020-03-21:
或许可能F芯片有好几种版本也说不定，我这个路由的闪存进BREED也说我是MX25L25635E。然后因为MWAN3有问题我试了官方的OPENWRT 19.07，发现也炸了。

yinzhefuyun, 2020-03-21:
> 或许可能F芯片有好几种版本也说不定，这个改了以后我用起来完全没问题。我这个路由的闪存进BREED也说我是MX25L25635E。然后因为MWAN3有问题我试了官方的OPENWRT 19.07，发现也炸了。

不是看能不能用，你还要测试在web页面点击重启能不能重启，这个才是关键

xjboss, 2020-03-22:
重启看起来是会出问题  直接卡住了 官方的和这个版本都会这样子

yinzhefuyun, 2020-03-22:
> 重启看起来是会出问题 直接卡住了 官方的和这个版本都会这样子
操，歌华链轮回了，下一个bug，是不是在web上传bin文件，显示不兼容？
只能等lean的高通spi驱动了

AmadeusGhost, 2020-03-22:
> 只能等lean的高通spi驱动了

你所说的是指`target/linux/ipq40xx/patches-4.14/303-spi-nor-enable-4B-opcodes-for-mx25l25635f.patch`吧，这个补丁ramips一样有的，而且强制指定为f对部分机子来说会无法启动

xjboss, 2020-03-22:
我按照给旧版本用的408-spi-nor-m25p-shutdown-for-mx25l25635f.patch C+V乐乐了一个补丁 好像可以用了

[408-spi-nor-m25p-shutdown-for-mx25l25635f.zip](https://github.com/coolsnowwolf/lede/files/4364907/408-spi-nor-m25p-shutdown-for-mx25l25635f.zip)

AmadeusGhost, 2020-03-22:
> 就发个文件不提交 pull request了，我感觉兼容性可能没那么吼。

看了下x-wrt也是这么处理的，那兼容性应该还好
https://github.com/x-wrt/x-wrt/commit/e0f7d20dda7e55e663d0c24887b5217cf03e0597

ify358, 2020-03-22:
看到2020-03-21的更新重新编译，今天刷了正常。

yinzhefuyun, 2020-03-22:
> 看到2020-03-21的更新重新编译，今天刷了正常。

你的正常是正常开机吗？在web下可以点击重启来重启吗？能在web下升级bin文件吗？

AmadeusGhost, 2020-03-22:
> 就发个文件不提交 pull request了，我感觉兼容性可能没那么吼。

这个跟`target/linux/ramips/patches-4.14/999-fix-m25p-shutdown.patch`差不多吧，就是第一个patch的位置不一样（

xjboss, 2020-03-22:
哦对我可能搞混了原版和这个版本 这个版本其实并不会重启死机 原版没这个补丁

ify358, 2020-03-22:
> > 看到2020-03-21的更新重新编译，今天刷了正常。
> 
> 你的正常是正常开机吗？在web下可以点击重启来重启吗？能在web下升级bin文件吗？

breed 刷的sysupgrade   openwrt-ramips-mt7621-gehua_ghl-r-001-squashfs-sysupgrade.bin

ify358, 2020-03-22:
> > 看到2020-03-21的更新重新编译，今天刷了正常。
> 
> 你的正常是正常开机吗？在web下可以点击重启来重启吗？能在web下升级bin文件吗？

刷完自动重启不行。拔了插头再插可以了

caoli5288, 2020-03-23:
你个浓眉大眼的也淘了个矿机吗，我也淘了个233

closed 2020-05-24 by xjboss