coolsnowwolf/lede #3897: ssrp在启动之前会检查节点状态，选择状态为无法连接的节点（即使该节点可以使用）无法启动ssrp。

shikiroot, 2020-03-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

        发现一个逻辑上的问题，ssrp在启动之前会检查节点状态，选择状态为无法连接的节点（即使该节点可以使用）无法启动ssrp。
        问题所在：为了防止dns污染和泄漏，我使用dns2socks（5335）和dnsmasq（53）防污染和泄漏。dnsmasq监控本地dns请求并转发到5335，wan、lan口设置dns为127.0.0.1。此时问题就来了，现在大多数机场的节点地址都是域名，而ssrp检测节点是否可用时需要解析域名，但是所有的解析最终都要转到dns2socks（5335），但是ssrp和dns2socks都没有运行，解析自然失败，解析失败，则显示节点不可用，所以ssrp无法启动。为了进行dns解析，需要启动ssrp和dns2socks，为了启动ssrp就需要检测端口，而ssrp和dns2socks不启动就无法进行节点地址解析。陷入循环。暂时的解决问题是先将wan口dns改为公共dns地址，使ssrp将节点域名解析出来然后启动，在将wan口dns改回127.0.0.1。
        我不太清楚ssrp在启动时选择的节点是否必须能够ping通，如果不是的话，可否将ssrp设置为在启动时不进行节点状态检测？如果此解决方法可行但普遍适用性不大同时会带来其他问题，大雕大可不必理会我，我每次开机设置一下wan口dns就好了。

shikiroot, 2020-03-20:
发现一个逻辑上的问题，ssrp在启动之前会检查节点状态，选择状态为无法连接的节点（即使该节点可以使用）无法启动ssrp。
        问题所在：为了防止dns污染和泄漏，我使用dns2socks（5335）和dnsmasq（53）防污染和泄漏。dnsmasq监控本地dns请求并转发到5335，wan、lan口设置dns为127.0.0.1。此时问题就来了，现在大多数机场的节点地址都是域名，而ssrp检测节点是否可用时需要解析域名，但是所有的解析最终都要转到dns2socks（5335），但是ssrp和dns2socks都没有运行，解析自然失败，解析失败，则显示节点不可用，所以ssrp无法启动。
        为了进行dns解析，需要启动ssrp和dns2socks，为了启动ssrp就需要检测端口，而ssrp和dns2socks不启动就无法进行节点地址解析。陷入循环。
        暂时的解决方法是先将wan口dns改为公共dns地址，使ssrp将节点域名解析出来然后启动，再将wan口dns改回127.0.0.1。
        我不太清楚ssrp在启动时选择的节点是否必须能够ping通，如果不是的话，可否将ssrp设置为在启动时不进行节点状态检测？如果此解决方法可行但普遍适用性不大同时会带来其他问题，大雕大可不必理会我，我每次开机设置一下wan口dns就好了。

coolsnowwolf, 2020-03-20:
……你多此一举了，ssr plus 本来就会自动用 pdnsd 或者 dns2socks 防污染

shikiroot, 2020-03-20:
> ……你多此一举了，ssr plus 本来就会自动用 pdnsd 或者 dns2socks 防污染

是的，但是仅仅开启pdnsd或者dns2socks的话运营商的dns服务器依然会参与解析。移动会拦截用户的解析转到自己的服务器上，其实我是不想让运营商的dns参与进来，有点强迫症😂

coolsnowwolf, 2020-03-20:
那就用Turbo acc 的dns 加速即可

closed 2020-03-20 by shikiroot