coolsnowwolf/lede #3900: 当tcp和游戏udp是同一个服务器的时候，能合并进程吗

yinzhefuyun, 2020-03-20:
当tcp和游戏udp是同一个服务器的时候，能合并进程吗
 "inbound": {
     "sniffing": {
       "enabled": true,
       "destOverride": [
         "http",
         "tls"
       ]
     },
     "port": "1234",
     "protocol": "dokodemo-door",
     "settings": {
       "network": "tcp",//两个json串只有这里有区别"network": "udp",
       "followRedirect": true
     }
   }

coolsnowwolf, 2020-03-20:
特意分开的

因为 tcp 是多进程 reuse port，但是这种方式对 udp 是增加延迟的，对游戏用户不友好

coolsnowwolf, 2020-03-20:
而且例如 Trojan，根本就官方不支持 UDP Relay，必须分开处理

yinzhefuyun, 2020-03-20:
> 特意分开的
> 
> 因为 tcp 是多进程 reuse port，但是这种方式对 udp 是增加延迟的，对游戏用户不友好

想问一下那v2的
"mux": {
       "enabled": true,
       "concurrency": 8
     },
这个应该也会产生延迟吧

coolsnowwolf, 2020-03-20:
因为 v2 本来就不适合 udp游戏，大部分都是其他用户，所以默认就用这个了

closed 2020-03-23 by coolsnowwolf