coolsnowwolf/lede #3907: ShadowSocksR Plus+ 没有效果

Cathgao, 2020-03-20:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

SSR+已经开启Trojan，状态页检查也通过，但所有下级设备都没有科学效果。
TurboACC中除了BBR外都开启了，请问是否与此有关？负载均衡和MWAN都没有安装

[image]
[image]

### (2) 路由器型号和固件版本

主机型号 | P&W R619AC 128M (CpuMark : 6054.184955 Scores)
架构 | ARMv7 Processor rev 5 (v7l) x 4
固件版本 | OpenWrt R20.3.11 / LuCI Master (git-20.071.17905-38b5cd7)
内核版本 | 4.14.172

### (3) 详细日志

2020-03-20 18:30:34 Welcome to trojan 1.14.1 , 4 Threads Started!
2020-03-20 18:30:04 Welcome to trojan 1.14.1 , 4 Threads Started!
2020-03-20 18:29:52 Welcome to trojan 1.14.1 , 4 Threads Started!

hcym, 2020-03-20:
不开乱七八糟的试试，哪有那么多事

Cathgao, 2020-03-20:
> 不开乱七八糟的试试，哪有那么多事

请问什么叫“乱七八糟”。加速全部关闭也没有任何区别

Cathgao, 2020-03-20:
进一步测试是只有GFW列表模式不工作，但dnsforwarder已经运行

hcym, 2020-03-20:
我没有那个DNS forward，只有一个ssr，都正常，今天下午的版本

dolphinpaopao, 2020-03-20:
检查一下dns转发设置，图上ssr使用的pdnsd转发，再加入了其他工具的话。各项设置得用点心思，毕竟复杂化了。

coolsnowwolf, 2020-03-20:
卸载 dnsforwarder 即可生效，尤其对于 gfwlist 模式

o0HalfLife0o, 2020-03-21:
没有安装dnsforwarder，gfw模式也不行

dolphinpaopao, 2020-03-21:
> 没有安装dnsforwarder，gfw模式也不行

最新版本正在机器上面编译，待会儿帮你测试一下。

o0HalfLife0o, 2020-03-21:
@dolphinpaopao 谢谢了，我就是用的默认的包，没装别的东西

coolsnowwolf, 2020-03-21:
如果你的客户端 DNS 没有指向 OP的 LAN IP ，那么很多模式都会失效

dolphinpaopao, 2020-03-21:
> @dolphinpaopao 谢谢了，我就是用的默认的包，没装别的东西

刷完最新拉取编译测试完毕，x86，gfw模式没有问题。按照作者所讲检查一下电脑和路由DNS设置。

o0HalfLife0o, 2020-03-21:
我是n1当旁路由，dhcp是主路由，但是网关、dns都是指向n1，所以不清楚是不是因为这个

closed 2020-03-22 by coolsnowwolf

o0HalfLife0o, 2020-10-20:
> > 我是n1当暂时由，dhcp是主路由，但是网关，dns都是指向n1，所以明显是不是因为这个
> 
> 你好，你最终有没找到原因
没有，换成直接用v2ray，搭配smartdns，和转发规则

WoodC2011, 2020-10-20:
> 没有，换成直接用v2ray，搭配smartdns，和转发规则
> 
> 

我试过openclash，能解决旁路由的大部分问题，比如我是cn上不去，就是速度有点慢

luckins, 2020-12-21:
我用的也是竞斗云，gfw默认GFW列表模式也是不行的