coolsnowwolf/lede #3909: x86系统安装好后 进web登录报下面这个错，求解。

LH1184793495, 2020-03-20:
/usr/lib/lua/luci/ccache.lua:52: error loading module 'luci.tools.ddns' from file '/usr/lib/lua/luci/tools/ddns.lua':
	/usr/lib/lua/luci/tools/ddns.lua:28: '=' expected near 'has_fetch'
stack traceback:
	[C]: in function 'l'
	/usr/lib/lua/luci/ccache.lua:52: in function </usr/lib/lua/luci/ccache.lua:46>
	[C]: in function 'require'
	/usr/lib/lua/luci/controller/ddns.lua:11: in main chunk
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:430: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:452: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:188: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

coolsnowwolf, 2020-03-21:
更新 feeds

LH1184793495, 2020-03-22:
> 更新 feeds

解决了 谢谢~   我重新pull了下

closed 2020-03-23 by coolsnowwolf