coolsnowwolf/lede #3910: 有没有方法在多拨的情况下实现upnp？

xhqpp, 2020-03-21:
实测多拨后upnp就失效了，有没有办法能同时使用的？

shenyz, 2020-03-21:
实测没有失效
[image]
[image]

xhqpp, 2020-03-21:
> 实测没有失效
> [image]
> [image]

我这里也有映射 但实测是不通的 你这个是通的吗？另外你能测一下v2模式通不通吗？

badcrazy, 2020-03-22:
我只知道之前是正常的，因为我半年多拨了，现在不知道，不过之前有个人说用所有lean编译的固件都不能开彗星的upnp，但是我测下来是正常的，大部分人正常，个别人确实不正常，所以应该不是源码问题，还是设备和网络环境的关系

closed 2020-03-26 by coolsnowwolf