coolsnowwolf/lede #3911: 广告屏蔽大师 Plus+似乎会自动停止运行

zy79937254, 2020-03-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

zy79937254, 2020-03-21:
[系统日志.txt](https://github.com/coolsnowwolf/lede/files/4362277/default.txt)
[内核日志.txt](https://github.com/coolsnowwolf/lede/files/4362278/default.txt)
路由器型号；ACRH17
固件版本；根据大雕的源码编译的OpenWrt R20.3.11 / LuCI Master (git-20.062.50551-a6189c5)
内核版本；4.19.106
运行时间 | 8d 16h 31m 44s
具体情况为，去广告大师似乎会经常性运行几天后自动停止，上一次我直接重启了路由器没有保存日志，这一次刚好保存下来了，因为前天突然发现不能打开恩山网页，登录进路由器后台发现去广告自动停止运行了，我重启后正常打开恩山网页

lx521603, 2020-03-21:
路由器型号；unraid 虚拟机运行
固件版本；N个固件版本。（Lienol，eSir, Lusty等编译的最新版本）（对，我虚拟机装了5个，一个不正常会试另外一个）
内核版本；4.19.108 AND 4.14.172

具体情况为 ：
    每天早上，大量国内DNS解析失败，nslookup能返回正确值，ping不通，网页打不开。
    手机显示WLAN没有连接到internet，但实际APP能联网，手机QQ里的课堂连不上，相册显示获取图片失败，但发送消息是正常的。ins里刷新不了。

     试用了几个固件，都是发现广告屏蔽大师 Plus+ 停止运行，√是勾了的，直接点下面的提交，上述的所有情况都消失，所有都恢复正常。

     怀疑是打勾了“每天凌晨6点更新规则并重启”，没有重启成功，因为情况都是出现在睡醒以后。

     用的是全局模式。

     没有日志，每次出问题都尝试了先重启，但重启后它也没自动运行。

d975xbx, 2020-03-21:
是啊，感谢楼主，你不说还真不知道。刚才才发现停止运行了。x64.

zy79937254, 2020-03-21:
> 是啊，感谢楼主，你不说还真不知道。刚才才发现停止运行了。x64。

我是前一段时间就有了 当时忘记保存日志，这几天复现了2次，

hplee0120, 2020-03-21:
你这个版本应该不是最新的，之前的确有这个问题，问题出现原因是大量用户6点同时访问那个规则服务器地址导致服务器炸了然后更新规则失败导致插件挂起一直在等待更新不自动运行。后面已经修复了，不过修复之后现在我发现的问题是那个adblock hosts自动更新失效了，因为我试了两三次早上六点后点了手动更新规则都能刷新出新规则。所以感觉那个自动更新没有生效。不过插件倒是能正常运行了。

lx521603, 2020-03-21:
> 你这个版本应该不是最新的，之前的确有这个问题，问题出现原因是大量用户6点同时访问那个规则服务器地址导致服务器炸了然后更新规则失败导致插件挂起一直在等待更新不自动运行。后面已经修复了，不过修复之后现在我发现的问题是那个adblock hosts自动更新失效了，因为我试了两三次早上六点后点了手动更新规则都能刷新出新规则。所以感觉那个自动更新没有生效。不过插件倒是能正常运行了。

我补充下情况，之前我一直用这个插件，从来没出现过这个问题，但我是adgurad home,adbyby plus一起用的，相互补充。前几天听说adbyby plus更新了，支持添加第三方规则。所以我特地刷了最新的固件，看了下确实是支持第三方规则的版本，所以应该是最新的吧。后来发现效果不错就吧ADH停了，单独使用adbyby plus.

仔细想了下，刚开始也没问题，大概是前天，还是昨天，算上今天，总共是2天还是3天连续出现这个问题。

coolsnowwolf, 2020-03-21:
版本太老，更新吧

hplee0120, 2020-03-21:
> 版本太老，更新吧

大雕，最新的虽然修复了那个一直被挂起的问题，不过那个hosts规则自动更新失效了，我连续试了3天6点过后手动更新规则都能刷新出新规则，所以判断是6点的时候没有自动更新

coolsnowwolf, 2020-03-21:
是么，我看看

hplee0120, 2020-03-21:
> 是么，我看看

[image]
请看时间，6点05分手动刷新可以刷新出新规则，所以推测6点时自动更新没有正常运行。

d975xbx, 2020-03-22:
第三方URL地址分享一下。谢谢

hplee0120, 2020-03-22:
> 第三方URL地址分享一下。谢谢

https://github.com/neoFelhz/neohosts

closed 2020-03-23 by coolsnowwolf