coolsnowwolf/lede #3921: 编译arm架构失败

rilyuuj, 2020-03-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
armhf架构编译失败

Q：是否搜索了issue (使用 "x" 选择)
查了最近的issue，有类似的但是尝试后依旧失败。
查了google，很多都是都是说python3问题，本机系统为python2.7，并已按要求安装python3
尝试再menuconfig里取消nodejs的language，发现第一个默认安装无法取消。

## 2. 详细叙述
我想编译全智A40系列的工程板，bpi-m2u.架构是arm的
编译时，只更改了平台和架构，以及输出目标镜像tar.gz格式，但是全部都有Makefile 101和105错误
尝试make clean, dirclean, dstclean,都依旧问题。
我用的windows的WSL，系统内核为4.4的bionic
附上我所安装的依赖：
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler

### (1) 具体问题
主要错误和部分issue类似，node 编译失败

### (2) 路由器型号和固件版本

allwinnerA40的bpi-m2u

### (3) 详细日志

tools/v8_gypfiles/generate_bytecode_builtins_list.target.mk:13: recipe for target '/home/opwrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h' failed
make[5]: *** [/home/opwrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h] Error 1
Makefile:101: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/opwrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0'
Makefile:150: recipe for target '/home/opwrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/.built' failed
make[3]: *** [/home/opwrt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v12.15.0/.built] Error 2
make[3]: Leaving directory '/home/opwrt/openwrt/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#17.53#49.23#136.47
package/Makefile:111: recipe for target 'package/feeds/packages/node/compile' failed
make[2]: *** [package/feeds/packages/node/compile] Error 2
make[2]: Leaving directory '/home/opwrt/openwrt'
package/Makefile:107: recipe for target '/home/opwrt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/opwrt/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/opwrt/openwrt'
/home/opwrt/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

麻烦请问该如何处理和查找原因。

hcym, 2020-03-21:
armv8正常编译

closed 2020-03-22 by coolsnowwolf

rilyuuj, 2020-03-22:
> armv8正常编译

请问怎么找原因呢?我用vmware重新试了一次还是一样