coolsnowwolf/lede #3925: 最新的lede，局域网的手机网关是op，dns是2个不存在的局域网的ip，一样可以上网？

norikohxb, 2020-03-21:
我是在今天半夜发现用手机智慧星控制易控设备无法控制经常显示网络无法连接。开头怀疑是adbyby的问题，关闭adbyby，问题依然存在，然后关闭ssrplus就好了。（手机的dns和网关都是op）
然后就觉得有必要找出博联智慧星究竟访问哪个ip，于是把我手机的dns指向adguardhome专门一台服务器。要嗅探出智慧星访问的域名，但是发现在adguardhome上面一个dns请求信息也没有。
在重启了手机重启了adguardhome仍然无法获得手机的dns请求（但是手机一样可以上大陆网）于是杜撰2个不存在的局域网ip，把手机的dns分配成为2个不存在的局域网ip，发现手机一样很愉快的上网。我瞬间崩溃了，难道我的手机已经不需要dns就可以上网了？但是问题出在op上面，手机的网关是op，就可以不需要dns了吗？

norikohxb, 2020-03-21:
我觉得这种有悖伦常的现象是不可能存在的。把我的台式机的dns设置为局域网一台不存在的ip，就无法上网了，所以这个问题和三星s10的手机有关系。究竟是什么问题导致三星的s10， dns的2个随便设置不存在都可以上网呢？

norikohxb, 2020-03-21:
有没有用三星s10手机+lean的op的
发现一个太奇葩的现象了
三星s10的手机，只要网关是op（旁路由），你随便设置2个不存在的dns，都可以上网
完全没有任何问题。
如果把三星的手机的网关设置为ros（主路由），dns设置2个不存在的就无法上网
但是如果吧三星的手机网关设置为op旁路由，你的dns随便填写2组不存在的不影响上网

相同的方法用台式机测试是没有这个现象的。
这个是什么原理？

norikohxb, 2020-03-21:
测试华为平板m6没有三星s10这个能力

tmpaxx, 2020-03-21:
拔掉手机卡或者禁用蜂窝网络试试

norikohxb, 2020-03-21:
禁用了三星s10手机流量还是可以随便填写两个虚假s10上网

norikohxb, 2020-03-22:
[image]
测试了只有三星的手机和平板都可以这样，华为平板不行，电脑不行。感觉好像就是dns2个都没有正确，一样可以通过op上网。
请问这个是什么原理？
如果把网关切换为ros就不行上网了，dns错了就不能上网
如果在op关闭ssrplus一样也可以上国内网站，可见和ssr没有关系。

coolsnowwolf, 2020-03-22:
OP 会自动纠正重定向你错误的 IP 设置，除非你是爱国华

mjyhj, 2020-03-22:
当然可以上网，防火墙自定义规格里已经劫持，无论你填什么DNS，都是指向路由
iptables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53
iptables -t nat -A PREROUTING -p tcp --dport 53 -j REDIRECT --to-ports 53

norikohxb, 2020-03-22:
> OP 会自动纠正重定向你错误的 IP 设置，除非你是爱国华

> OP 会自动纠正重定向你错误的 IP 设置，除非你是爱国华

lean大谢谢你的百忙之中的回帖。但是现在困扰我的是，我其实本意是要把我的手机dns填写成为我的adguardhome的服务器，ip是192.168.16.251（真实存在）我也填写了，但是在adguardhome那边无法收到任何纪录。
[image]

norikohxb, 2020-03-22:
> OP 会自动纠正重定向你错误的 IP 设置，除非你是爱国华

三星s10手机，如果网关是ROS（192.168.16.252）那么dns错误无法上网，
把三星s10的手机的网关设置为旁路由192.168.16.248，dns故意填错2个一样可以上网。关闭了op的ssr一样可以上网。但是如果填写正确为内网的192.168.16.251（单独一台adguardhome）竟然在adguardhome上面找不到任何解析记录。（其他小度在家都可以）
昨天半夜特别把华为M6的平板（网关为op），，dns故意设置2个不存在无法访问网页。
昨天半夜也把我的台式机（网关为op，dns故意设错）无法打开网页
就明显三星s10可以，而且也只可以网关是op才可以，如果网关是ros一级路由，dns设错无法访问网页
[image]
）

norikohxb, 2020-03-22:
> OP 会自动纠正重定向你错误的 IP 设置，除非你是爱国华
我想要让手机s10的的域名请求纪录可以在adguardhome里面可以看到。问题是没有
[image]

closed 2020-03-23 by norikohxb

norikohxb, 2020-03-23:
感谢一下lean的贴心，可以自动绕过错误的dns，我之前反映这个问题，还有一个是自己搞错了，在设置ros的dhcp的时候，忘记给我的手机设置255.255.255.0了。所以导致都不需要dns就可以上网了。虽然我也不是很明白为何会如此，但是我知道我那样设置是错误的