coolsnowwolf/lede #3927: SSR Plus+开启后，谷歌、百度连通性检查均失败，网络->诊断，只能ping通IP，不能ping通域名

ichris007, 2020-03-21:
（1）具体问题
SSR Plus+开启后，谷歌、百度连通性检查均失败，网络->诊断，只能ping通IP，不能ping通域名
SSR Plus+关闭后，网络->诊断，ping域名和IP均没问题

 (2) 路由器型号和固件版本

A：群晖里用docker安装openwrt_r9:20.1.24-x86_64固件

(3) 详细日志

Sun Mar 22 02:48:10 2020 cron.info crond[14357]: crond (busybox 1.30.0) started, log level 5
Sun Mar 22 02:48:11 2020 user.err ssr-rules[14401]: Start failed!
Sun Mar 22 02:48:11 2020 daemon.info pdnsd[14547]: pdnsd-1.2.9b-par starting.
Sun Mar 22 02:48:12 2020 daemon.info dnsmasq[10977]: exiting on receipt of SIGTERM
Sun Mar 22 02:48:12 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Mar 22 02:48:12 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Mar 22 02:48:12 2020 daemon.crit dnsmasq[14650]: failed to create IPset control socket: Protocol not supported
Sun Mar 22 02:48:12 2020 daemon.crit dnsmasq[14650]: FAILED to start up
Sun Mar 22 02:48:17 2020 daemon.crit dnsmasq[14746]: failed to create IPset control socket: Protocol not supported
Sun Mar 22 02:48:17 2020 daemon.crit dnsmasq[14746]: FAILED to start up
Sun Mar 22 02:48:22 2020 daemon.crit dnsmasq[14803]: failed to create IPset control socket: Protocol not supported
Sun Mar 22 02:48:22 2020 daemon.crit dnsmasq[14803]: FAILED to start up
Sun Mar 22 02:48:27 2020 daemon.crit dnsmasq[14860]: failed to create IPset control socket: Protocol not supported
Sun Mar 22 02:48:27 2020 daemon.crit dnsmasq[14860]: FAILED to start up
Sun Mar 22 02:48:32 2020 daemon.crit dnsmasq[14899]: failed to create IPset control socket: Protocol not supported
Sun Mar 22 02:48:32 2020 daemon.crit dnsmasq[14899]: FAILED to start up
Sun Mar 22 02:48:37 2020 daemon.crit dnsmasq[14961]: failed to create IPset control socket: Protocol not supported
Sun Mar 22 02:48:37 2020 daemon.crit dnsmasq[14961]: FAILED to start up
Sun Mar 22 02:48:37 2020 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 1 seconds since last crash

coolsnowwolf, 2020-03-22:
群晖不具备 IPSET 内核模块，请编译群晖内核时加入

ichris007, 2020-03-22:
@coolsnowwolf 感谢大神。本人小白，不懂技术，只能等大神下次更新时考虑一下群晖用户了：）

hxlls, 2020-03-22:
docker 是没办法改变宿主内核的，你要求助请找群晖

ichris007, 2020-03-22:
@hxlls 明白了，谢谢！
我的是群晖218+（文件系统是ext4），还有别的方法可以安装Openwrt，并使用SSR Plus+或Passwall吗？

closed 2020-03-22 by coolsnowwolf

matthuo333, 2021-04-02:
除了加载ipset模块外，还需要加入其它模块吧？都是需要哪些？可否列出一下