coolsnowwolf/lede #3928: ipv6的负载均衡 有没有可能整合进源码中

xhqpp, 2020-03-21:
https://www.right.com.cn/forum/thread-760177-1-1.html

lwz322, 2020-03-22:
楼主用了这个帖子的IPv6负载均衡，有发现什么问题吗？

我在没有PD的教育网用过，功能是没有问题的，因为没有PD，所以用的IPv6 NAT，故也不存在下面这个问题
> ipv6只能路由自身能用，不能在客户端使用且只有内外ipv6地址，除非直接拨号有方法的也请告知一下

具体在家用（有PD），可以多拨的环境下表现如何应该还需要测试反馈，如果你有这个测试条件的话，我可以跟进下

翻了下原帖的回复，征求作者对PR的意见，但是作者没回复；从设计上来说,，IPv6单独设计了开关，应该是不会影响IPv4的正常功能，最终是否合入还是要看管理的意见

coolsnowwolf, 2020-03-22:
我没有没有测试环境，自行测试

dongjuanyong, 2020-03-24:
上面lwz322已经说了，ipv6搞负载均衡除非你用ipv6 nat，否则根本实现不了
ipv6 pd的情况下，假设你有两条ipv6线路2408::（联通）和2409::（移动），在路由正常通告（ra）或分配（dhcpv6）v6地址的时候，客户端会得到2408::和2409::两个v6的地址，虽然网关只有一个
这里和v4不一样的地方就来了，客户端用v6访问的时候必须在发包中表明源地址，你用2408::的源地址，到了路由自然给你走联通的route，2409::自然走移动。也就是说路由控制不了你的路线，只能靠客户端，所以你在路由器上搞不了ipv6的负载均衡
你要是用nat6，就不一样了，管你几条路线进来，我路由只给下面特定的地址，发包到我这我在改你的源地址，这正是ipv4时期的做法。
你要是问为什么不用nat6，那我只能问你为什么用v6不想用v4呢？

aiden-5154, 2020-03-30:
可不可以这样实现，只让vwan1有ipv6，然后剩下的vwan全部关闭ipv6

dongjuanyong, 2020-03-30:
这倒是简单，network这么配就行
```
config interface 'wan'
        option ifname 'eth1'
        option proto 'pppoe'
        option username '*************'
        option password '*************'
        option ipv6 '1'
        option metric '20'

config interface 'wan_6'
        option proto 'dhcpv6'
        option ifname '@wan'
        option reqaddress 'try'
        option reqprefix 'auto'
```
所有的接口添加`option ipv6 '1'`让ipv6手动配置
需要用v6的接口添加虚拟接口（上面wan_6），物理接口@原生接口即可

closed 2020-07-04 by coolsnowwolf