coolsnowwolf/lede #3932: openwrt作为旁路由丢包问题

liuqi0270, 2020-03-22:
反馈bug/问题模板，提建议请删除

## 1.openwrt作为旁路由丢包的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

情况简介：我的平台是exsi平台上ROS+OP的模式。OP作为旁路由，主路由为ROS（10.10.138.253）同时是DHCPserver，OP用SSRplus+富强上网。软路由的网口都通过ROS桥接，只有一个虚拟交换机给OP做LAN口。

问题描述：MTR百度发现丢包率很高，如果绕开OP（10.10.138.254），网关设成ROS（10.10.138.253）丢包率为0

请问这是固件的问题，还是配置那里有问题呢？

使用OP的mtr报告
HOST: MacBookPro                  Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- 10.10.138.254             96.0%    50    3.3   2.7   2.2   3.3   0.8
  2.|-- 10.10.138.253              0.0%    50    2.0   3.1   1.6   9.4   1.3
  3.|-- 221.219.96.1               0.0%    50    4.1   9.2   3.3  76.0  14.1
  4.|-- 125.34.174.73              0.0%    50    5.1   9.1   4.3  26.1   5.4
  5.|-- 123.126.8.178              0.0%    50    8.7  12.2   4.3  78.0  17.4
  6.|-- 123.125.248.94             0.0%    50    5.2  14.8   5.0 102.5  23.0
  7.|-- ???                       100.0    50    0.0   0.0   0.0   0.0   0.0
  8.|-- ???                       100.0    50    0.0   0.0   0.0   0.0   0.0
  9.|-- 61.135.169.125             2.0%    50    6.6   6.9   4.7  17.0   2.3

未使用OP的mtr报告
HOST: MacBookPro                  Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- 10.10.138.253              0.0%    50    2.0   4.0   1.4  19.9   3.3
  2.|-- 221.219.96.1               0.0%    50    4.2   9.5   3.2 123.8  19.3
  3.|-- 125.34.175.205             0.0%    50    9.9  10.0   5.6  29.7   4.9
  4.|-- 124.65.58.62               0.0%    50    9.0  36.1   6.1 205.6  55.9
  5.|-- 123.125.248.122            0.0%    50    7.8  11.7   5.8  80.2  12.9
  6.|-- ???                       100.0    50    0.0   0.0   0.0   0.0   0.0
  7.|-- 61.135.169.121             0.0%    50    5.6   7.1   5.2  14.7   2.0

### (2) 路由器型号和固件版本

A：固件版本：OpenWrt R20.1.24 / LuCI Master (git-20.032.31649-0d31472)
B：内核版本：4.19.101

### (3) 详细日志

A：

closed 2020-03-22 by coolsnowwolf

liuqi0270, 2020-03-22:
关闭了是什么情况？

EnnawYang, 2020-03-22:
> 关闭了是什么情况？

你猜

kidd31, 2020-03-24:
我好像也碰到丢包问题了，请问有什么解决方案吗

liuqi0270, 2020-03-24:
> > 关闭了是什么情况？
> 
> 
> 
> 你猜

猜不到啊

liuqi0270, 2020-03-24:
> 我好像也碰到丢包问题了，请问有什么解决方案吗

那你也是做的旁路由？

kidd31, 2020-04-06:
> > 我好像也碰到丢包问题了，请问有什么解决方案吗
> 
> 那你也是做的旁路由？

对，我在nas里虚拟了给旁路由

Yrp, 2020-07-14:
这个问题解决了吗

liuqi0270, 2020-07-14:
没有解决。没啥影响就没管了