coolsnowwolf/lede #3936: 树莓派pi 1 B型， trojan+youtube速度被限制在百兆

bigguyong, 2020-03-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：树莓派pi 1 B型，百兆网口； trojan+youtube速度被限制在百兆，约12MB/s; 一个月前的版本测速可以达到150MB/s，忘了具体时间点了，试了之前的几个版本速度都回不去
机场速度是OK的，macOS客户端可以达到150MB/s，但是走OP就达不到

### (2) 路由器型号和固件版本

A：树莓派pi 1 B型 作为旁路由，百兆网口，固定IP192.168.0.2；主路由是千兆，设置lan 网关和DNS为旁路由IP 192.168.0.2

### (3) 详细日志

A：
[image]

EnnawYang, 2020-03-22:
你自己都说了是百兆网口了

hxlls, 2020-03-22:
百兆网线不是限制在百兆你还想多少？

coolsnowwolf, 2020-03-22:
.. 你还真以为是 “旁”路由啊

ElectronicElephant, 2020-03-22:
YouTube 不是测速软件…

EnnawYang, 2020-03-22:
可以关闭了，这个不是issue

bigguyong, 2020-03-22:
> 可以关闭了，这个不是issue

怎么说？

bigguyong, 2020-03-22:
> YouTube 不是测速软件…

我重点说的是trojan两种方式的对比，速度可能差异到达10倍了

agutools, 2020-03-22:
> > 可以关闭了，这个不是issue
> 
> 怎么说？

硬件上限，百兆网络理论最大速度12.5M/s，Pi2B+都还是由USB2.0 转RJ45的，还早的版本就……

hcym, 2020-03-22:
我的n1和电脑肉测差距有一点点，不影响4K，太大差距应该有原因，不过这个trojan根本不吃性能，硬件配置不是主要原因吧

bigguyong, 2020-03-22:
> > > 可以关闭了，这个不是issue
> > 
> > 
> > 怎么说？
> 
> 硬件上限，百兆网络理论最大速度12.5M/s，Pi2B+都还是由USB2.0 转RJ45的，还早的版本就……

b

bigguyong, 2020-03-22:
> 我的n1和电脑肉测差距有一点点，不影响4K，太大差距应该有原因，不过这个trojan根本不吃性能，硬件配置不是主要原因吧

是的，我也觉得不是硬件的问题；因为从响应看，op配置的trojan性能更好

ElectronicElephant, 2020-03-22:
> 
> 
> > YouTube 不是测速软件…
> 
> 我重点说的是trojan两种方式的对比，速度可能差异到达10倍了

但你的这个“可能”并不成立啊——YouTube又不是测速软件。

bigguyong, 2020-03-22:
> > > YouTube 不是测速软件…
> > 
> > 
> > 我重点说的是trojan两种方式的对比，速度可能差异到达10倍了
> 
> 但你的这个“可能”并不成立啊——YouTube又不是测速软件。
> 
> 那些说百兆网卡的人才曲解了你的意思

不怪别人，一开始自己搞错了单位，以为是被限制在百兆
可以从两种客户端表现看待这个问，ssr+ 和 macOS的trojan

EnnawYang, 2020-03-22:
自己再去好好学学网速单位吧，`百兆网口`的理论最高速度为`12.5M/s` ，对于百兆网卡来说这个速度已经到极限了，难不成你还想跑成千兆网速`125M/s`？ 不管你的主路由是不是千兆，只要流量经过旁路由`Pi1B`的`百兆网卡`，那你的网速就不可能超过`12.5M/s`，如果流量同时进出则网速`对半分`

bigguyong, 2020-03-22:
> 自己再去好好学学网速单位吧，`百兆网口`的理论最高速度为`12.5M/s` ，对于百兆网卡来说这个速度已经到极限了，难不成你还想跑成千兆网速`125M/s`？ 不管你的主路由是不是千兆，只要流量经过旁路由`Pi1B`的`百兆网卡`，那你的网速就不可能超过`12.5M/s`，如果流量同时进出则网速`对半分`

请看图，单位是小b

ElectronicElephant, 2020-03-22:
好歹换个正经测速工具吧…比如 fast.com

Wilson-coder, 2020-03-23:
楼主一开始的表达不准确导致了误解。我说说我的使用体验：

我最新编译的OP，20.3.19版本，Trojan客户端确实效率不高，甚至还不如自带的V2Ray，同一个VPS，用路由器设置Trojan节点，GFW模式看YT；和路由器停用代理，直接在手机上运行Igniter，YT的速度和缓冲提升了起码一倍。

另外为了对比，在Vultr硅谷同时搭建了两个VPS，一个部署Trojan，另一个部署V2Ray，路由器这边在两个节点之间简单切换，打开YT上这个时代广场的直播链接（https://www.youtube.com/watch?v=mRe-514tGMg），会发现V2Ray节点能更快建立稳定连接，缓冲速度也更快，同时在过程中连接速度也明显比Trojan节点的速度更高。这个差别在白天没有被电信QoS的时候对比还不太明显，到了晚上，V2Ray节点基本上还能稳定在720P，Trojan节点连480P都卡到不行。所以现在我又回到V2Ray了。

按说，Trojan是要比V2Ray快一些才合理吧？希望Lean能抽时间再检查一下问题在哪里。

bigguyong, 2020-03-23:
> 楼主一开始的表达不准确导致了误解。我说说我的使用体验：
> 
> 
> 
> 我最新编译的OP，20.3.19版本，Trojan客户端确实效率不高，甚至还不如自带的V2Ray，同一个VPS，用路由器设置Trojan节点，GFW模式看YT有点慢；如果路由器停用代理，直接在手机上运行Igniter客户端，YT的速度和缓冲提升明显，粗略估计起码一倍。
> 
> 
> 
> 另外为了对比，在Vultr硅谷同时搭建了两个VPS，一个部署Trojan，另一个部署V2Ray，路由器这边在两个节点之间简单切换，打开YT上这个时代广场的直播链接
> 
> 
> 
> https://www.youtube.com/watch?v=mRe-514tGMg
> 
> 
> 
> 会发现V2Ray节点能更快建立稳定连接，缓冲速度也更快，同时在过程中连接速度也明显比Trojan节点的速度更高。这个差别在白天没有被电信QoS的时候对比还不太明显，到了晚上，V2Ray节点基本上还能稳定在720P，Trojan节点连480P都卡到不行。所以现在我又回到V2Ray了。
> 
> 
> 
> 按说，Trojan是要比V2Ray快一些才合理吧？希望Lean能抽时间再检查一下问题在哪里。

这是我的问题，哈哈

Wilson-coder, 2020-03-23:
> 这是我的问题，哈哈

Sorry, 我觉得问题有关联性，就在这里一并提出来了。

closed 2020-03-23 by coolsnowwolf