coolsnowwolf/lede #3937: 是否应该放弃对 8M ROM + 64M RAM 的支持？

coolsnowwolf, 2020-03-22:
2020 年了。目前的插件体积随着功能的增加，体积在增大。

OpenWrt 要预留 Overlay 空间，所以如果是 8M ROM 最大的安全体积不能超过 6.75MB。

同时，Golang 之类的大体积和占用大内存，导致 64MB 连开启的机会可能都没有。

这个限制也制约了很多 IPK 好的功能和想法

是否应该正式放弃对 这些型号的支持呢？

jackyzhangyc, 2020-03-22:
支持放弃！

rvgty, 2020-03-22:
放弃

hnotcold, 2020-03-22:
感觉可以放弃了，都0202年了，少抽两包烟，少喝两瓶酒不就能换个机器了嘛？铁汁们说我说的对吗？觉得我对的在公屏扣个1让我看到好嘛？

adamw92, 2020-03-22:
放弃一部分老古董吧…

guochan2019, 2020-03-22:
支持放弃

qianxu2001, 2020-03-22:
放弃

gpeditmsc1986, 2020-03-22:
支持放弃

AmadeusGhost, 2020-03-22:
没法投票，那就用表情吧，支持放弃点赞，不支持点踩

mjyhj, 2020-03-22:
说支持，好像是支持放弃的意思
说放弃，就肯定就是放弃的意思
说放弃支持，那肯定是放弃的意思
说支持放弃，那也是放弃的意思

所以无论大家说支持还是放弃，都是放弃的意思

hcym, 2020-03-22:
支持放弃，硬路由做AP蛮好

ldy8182, 2020-03-22:
可以放弃了

ja5onwu, 2020-03-22:
支持放弃!

hxlls, 2020-03-22:
都什么年代了，还8M的flash ，该加钱了

hardy-one, 2020-03-22:
放弃 8+64的千兆路由很少见吧，现在还用百兆的是该考虑换了

xayane, 2020-03-22:
支持放弃

ElectronicElephant, 2020-03-22:
支持放弃

Going2333, 2020-03-22:
可以放弃了，软路由还是买得起的。K2 K2P之类的当AP就好了

yoseg, 2020-03-22:
支持放弃

willan007, 2020-03-22:
支持放弃

dolphinpaopao, 2020-03-22:
要不暂时保留一个旧的分支来支持。给一点过渡时间。

xjboss, 2020-03-22:
8M RAM + 64G ROM 我感觉吧装原版OPENWRT就行了吧 并不需要插件支持 反正也带不动

mingxiaoyu, 2020-03-22:
支持

xingpo, 2020-03-22:
我觉得可以放弃，现在的固件功能越来越多，8M+64M的刷原版openwrt就可以了！

rageq, 2020-03-22:
放弃吧

getopenwrt, 2020-03-22:
早点放弃吧，不然大雕你也没法放开手脚干，古董就该有古董的觉悟，早点吃灰。

badcrazy, 2020-03-22:
支持，8+64的性能跟不上了，总有说再见的一天

agutools, 2020-03-22:
用过相对“宽松”的硬件运行OpenWrt，不会想“倒退/将就”的了😅

zhengpaiboy, 2020-03-22:
留一个历史版本，然后正式放弃支持。后续有人玩，可以用旧版本编译。

Bnosora, 2020-03-22:
同意，不能让旧设备限制发展。

eskimokk, 2020-03-22:
建议保留一下分支支持老的机型，如果L大有精力的情况下，仅修复bugs，如果没有精力就算了。

wongwenj, 2020-03-22:
放弃！

nnssz, 2020-03-22:
支持大雕！

jinsm, 2020-03-22:
支持放弃

zcgmt, 2020-03-22:
建议保留一下分支支持老的机型，如果L大有精力的情况下，仅修复bugs，如果没有精力就算了。

hhmaomao, 2020-03-22:
闲鱼几十块钱买到的设备随便吊打8+64的老古董，技术的创新创新总要以淘汰老旧设备作为代价。如果你是x86平台当我没说，d525坚挺这么多年，到现在依然牛逼，上个无线网卡一样可以起飞兼职无线功能

Promix953, 2020-03-22:
直接放弃，也没必要留个修bug的分支了，谁还用老古董刷阿

cooguyan, 2020-03-23:
这个配置也太古董了

Mcxunyun666, 2020-03-23:
16MB + 64 MB 如何看待

jwf8280031, 2020-03-23:
放弃

mcucore, 2020-03-23:
虽然很反人类，但是我觉得8M 64M可以促进代码优化，某些IPK可以不支持，但是底层基础必须保持高效低资源需求。如果真这样放弃，我觉得变相放弃了X86和ARM之外的所有路由。

dolphinpaopao, 2020-03-23:
> 虽然很反人类，但是我觉得8M 64M可以促进代码优化，某些IPK可以不支持，但是底层基础必须保持高效低资源需求。如果真这样放弃，我觉得变相放弃了X86和ARM之外的所有路由。

这样的话不如上官方原版了。

mcucore, 2020-03-23:
> > 虽然很反人类，但是我觉得8M 64M可以促进代码优化，某些IPK可以不支持，但是底层基础必须保持高效低资源需求。如果真这样放弃，我觉得变相放弃了X86和ARM之外的所有路由。
> 
> 这样的话不如上官方原版了。

你没考虑过骨牌效应，每个代码都放开性能约束，优先考虑方便和冗余，最终代码效率会变啥样。
就好比最近的更新，X86上CPU 30%+，
放不放弃低端配置不是重点，而是整个项目的初心变了。

coolsnowwolf, 2020-03-23:
和效率没关系，纯粹是 ROM 和 RAM 不够了

Go 重写不如直接用 Clash 了

dolphinpaopao, 2020-03-23:
> > > 虽然很反人类，但是我觉得8M 64M可以促进代码优化，某些IPK可以不支持，但是底层基础必须保持高效低资源需求。如果真这样放弃，我觉得变相放弃了X86和ARM之外的所有路由。
> > 
> > 
> > 这样的话不如上官方原版了。
> 
> 你没考虑过骨牌效应，每个代码都放开性能约束，优先考虑方便和冗余，最终代码效率会变啥样。
> 就好比最近的更新，X86上CPU 30%+，
> 放不放弃低端配置不是重点，而是整个项目的初心变了。我还是比较喜欢如ASM OPENSSL那种精神。这毕竟是个嵌入式设备。
> 
> 另外，我是支持大雕用GO重写SSR PLUS+的。

徒增工作量啊，随着功能进化，体积变化也很正常。要是作为极简项目那体积限制暂时没什么问题，但老有增加这样增加那样的需求，作者也疲于应对。另外占用问题，我最近全新编译了五个不同版本都没有遇到。x86物理机，默认编译。

KevinZjYang, 2020-03-23:
支持放弃。

mcucore, 2020-03-23:
> 和效率没关系，纯粹是 ROM 和 RAM 不够了
> 
> Go 重写不如直接用 Clash 了

如果性能还是优先选项，当然支持抛弃包袱，
SSR这边，看大雕提到GO, 还以为大雕准备用GO重写ALL IN ONE。

letmekillthemall, 2020-03-23:
发一个release，以后有人想玩也能捡起来，就此不再支持8M，如何

KevinZjYang, 2020-03-23:
> 发一个release，以后有人想玩也能捡起来，就此不再支持8M，如何

这个解决方法挺好！

madcat0513, 2020-03-23:
最好留个分支不再维护，可以有个过渡

Mcxunyun666, 2020-03-23:
或者对cpu芯片支持，硬改256m ram ，usb启动

Clackz, 2020-03-23:
会自己刷第三方ROM的人……也应该不会死认一个没有可玩性的路由器吧……

BraveSail, 2020-03-23:
能刷固件的我觉得没几个会没个大内存的机器吧

star-o0O, 2020-03-23:
应该放弃，玩路由的大多数应该都16M以上了，是时候放弃了

bearbazooka, 2020-03-23:
滋瓷，肯定滋瓷

Mcxunyun666, 2020-03-23:
16+64如何活

&mdash;

keivenliao, 2020-03-24:
有投票的话，估计是一边倒吧，所以我支持！
扔掉老古董吧！

swxk521, 2020-03-24:
以X86软路由为主好吧

BOBINIUNIU, 2020-03-24:
我个人觉得除了x86，其他都放弃都行。现在一个j1900才多少钱，再配几个AP无缝漫游难道不香吗，价格和一台高档硬路由也差不多。硬路由刷openwrt因为无线驱动的问题很容易各种毛病，硬路由还是做AP最好。

Clackz, 2020-03-24:
> 我个人觉得除了x86，其他都放弃都行。现在一个j1900才多少钱，再配几个AP无缝漫游难道不香吗，价格和一台高档硬路由也差不多。硬路由刷openwrt因为无线驱动的问题很容易各种毛病，硬路由还是做AP最好。

你这个就太极端了吧

marcosteam, 2020-03-24:
建议专门为老机型开个存档就行了

[60 more replies]

closed 2020-04-19 by coolsnowwolf