coolsnowwolf/lede #3945: 计划任务重启无法保存

vobasf, 2020-03-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
计划任务重启无法保存
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在web界面添加计划任务后，重启消失
在/etc/crontabs/root文件中，写入计划任务后，重启也消失

### (2) 路由器型号和固件版本

A：J1900 X86 OpenWrt R20.3.11

### (3) 详细日志

A：暂无

aptx17, 2020-03-22:
先回车一行，再添加命令。

vobasf, 2020-03-22:
> 先回车一行，再添加命令。

我就是另起一行添加命令的，刚才试了空一行，还是重启失效。

aptx17, 2020-03-22:
> > 先回车一行，再添加命令。
> 
> 我就是另起一行添加命令的，刚才试了空一行，还是重启失效。

[image]
不能啊，下边再加个回车试试，要不就直接改crontab -e。

jyy5206, 2020-03-23:
我也遇到过此问题。
后来尝试从文头输入，ok了。
从文末输入，也是重启后丢失改动。
试试看？

coolsnowwolf, 2020-03-23:
看看是不是用了关键字

mjyhj, 2020-03-23:
部分关键字会被对应LUCI 去掉

vobasf, 2020-03-23:
> 看看是不是用了关键字

应该是因为关键字adblock的原因，改掉后就没问题了，谢谢~

closed 2020-03-23 by vobasf

mayichuan2010, 2020-11-21:
能讲一下具体是怎么回事吗，关键字adblock是什么？

Cathgao, 2022-08-01:
reboot也算是关键字吗？我用计划任务做自动重启，重启完就没了

taoletsgo, 2022-12-30:
> reboot也算是关键字吗？我用计划任务做自动重启，重启完就没了

同问

taoletsgo, 2022-12-30:
> reboot也算是关键字吗？我用计划任务做自动重启，重启完就没了

问题解决了，去软件包里搜reboot删除别人编译的里面那个定时重启插件