coolsnowwolf/lede #3946: 升级到iptables 1.8.4后编译不过去了，，没有libnftnl

bluehj777, 2020-03-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
22日iptables升级到1.8.4，nftables模块的
错误日志提示它需要系统安装 libnftnl包，但14版本的
ubuntu 没有这个依赖可以用来安装
之前的1.8.2的iptables一直没有问题的

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

hxlls, 2020-03-22:
安装libnftnl-dev试试

bluehj777, 2020-03-22:
14版本里也没有这个，16以上版本才有 @hxlls

hzmax0416, 2020-03-22:
那是不是要重新安装16以上版本的Ubuntu来编译才行？

hxlls, 2020-03-22:
不需要
我就是14编译的，屁事都没

bluehj777, 2020-03-22:
您的编译项目里选择了nftables了吗？

coolsnowwolf, 2020-03-22:
安装libnftnl-dev试试

bluehj777, 2020-03-22:
这几天一直在编译都没有问题，今天pull了以后出现的。最后干脆全部删除了编译目录重新克隆了，重新开始。还是这个问题。
google到类似的情况
https://forum.openwrt.org/t/iptables-1-8-3-ar71xx-mikrotik-compiler-error/40244
https://github.com/openwrt/openwrt/pull/2179

hxlls, 2020-03-22:
我编译的nfs qos 就需要这个吧
[image]
[image]

closed 2020-03-23 by coolsnowwolf