coolsnowwolf/lede #3947: shadowsocks plus+绕过大陆模式下的问题

Humsan, 2020-03-22:
## 1.绕过大陆IP模式下，强制不走代理的域名设置了无效

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在ssr plus+中设置为绕过大陆IP模式时，访问控制走强制不走代理的域名设置保存应用后，还是走了代理。
如： ip.sb 该域名用来获取宽带的公网IP，设置了绕过大陆IP模式后，在访问控制中加入该域名保存应用后，获取到的ip是节点服务器的ip。希望可以修复下。

### (2) 路由器型号和固件版本

A：2020.03.22日clone到Ubuntu中自编译的版本

### (3) 详细日志

A：

coolsnowwolf, 2020-03-22:
[image]
[image]

Humsan, 2020-03-22:
@coolsnowwolf 在命令行下使用curl ip.sb试下

Humsan, 2020-03-22:
@coolsnowwolf 正常web访问是可以拿到正确的公网IP, 不过在命令行下面使用curl ip.sb就不行

Humsan, 2020-03-23:
运行环境:

- 基于Ubuntu 18.04全局代理环境下编译的固件，编译版本为2020.03.22时最新代码
- 运行在Ubuntu 18.04下的virtual box，使用桥接网络连接
- 安装后，修改对应静态ip地址，局域网内设备修改网关地址为openwrt的静态ip地址（在同一网段）
- 开启SSR PLUS+服务后，可以正常访问Google YouTube等网站
- 使用绕过大陆IP模式，web访问http://ip.sb可获得正确的公网ip地址，但在Ubuntu的terminal中使用 curl ip.sb 获得的是节点服务器的ip地址

coolsnowwolf, 2020-03-23:
不可能，请检查自己的设置

closed 2020-03-23 by coolsnowwolf