coolsnowwolf/lede #3949: 202003221版dnsmasq.ssr的问题

xxsunyxx, 2020-03-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
启用了ssr里的广告过滤后，过一段时间就会报错，这次是30个小时左右，上次没计时，就重刷了
### (1) 具体问题

A：不能分配ip，路由卡死

### (2) 路由器型号和固件版本

A：acrh17，前天编译的版本

### (3) 详细日志

A：
Mon Mar 23 06:35:42 2020 daemon.info procd: Instance dnsmasq::cfg01411c pid 2940 not stopped on SIGTERM, sending SIGKILL instead
Mon Mar 23 06:35:43 2020 daemon.crit dnsmasq[4073]: bad option at line 33741 of /tmp/dnsmasq.ssr/ad.conf
Mon Mar 23 06:35:43 2020 daemon.crit dnsmasq[4073]: FAILED to start up
Mon Mar 23 06:35:48 2020 daemon.crit dnsmasq[4092]: bad option at line 33741 of /tmp/dnsmasq.ssr/ad.conf
Mon Mar 23 06:35:48 2020 daemon.crit dnsmasq[4092]: FAILED to start up
Mon Mar 23 06:35:54 2020 daemon.crit dnsmasq[4166]: bad option at line 33741 of /tmp/dnsmasq.ssr/ad.conf
Mon Mar 23 06:35:54 2020 daemon.crit dnsmasq[4166]: FAILED to start up
Mon Mar 23 06:35:56 2020 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0.2)
Mon Mar 23 06:35:57 2020 cron.info crond[4294]: crond (busybox 1.30.0) started, log level 8
Mon Mar 23 06:35:57 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Mar 23 06:35:57 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Mar 23 06:35:58 2020 daemon.crit dnsmasq[4357]: bad option at line 33741 of /tmp/dnsmasq.ssr/ad.conf
Mon Mar 23 06:35:58 2020 daemon.crit dnsmasq[4357]: FAILED to start up
Mon Mar 23 06:35:58 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Mar 23 06:35:58 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Mar 23 06:35:59 2020 daemon.info hostapd: wlan0: STA xx IEEE 802.11: authenticated
Mon Mar 23 06:35:59 2020 daemon.info hostapd: wlan0: STA xx IEEE 802.11: associated (aid 1)
Mon Mar 23 06:35:59 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTEDxx
Mon Mar 23 06:35:59 2020 daemon.info hostapd: wlan0: STA xx RADIUS: starting accounting session AE695CC8AEDE16BC
Mon Mar 23 06:35:59 2020 daemon.info hostapd: wlan0: STA xx WPA: pairwise key handshake completed (RSN)
Mon Mar 23 06:35:59 2020 daemon.crit dnsmasq[4475]: bad option at line 33741 of /tmp/dnsmasq.ssr/ad.conf
Mon Mar 23 06:35:59 2020 daemon.crit dnsmasq[4475]: FAILED to start up
Mon Mar 23 06:36:05 2020 daemon.crit dnsmasq[4588]: bad option at line 33741 of /tmp/dnsmasq.ssr/ad.conf
Mon Mar 23 06:36:05 2020 daemon.crit dnsmasq[4588]: FAILED to start up
Mon Mar 23 06:36:05 2020 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 1 seconds since last crash

xxsunyxx, 2020-03-23:
看起来像ad.conf没了

hcym, 2020-03-23:
木有问题，只用ssr的内置广告功能，没有编译adbyby，非常稳定撒

coolsnowwolf, 2020-03-23:
bad option at line 33741 of /tmp/dnsmasq.ssr/ad.conf

订阅的去广告规则的  33741 行有问题，导致启动失败

closed 2020-03-23 by coolsnowwolf

xxsunyxx, 2020-03-23:
提示确实是33741行有问题，但运行了30个小时之后出问题就有点奇怪了。ad.conf是编译自带的，没更改过，为什么会出问题，这是个问题。

hcym, 2020-03-23:
这个麻烦，大雕的内置不知道有没有test功能，我是用原版op固件的dnsmasq测试过替换大雕源码的ad.conf，哪个准确的说