coolsnowwolf/lede #3954: 上个星期的lean大的ssr让博联智能家居的app打开经常超时

norikohxb, 2020-03-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
没有
## 2. 详细叙述
设备lede的op作为旁路由，主路由是ros，手机智慧星app打开以后，看开关的状态，经常会转圈圈设置是连接超时现在基本上都已经无法通过app控制开关，关闭op里面的ssrplus，则可以正常用app控制开关，把op里面的ssrplus设置为 gfw模式，避开大陆模式都一样无法通过智慧星app遥控开关。确凿问题来自ssrplus，然后还尝试吧手机的dns设置为厦门电信isp的dns，也是相同问题存在。结论是ssrplus导致了智慧星app对于开关的操作问题。（大约2个星期前的版本是没有这个问题的）
因为我是几乎每天来更新lean大的lede编译一次。所以最近如果要用手手机app控制开关都需要用4G的电信流量。
[image]
[image]
[image]
[image]
[image]

hxlls, 2020-03-23:
我用最新的，直接OP主路由，不存在这个问题，包括其他一些协议都正常

norikohxb, 2020-03-23:
> 我用最新的，直接OP主路由，不存在这个问题，包括其他一些协议都正常

你也是博联的智慧星吗，？

[image]
这种设备吗？

hxlls, 2020-03-23:
博联的设备协议都一样的，所以不存在什么型号设备不同，只有不同品牌的，或者特定走某些天猫京东的那种不一样

norikohxb, 2020-03-23:
> 博联的设备协议都一样的，所以不存在什么型号设备不同，只有不同品牌的，或者特定走某些天猫京东的那种不一样

我如果启用ssrplus就这样。app无法对开关操作，也看不到安防管家s1的信息。都是空白。关闭ssrplus则可以，如果你的没有问题，应该就是我采用了旁路由的关系。你说这个旁路由要不要开upnp？
我编译的时候是没有upnp编译进去的。没有打钩的。ssrplus就编译了一个trojan和v2ray，其他也都没有编译。
[image]
[image]

hxlls, 2020-03-23:
大概率是你的什么2级路由问题，也就是你说的“旁”路由

norikohxb, 2020-03-23:
> 大概率是你的什么2级路由问题，也就是你说的“旁”路由
应该这个也有关系对的。因为作为旁路由我编译的时候是没有启用upnp，想说手机平板通过op路由到ros，都是路由关系并不是采用伪装ip方式，都是直挺挺链接，而且都是同属于255.255.255.0的网段。应该不需要upnp吧，之前我也发帖咨询过。然后现在表面现象就是我如果关闭ssrplus，整个智慧星很通畅，如果开启ssrplus，智慧星的管家，开关全部都不可以控制。
下面是我编译的选项。你帮我看看

[image]

norikohxb, 2020-03-23:
> 大概率是你的什么2级路由问题，也就是你说的“旁”路由

不过大约2个星期前，一样是旁路由，我的博联智慧星app是可以用内网对开关进行操作。
就是最近这两个星期才怪怪的。

hxlls, 2020-03-23:
软件这些按照道理是没问题的最近更改的网络规则有点多，并且你是2级路由，所以设置1级到2级也相对更麻烦，可能要自己修改对应的设置。当然这个只是个人看法，具体落实，只能自己去研究了，我没用2级，并且我的是正常的，所以我认为大概率上是因为2级路由引起的故障，并不是固件自身的问题

norikohxb, 2020-03-23:
> 软件这些按照道理是没问题的最近更改的网络规则有点多，并且你是2级路由，所以设置1级到2级也相对更麻烦，可能要自己修改对应的设置。当然这个只是个人看法，具体落实，只能自己去研究了，我没用2级，并且我的是正常的，所以我认为大概率上是因为2级路由引起的故障，并不是固件自身的问题

我知道你也是和我在探讨，我想要表达的是大约一个月之前的lede是可以的，然后最近的这两个星期不行。然后关闭ssrplus就可以，所以根源应该还是在于ssrplus。但是我还特别去分析了博联的所有的域名（利用一台adguardhome去分析手机app访问的域名）博联的域名里面解析的ip都是中国ip，我甚至把这些中国ip都加入到ssr的wanip白名单和不走代理的域名，还是无效。
说明应该还是和传输协议之间的问题。
这个该死的小公司，几个设备还动用了各种各样很难记住的域名。

mbd.baidu.com
10002access.ibroadlink.com
10002backup.broadlink.com.cn
10002main.broadlink.com.cn
10002tcp.broadlink.com.cn
10004access.ibroadlink.com
10004backup.broadlink.com.cn
10004main.broadlink.com.cn
10004main.ibroadlink.com
10004tcp.broadlink.com.cn
10015access.ibroadlink.com
10018access.ibroadlink.com
10018backup.broadlink.com.cn
10018main.broadlink.com.cn
10018tcp.broadlink.com.cn
10024access.ibroadlink.com
10024backup.broadlink.com.cn
10024main.broadlink.com.cn
10024tcp.broadlink.com.cn
10026access.ibroadlink.com
10119access.ibroadlink.com
10119backup.broadlink.com.cn
10119main.broadlink.com.cn
10119tcp.broadlink.com.cn
10141access.ibroadlink.com
10141backup.broadlink.com.cn
10141main.broadlink.com.cn
10141tcp.broadlink.com.cn
149access.ibroadlink.com
149backup.broadlink.com.cn
149main.broadlink.com.cn
149tcp.broadlink.com.cn
30014access.ibroadlink.com
30014backup.broadlink.com.cn
30014main.broadlink.com.cn
30014main.ibroadlink.com
30014tcp.broadlink.com.cn
30083access.ibroadlink.com
32007access.ibroadlink.com
32007backup.broadlink.com.cn
32007main.broadlink.com.cn
32007tcp.broadlink.com.cn
32017access.ibroadlink.com
account.ibroadlink.com
app-service-chn-467a8f05.ibroadlink.com
backup.broadlink.com.cn
broadlink.com.cn
broadlink.tmall.com
cloud.broadlink.com.cn
clouddb.broadlink.com.cn
d0f94faa04c63d9b7b0b034dcf895656bizappmanage.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5appfront.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5auth.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5bizaccount.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5bizappdata.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5bizappmanage.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5bizihcv0.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5bizpd.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5cn-clouddb.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5cn-ifttt.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5devinfo.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5fwversions.ibroadlink.com
d71c3dda3b31d66c9033e9b622e746b5pushproxy.ibroadlink.com
dnakit.ibroadlink.com
e9a19a0e0d099eb8288d878687b4883aauth.ibroadlink.com
ec3bizappmanage.ibroadlink.com
ibroadlink.com
ihcv0.ibroadlink.com
s1data.ibroadlink.com
spminidata.ibroadlink.com
upgrade.ibroadlink.com
www.broadlink.com.cn

101.124.6.180
101.132.113.211
101.132.116.34
112.124.35.104
112.124.42.42
114.215.253.64
114.55.179.112
115.29.202.183
121.40.130.34
121.40.132.212
121.40.139.228
121.40.221.50
121.40.93.169
139.217.206.178
139.219.239.183
203.119.169.90
42.159.227.130
42.159.237.10
47.100.214.116
47.103.46.229
47.110.77.209
47.111.118.175
47.111.130.99
47.99.200.70
52.130.67.116

norikohxb, 2020-03-23:
> 软件这些按照道理是没问题的最近更改的网络规则有点多，并且你是2级路由，所以设置1级到2级也相对更麻烦，可能要自己修改对应的设置。当然这个只是个人看法，具体落实，只能自己去研究了，我没用2级，并且我的是正常的，所以我认为大概率上是因为2级路由引起的故障，并不是固件自身的问题

找到问题了，实在是无法接受手机要控制博联还要断开wifi，我找到了根源所在，要解决这个问题，
游戏模式udp中继服务器必须停用，如果勾选了，会导致无法在内网控制。

请问lean大能否在将来解决这个问题吗？

norikohxb, 2020-03-23:
> 软件这些按照道理是没问题的最近更改的网络规则有点多，并且你是2级路由，所以设置1级到2级也相对更麻烦，可能要自己修改对应的设置。当然这个只是个人看法，具体落实，只能自己去研究了，我没用2级，并且我的是正常的，所以我认为大概率上是因为2级路由引起的故障，并不是固件自身的问题

问题出在我开启了游戏udp代理，麻烦你看一下你的一级路由也就是这个op是否开启了，游戏udp代理，如果没有开启能否开启看看，开启了以后就无法远程控制了

norikohxb, 2020-03-23:
> 软件这些按照道理是没问题的最近更改的网络规则有点多，并且你是2级路由，所以设置1级到2级也相对更麻烦，可能要自己修改对应的设置。当然这个只是个人看法，具体落实，只能自己去研究了，我没用2级，并且我的是正常的，所以我认为大概率上是因为2级路由引起的故障，并不是固件自身的问题

能否帮我做一下实验，游戏模式UDP中继服务器，开启和关闭看看对比一下博联智慧星的区别。你是一级路由，我二级路由已经有了答案并且找到问题了。你的一级路由能否测试。

coolsnowwolf, 2020-03-23:
能不能用tcpdump抓包看看连接到什么udp端口和地址

norikohxb, 2020-03-23:
> 能不能用tcpdump抓包看看连接到什么udp端口和地址
我用ros查询的，这个有没有参考意义？
[image]

norikohxb, 2020-03-23:
请问lean大这个有参考意义吗？我现在op是关闭udp，因为我知道他最终要通过主路由出去。所以我在ros上的connenct看的。
[image]

norikohxb, 2020-03-23:
会不会是如此问题？
[image]

norikohxb, 2020-03-23:
通过这个方式也是无法解决。
[image]

norikohxb, 2020-03-23:
lean大我提供的这个ros的udp截图可以吗？如果不行我就去研究你说的那个软件。我记得大约一个月以前的固件是不会有这个问题的。虽然我不打游戏但是那个udp几乎都是开的。估计是这两个星期。我现在先关闭upd先用。其实我也不是每天都在用手机控制博联设备，偶尔要用手机控制，平时都是和红外设备门磁进行联动。。

norikohxb, 2020-03-24:
今天我有刷了最新的lede，好像这个问题依旧，然后我又依旧关闭udp代理。

Promix953, 2020-03-24:
[image]
看看有没有图中的这条
防火墙丢弃了UDP80端口任意端口，任意IP的所有的数据包
从你的截图能看出控制连接可能是192.168.16.*和192.168.8.*的udp 80端口连接

norikohxb, 2020-03-24:
> [image]
> 看看有没有图中的这条
> 防火墙丢弃了UDP80端口任意端口，任意IP的所有的数据包
> 从你的截图能看出控制连接可能是192.168.16.*和192.168.8.*的udp 80端口连接

我这个是旁路由，其实并没有用到NAT，我截图给你看看，下午lean大又更新了，我又编译了，还是一样不可以打开upd。
[image]

Promix953, 2020-03-24:
不是N不NAT的问题，这条规则在Mangle表的PREROUTING链上，和NAT毛关系没有
在 状态--防火墙--表RAW--链SS_SPEC_TPROXY里面找
刚看了源码试了下，SSR Plus+在开启UDP后会添加这条DROP，除了关游戏模式UDP，没有简单的解决方法，防火墙自定义规则也不行，因为SSRP+比防火墙后启动，SSRP+添加的规则优先

norikohxb, 2020-03-24:
> 不是N不NAT的问题，这条规则在Mangle表的PREROUTING链上，和NAT毛关系没有
> 在 状态--防火墙--表RAW--链SS_SPEC_TPROXY里面找
> 刚看了源码试了下，SSR Plus+在开启UDP后会添加这条DROP，除了关游戏模式UDP，没有简单的解决方法，防火墙自定义规则也不行，因为SSRP+比防火墙后启动，SSRP+添加的规则优先

您说的是这里吗？但是但是大约半个月之前的lede应该是可以让我智慧星app在局域网控制博联开关而且还正常的。不然我现在去试试看之前的版本
[image]

Promix953, 2020-03-24:
[https://github.com/coolsnowwolf/lede/commit/2b09def652959baddb7c80e950bc2046f4d21313#diff-3c85d4a85cf63696b2820c2a412ee3ba](url)
在5天前加的

可以在SSRP+打开游戏模式中继启动后，SSH连上去，执行iptables -t mangle -A PREROUTING -p udp --dport 80 -j ACCEPT试试有没有效果

norikohxb, 2020-03-24:
我编译的时候把ssh去掉了，就是lean大批评的小白还乱折腾。ssh平时感觉都没啥用，我晚一点有空编译一个进去执行。先要吧ssh加上去。

Promix953, 2020-03-24:
> 我编译的时候把ssh去掉了，就是lean大批评的小白还乱折腾。ssh平时感觉都没啥用，我晚一点有空编译一个进去执行。先要吧ssh加上去。

其实你git revert撤销掉这个commit就可以了

norikohxb, 2020-03-24:
不用ssh在虚拟机的界面应该也可以吧
[image]

Promix953, 2020-03-24:
可以，一样的
可以在SSRP+打开游戏模式中继启动后，SSH连上去，执行
`iptables -t mangle -A PREROUTING -p udp --dport 80 -j ACCEPT`
试试有没有效果

norikohxb, 2020-03-24:
加了运行了还是不行
[image]

norikohxb, 2020-03-24:
> 可以，一样的
> 可以在SSRP+打开游戏模式中继启动后，SSH连上去，执行
> `iptables -t mangle -A PREROUTING -p udp --dport 80 -j ACCEPT`
> 试试有没有效果

在界面运行了你给我的命令，没效果。还是一样智慧星无法控制设备。

Promix953, 2020-03-24:
表Mangle截图

norikohxb, 2020-03-24:
mangle截图给你，谢谢你的热心指导
[image]

norikohxb, 2020-03-24:
我自己多执行了一条也无效。
iptables -t mangle -A PREROUTING -p udp --dport 1999 -j ACCEPT
[image]

norikohxb, 2020-03-24:
能顺便帮我看看系统日志里面这个错误指的是什么吗？

Promix953, 2020-03-24:
> 我自己多执行了一条也无效。
> iptables -t mangle -A PREROUTING -p udp --dport 1999 -j ACCEPT
> [image]

问题是出在游戏模式UDP中继，但具体处在哪里，还要排查，我这没这些设备，没条件

Promix953, 2020-03-24:
在防火墙--自定义规则 重启下防火墙，SSRP+开UDP，应用后，去终端运行下这个命令试试看：
`iptables -t mangle -D SS_SPEC_TPROXY -p udp --dport 80 -j DROP`

norikohxb, 2020-03-24:
按照1,2,3 的顺序还是不行，还是进入智慧星的设备转圈圈，无法连接。
[image]
[image]
[image]

Promix953, 2020-03-24:
`iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 80 -j RETURN`
重启SSRP，改用这个命令试试

norikohxb, 2020-03-24:
> `iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 80 -j RETURN`
> 重启SSRP，改用这个命令试试

1:网络-》防火墙-》自定义规则-》重启防火墙
2：重新保存和应用ssrplus+udp
3：运行iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 80 -j RETURN

norikohxb, 2020-03-24:
无效，还是无法智慧星控制设备

norikohxb, 2020-03-24:
3：运行iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 80 -j RETURN
4:   运行iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 1999 -j RETURN
这两个命令都运行就ok了。

Promix953, 2020-03-24:
> 3：运行iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 80 -j RETURN
> 4: 运行iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 1999 -j RETURN
> 这两个命令都运行就ok了。

。。。两条命令都是改第二条规则，其实只有1999那条有用，第一条不用的

norikohxb, 2020-03-24:
我可以再试试看只有1999的这条然后给你反馈，我的疑问来了
[image]

norikohxb, 2020-03-24:
> iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 1999 -j RETURN

iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 1999 -j RETURN
经过实践只用一条命令不行
必须80和1999都运行
ssrplus把内网的upd全部狙击了。 无论是直接连接博联设备的192.168.8.x 还是连接服务器的
统统咔嚓。

Promix953, 2020-03-24:
> > iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 1999 -j RETURN
> 
> iptables -t mangle -R SS_SPEC_TPROXY 2 -p udp --dport 1999 -j RETURN
> 经过实践只用一条命令不行
> 必须80和1999都运行
> ssrplus把内网的upd全部狙击了。 无论是直接连接博联设备的192.168.8.x 还是连接服务器的
> 统统咔嚓。

按理来说运行两条和只运行第二条效果是一样的，怪了

norikohxb, 2020-03-24:
我实践过了，专门重启了，只加1999，无效，，之前你第一次让我测试只加80也是无效。
我的192.168.8.x的博联设备是通过ros进行数据交换（上级路由器）192.168.16.x的手机+智慧星app，没有和192.168.8.x的设备在同一个网段，所以80端口的也是给OP进行block掉了，也就是SSR其实不管是通往国外还是国内的UDP全部都block了。

然后请教你一个问题在lean大没有更新处理之前，我能否把你告诉我的这两个命令加入到防火墙的自定义规则里面。

问题2：是不是其他人的如果局域网有udp也会给block，除非在同一个网段。

norikohxb, 2020-03-24:
你给我的2个命令好像有时效性。大约过了几分钟又不行了，然后我又必须运行这两条命令，又客户了。很奇怪的说。是不是SSR开通了自动切换工鞥？

norikohxb, 2020-03-24:
又不行了，又是转圈圈，你给我的2个命令好像就一分钟内有效。过了一分钟依旧转圈圈，无法连接，然后还必须再次输入这两个命令才可以用

Promix953, 2020-03-24:
> 又不行了，又是转圈圈，你给我的2个命令好像就一分钟内有效。过了一分钟依旧转圈圈，无法连接，然后还必须再次输入这两个命令才可以用

你这问题要改SSRP的
/usr/bin/ssr-rules
这个文件第200行之前插入
`$ipt -A SS_SPEC_TPROXY -p udp -d 192.168.0.0/16 --dport 80 -j RETURN`
`$ipt -A SS_SPEC_TPROXY -p udp -d 192.168.0.0/16 --dport 1999 -j RETURN`
重启SSRP

Promix953, 2020-03-24:
出问题的关键点还是你配置两个内网导致必须在防火墙里面走一圈，这种修改没提PR的必要，以后每升级一次你都要自己改或者维护一份自己的fork

norikohxb, 2020-03-24:
不只是内网的问题，外网用1999的也给屏蔽了。对于和博联服务器的数据交换采用的是和外网47.111.130.99还有47.100.214.116两个ip的1999进行通讯，对于内网机器是用192.168.8.x和每个单一设备进行通讯。所以总的来说其实还是ssrplus对于所有的不管是否gfw，还是非gfw的统一都采取了管制。才导致的。，而且可能还存在刷新问题，我按照你的命令执行了大约一分钟就无效了。
等lean大以后更新版本吧，我哦现在关闭udp其实也无妨，根本就没有玩游戏。
但是感觉ssrplus不应该block掉所有的udp啊，如果是其他软件或者程序要跑外网的udp，也一样给ssrplus咔嚓掉了
没事了，我等lean大以后慢慢更新代码。

norikohxb, 2020-03-24:
其实我觉得这样必须要绕过udp才对，可惜没有
[image]

norikohxb, 2020-03-24:
> 出问题的关键点还是你配置两个内网导致必须在防火墙里面走一圈，这种修改没提PR的必要，以后每升级一次你都要自己改或者维护一份自己的fork
我加了这三条您帮我看看对不对。
[image]

norikohxb, 2020-03-24:
/usr/bin/ssr-rules
这个文件第200行之前插入
$ipt -A SS_SPEC_TPROXY -p udp -d 192.168.0.0/16 --dport 80 -j RETURN
$ipt -A SS_SPEC_TPROXY -p udp -d 47.100.214.116 --dport 1999 -j RETURN
$ipt -A SS_SPEC_TPROXY -p udp -d 47.111.130.99 --dport 1999 -j RETURN
重启SSRP

我用你的方案加入到这个文件里面。目前已经好了，好像也不会有时效问题。

为什么之前你让我运行的这个代码会有时效问题？

norikohxb, 2020-03-24:
lean大是不是可以这样理解，这个udp的开关打开就是全局的，已经不在乎什么gfw或者是绕过大陆ip，清一色，都做全局布局，对吗？

norikohxb, 2020-03-24:
> 博联的设备协议都一样的，所以不存在什么型号设备不同，只有不同品牌的，或者特定走某些天猫京东的那种不一样

你能否开一下udp代理看看，我感觉你虽然是一级路由，没有像我一样划分子网，但是感觉你那个应也是无法用手机app控制的。因为只要开了udp，就是全局udp。

norikohxb, 2020-03-24:
> 出问题的关键点还是你配置两个内网导致必须在防火墙里面走一圈，这种修改没提PR的必要，以后每升级一次你都要自己改或者维护一份自己的fork
请教你一个问题啊，我在编译lede的时候，dns2sock是精简掉的请问这个会不会影响我的trojan科学上网啊，看其他帖子总是出现dns2sock，我目前用lede的旁路由，用trojan和v2ray科学上网会不会也需要用到dns2sock？请赐教。

norikohxb, 2020-03-24:
再咨询一下，我突然想起来我专门下载的一台老爷机，上面用比特彗星和迅雷他们也用udp，也就是如果游戏模式UDP中继服务器开启了，下载的电脑也跟着udp全部通过trojan出去了？那么这个 需要代理的端口，我选择了 仅常用端口（不走p2p流量大代理）是不是也是无效了。因为udp是全局的。

Promix953, 2020-03-24:
1.UDP中继并没有全局啊，下面一堆的判定规则，刚去看了下国内IP段数据库，你这47.100和47.111全是阿里云段的，没更新到国内IP数据库里面，肯定要被代理，你这问题要背锅也是网络结构和国内IP数据库背，和固件本身没什么大关系
2.玩BT的话建议把老爷机IP加到不代理的LAN里面
3.精简掉v2ray和dns2socks有什么影响没实际试过，理论上应该不影响，我用的软路由，不差这点空间

norikohxb, 2020-03-25:
> Close这个issue吧，问题帮你解决了
但是我自己尝试吧47.100和47.111加入到ssr的不走代理的ip里面，为什么还会通过udp？
[image]

我自己的想法我按照你的手法去修改ssrplus，不也就是把这几个ip加入到不走代理里面吗。为什么这样加了不行必须用你的方式去修改

[7 more replies]

closed 2020-03-25 by norikohxb