coolsnowwolf/lede #3955: 最新xxr+无法连接自带中转的节点

White-Clouds, 2020-03-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
[image]
自带中转的节点测试ping会显示很多个，而且大多无法连接到节点。

### (2) 路由器型号和固件版本

A：ASUS RT-ACRH17   OpenWrt R20.3.19 / LuCI Master (git-20.071.17905-38b5cd7)

### (3) 详细日志

A：并没有什么特别的

hxlls, 2020-03-23:
[image]
没什么问题啊，我一直用中转，上面的是最新的源码编译的，没问题

White-Clouds, 2020-03-23:
> [image]
> 没什么问题啊，我一直用中转，上面的是最新的源码编译的，没问题

是在机场设置的中转，不是路由

White-Clouds, 2020-03-23:
> [image]
> 没什么问题啊，我一直用中转，上面的是最新的源码编译的，没问题

[image]
我试了两次固件都是这玩意儿。。。。

hxlls, 2020-03-23:
显示多少个延时都无所谓，正常使用就行了，不是么？测试延时只是代表当前的这个节点的连接响应速度而已，并不一定代表不能连接，OK这个才是主要能不能用

White-Clouds, 2020-03-23:
> 显示多少个延时都无所谓，正常使用就行了，不是么？测试延时只是代表当前的这个节点的连接响应速度而已，并不一定代表不能连接，OK这个才是主要能不能用

就是因为并不能连接，所有显示这个的都连接不上

hxlls, 2020-03-23:
我的没问题啊

coolsnowwolf, 2020-03-23:
把节点信息私信我看看

closed 2020-03-23 by coolsnowwolf