coolsnowwolf/lede #3961: 3.23上午11点左右刷了最新的版本，CPU占用似乎有点不正常

keivenliao, 2020-03-23:
[image]

忽然发现CPU占用高了不少，进去看了看，是以下两个进程，这个正常吗？

[image]

kl9800, 2020-03-23:
我删掉ddns就好了

keivenliao, 2020-03-23:
> 我删掉ddns就好了

我DDNS是必选项。
刚才重新编译了个版本，貌似可以了， 今晚再观察一下

coolsnowwolf, 2020-03-23:
tcp ping 的时候占用cpu 比普通ping要多

keivenliao, 2020-03-23:
> tcp ping 的时候占用cpu 比普通ping要多

感谢大佬回复，问题是为啥一直持续着？不会自己结束进程吗？

VNY0920, 2020-03-24:
我跟你情况类似，cpu常态使用15~20%。3865U
昨晚在Github云编译的版本。
其实只开了ssr plus+和 softherVPN 两个服务。

VNY0920, 2020-03-24:
太神奇了，我删掉在软件包里删掉ddns就好了。ddns服务本来就没开启的。

VNY0920, 2020-03-24:
太神奇了，我在软件包里删掉ddns就好了，cpu使用率掉到个位数。不过ddns服务本来就没开启的。

keivenliao, 2020-03-24:
今天 刷了之后，tcpping没有了。 但是 ip2socks 还是比较高啊
[image]

VNY0920, 2020-03-24:
你这cpu占用率怎么看的？ 不像是top命令。

kfen, 2020-03-24:
> 太神奇了，我在软件包里删掉ddns就好了，cpu使用率掉到个位数。不过ddns服务本来就没开启的。

问下老哥，我也遇到了同样的问题，删除luci-app-ddns之后，于此相关的软件包是否也一并移除？比如：ddns-scripts_dnspod和ddns-scripts_aliyun?????

keivenliao, 2020-03-24:
> 你这cpu占用率怎么看的？ 不像是top命令。

就是top啊

keivenliao, 2020-03-24:
我发现把这个停用了，ipt2socks 就关闭了。

VNY0920, 2020-03-24:
这个游戏udp中继我也没开。开了后Switch的nat类型会变成nat F，ps4玩实况会连不上服务器。
可能是因为我那个服务器是ssr不是ss类的。

closed 2020-03-24 by coolsnowwolf