coolsnowwolf/lede #3966: 关于dns污染的问题

ypx5, 2020-03-23:
版本 R20.3.5，在ShadowSocksR Plus中使用ss科学上网，运行方式选择绕过中国大陆ip模式，dns解析方式为使用PDNSD Tcp查询并缓存。还是有很多域名会被污染无法解析出正确的ip地址，比如www.duyaoss.com，会被污染为31.13.86.8，将这个域名加入强制走代理的域名列表，则可以解析出正确ip。我猜测目前是不是在gfwlist中的域名才会采用ss代理来远程解析？问题是目前有大量域名实际是被污染但不在列表内的，除非采用全局代理，否则这部分网站是上不去的。个人觉得chinadns-ng这个项目的dns防污染思路非常不错，大神不知道是否可以参考？https://github.com/zfl9/chinadns-ng   chinadns-ng在pandavan系统中是有的，个人实测确实可以解决类似的问题，但可惜似乎不太稳定，不能稳定使用。个人觉得其实可以简化一下，gfwlist中的域名直接远程解析，其他域名本地dns解析，如果解析出来的地址是国内的，则直接使用，如果是国外的，则用远程可信dns再次解析一下，应该就可以解决绝大部分问题了，毕竟目前dns污染给的ip地址似乎都是国外的地址。

hcym, 2020-03-23:
passwall用的是你说的东西，感觉稳定一点点，不过他的系统dnsmasq都完全劫持的，不方便也不规范，还是大雕的规范一些

ypx5, 2020-03-23:
感谢告知，原来还不知道系统里有个passwall，刚才试了一下，虽然里面有chinadns-ng，但是还是解决不了上述问题，不知道是什么原因

Bnosora, 2020-03-23:
chinadns-ng本身就可以指定只走国外dns的域名：`gfwlist-file` 选项指定黑名单域名文件，命中的域名只走可信 DNS。

norikohxb, 2020-03-23:
> 版本 R20.3.5，在ShadowSocksR Plus中使用ss科学上网，运行方式选择绕过中国大陆ip模式，dns解析方式为使用PDNSD [...]

我也是对dns污染很好奇，你提供的这个例子算是我用不上的域名但是确实是给劫持的。在我的电脑没有加入强制ssr的时候。解析出来的是31.13.72.17.无法打开网站，加入解析以后是31.13.75.18可以打开网站了。一个是爱尔兰，一个是法国马赛。我一直以为lean大的ssr在dns解析已经是战无不摧了。（因为说实话我几乎不上你这个网站。所以也就没有遇到）还有其他给污染的域名吗？多扔几个给我也测试看看饱饱眼福。最好是知名一点的。比你给的https://www.duyaoss.com/ 看能否知名一点，普遍一点，多给几个，我也测试测试。

coolsnowwolf, 2020-03-23:
怎么说呢， 用 chinadns-ng 很难保证返回 DNS IP 的可预知性。

例如 Netflix.com ，用 chinadns-ng 返回的 IP 是根据你从“路由器源地址” 发出而决定的。而这种情况下，你是基本不可能逃过 Netflix 的反代理检测的了。。

而且只要从“路由器源地址” 发出返回的A记录，并不能该服务器保证离你的 VPS 最近，而是离“你的路由最近”

目前根据列表来决定保证了分流节点的 DNS 一定是从分流节点发出，且 IPSET 的获取唯一性（同一个域名只能命中一次 IPSET)，只需要及时更新 GFWLIST 列表即可

coolsnowwolf, 2020-03-23:
另外说一句， 对于 www.duyaoss.com， 31.13.72.17 和 31.13.75.18 都是它的正确 IP ，并不存在污染

norikohxb, 2020-03-23:
lean大我也是觉得不应该是劫持。那个爱尔兰ip无法打开是不是他有针对访问区域做限制。所以只能是法国马赛可以打开，不过说实话这些都是小众网站，说白了大部分经常访问的国外网站都可以正确打开就很满意了。总之我用您的lede，是连dns劫持都不开的。

Bnosora, 2020-03-23:
> 怎么说呢， 用 chinadns-ng 很难保证返回 DNS IP 的可预知性。
> 
> 例如 Netflix.com ，用 chinadns-ng 返回的 IP 是根据你从“路由器源地址” 发出而决定的。而这种情况下，你是基本不可能逃过 Netflix 的反代理检测的了。。
> 
> 而且只要从“路由器源地址” 发出返回的A记录，并不能该服务器保证离你的 VPS 最近，而是离“你的路由最近”
> 
> 目前根据列表来决定保证了分流节点的 DNS 一定是从分流节点发出，且 IPSET 的获取唯一性（同一个域名只能命中一次 IPSET)，只需要及时更新 GFWLIST 列表即可

为什么会难以保证返回dns ip的可预知性呢？例如您说的netflix.com，因为国内dns的解析结果不是国内ip，所以用chinadns-ng的查询结果一定是可信dns的结果。
chinadns-ng的逻辑简化一下就是“解析结果是国内ip才会引用国内dns的解析结果，否则一律引用可信dns的结果。这和大陆白名单的逻辑完全一致，因此个人认为非常适合大陆白名单的设备使用。

NF的反代理检测确实是个问题，不知道能否靠`gfwlist-file`这个选项配合您现有的nf域名表解决？这个选项可以让列表内的域名不查询国内dns，只向可信dns发出解析请求。

cdn友好的问题对gfwlist的设备存在，但对大陆白名单的设备不存在。因为chinadns-ng的逻辑和大陆白名单完全一致。因此只要可信dns走代理，那结果一定是离vps最近的。实测也是如此。

目前pdnsd的问题和chinadns-ng正好相反，完美契合gfwlist模式的设备，但不适合大陆白名单模式的设备。比如cdn.data.fate-go.jp，这个域名不会被pdnsd查询，但在大陆白名单模式下会走代理。这就导致了cdn和vps离的很远。

hcym, 2020-03-24:
听大雕的没错，毕竟专业人士，就像n1单臂，我就是学了hanwckf方法才永久安定，恩山五花八门的方法都是有隐患的

ypx5, 2020-03-24:
> 我也是对dns污染很好奇，你提供的这个例子算是我用不上的域名但是确实是给劫持的。在我的电脑没有加入强制ssr的时候。解析出来的是31.13.72.17.无法打开网站，加入解析以后是31.13.75.18可以打开网站了。一个是爱尔兰，一个是法国马赛。我一直以为lean大的ssr在dns解析已经是战无不摧了。（因为说实话我几乎不上你这个网站。所以也就没有遇到）还有其他给污染的域名吗？多扔几个给我也测试 [...]

知名的网站应该没有，肯定都在gfwlist里面，不会有问题，但是确实有大量实际被墙但是不在列表里的网站是被污染打不开的，比如我因为个人原因经常看看澳门的新闻，这个网站http://macaudailytimes.com.mo/  似乎也是被污染的，你可以测试一下。

ypx5, 2020-03-24:
> 为什么会难以保证返回dns ip的可预知性呢？例如您说的netflix.com，因为国内dns的解析结果不是国内ip，所以用chinadns-ng的查询结果一定是可信dns的结果。
> chinadns-ng的逻辑简化一下就是“解析结果是国内ip才会引用国内dns的解析结果，否则一律引用可信dns的结果。这和大陆白名单的逻辑完全一致，因此个人认为非常适合大陆白名单的设备使用。 [...]

是啊，chinadns-ng的原理听起来很美，和大陆白名单模式应该是完美契合，但是实际用下来就是不稳定，不知道是配置问题还是代码没有完善。。。

ypx5, 2020-03-24:
> 怎么说呢， 用 chinadns-ng 很难保证返回 DNS IP 的可预知性。
> 
> 例如 Netflix.com ，用 chinadns-ng 返回的 IP 是根据你从“路由器源地址” 发出而决定的。而这种情况下，你是基本不可能逃过 Netflix 的反代理检测的了。。
> 
> 而且只要从“路由器源地址” 发出返回的A记录，并不能该服务器保证离你的 VPS 最近，而是离“你的路由最近”
> 
> 目前根据列表来决定保证了分流节点的 DNS 一定是从分流节点发出，且 IPSET 的获取唯一性（同一个域名只能命中一次 IPSET)，只需要及时更新 GFWLIST 列表即可

感谢大神回复，这个问题确实只要能及时更新GFWLIST列表也是可以接受的，我想再请教一下，现在用的GFWLIST是哪里维护的，我们可以提交上面没有但实际被墙的网址吗？

norikohxb, 2020-03-24:
说实话，感觉lean的整个op还是比较持之以恒的，之前我也编译passwall（好像反映速度感觉快一丢丢）现在我也不编译passwall了，感觉lede的考虑性是经过好几年的沉淀下来。这种域名从马赛变成爱爱尔兰应该还可以改进程序来解决，但是毕竟是小众网站。大部分访问国外的可以正常感觉就可以了，不行自己加一下黑白名单，当然能改进就最好。之前passswall比较快的时候我还去编译lienol的openwrt，三天两头要重新git clone，那个git pull无法用了。这个也很是头疼。现在都专心用lede了。就是不知道lean大能否借鉴passwall里面某些响应速度比较快的一些方法。还有那个helloworld，仅仅是开发者心血来潮来那么一次花样修改。之后就没有更新了。还是咱们的ssrplus和lede更新最持之以恒。

norikohxb, 2020-03-24:
> 知名的网站应该没有，肯定都在gfwlist里面，不会有问题，但是确实有大量实际被墙但是不在列表里的网站是被污染打不开的，比如我因为个人原因经常看看澳门的新闻，这个网站http://macaudailytimes.com.mo/ 似乎也是被污染的，你可以测试一下。

没有加入强制走ssr的时候解析ip是 31.13.66.23 加入ssr强制以后得到的ip是192.254.208.42确实是可以了感觉gcd污染的东西不少啊。这个就觉得应该是zf污染的了。那个爱尔兰还有法国马赛可能还是不同地方请求给不同ip，这个感觉就是真的给污染了。好在小众，哈哈，一打开都是英文新闻，看起来很吃力。多给几个网站我来试试看，我很喜欢试试看这种劫持的，一直自己上网的都没有遇到。难怪总有人叫喊着要smartdns加进来。估计是要满足你这种需要的，我就纳闷了，我自己平时上的国外的网站，我连lede的dns劫持都不开，怎么还有人那么多要求要smartdns，设置内外分开解析。我是都给lede的ssr包办了

ypx5, 2020-03-24:
> 没有加入强制走ssr的时候解析ip是 31.13.66.23 [...]

是的，不同的人上外网的需求不一样，很多人只上一些著名的外网网址，那其实用gfw模式就够了，但我呢需要经常上各种外网，再加上电信163线路的出口拥堵一塌糊涂，所以弄一个cn2好梯子然后开国内ip白名单模式，外网全部交给梯子就是最适合的，在这种情况下就会经常遇到各种网站实际被墙但不在gfwlist里面的情况了。这种情况确实是很多的，比如https://jetstream.surf/zh-TW/home 我以前用过的梯子网站，也是打不开的

norikohxb, 2020-03-24:
没有加入ssr强制名单的时候 jetstream的ip是104.16.53.111；104.16.51.111；104.16.52.111；104.16.54.111 加入强制ssr以后的ip是104.27.178.188可以打开。用梯子可以打开，感觉你最好的办法是用自己的域名库反馈给gfwlist的人，这样更新进去了就不用自己做策略了。

coolsnowwolf, 2020-03-24:
应该认识到，并不会有一种方法是完美适合所有人的所有情况的。

就例如理论上 smartdns 和 chinadns-ng 理论上对 A 记录分流更智能，但这两个就是更容易不正常工作。我这边一天里面观察 smartdns 会 crash 几十次以上，chinadns-ng 也有稳定性问题

相反 pdnsd 运行的时候甚至不需要进程守护。。。我都没看过这个程序crash 的

对开发者来说，流程和结果预期可控是更容易维护的项目，也更容易定位问题，所以尽量减少这种“黑盒子”

PS: 今天的新版本加入了可自定义订阅 GFWLIST ，推荐用 https://github.com/Loukky/gfwlist-by-loukky 这个 fork

norikohxb, 2020-03-24:
> 应该认识到，并不会有一种方法是完美适合所有人的所有情况的。
>
> 就例如理论上 smartdns 和 chinadns-ng 理论上对 A 记录分流更智能，但这两个就是更容易不正常工作。我这边一天里面观察 smartdns 会 crash 几十次以上，chinadns-ng 也有稳定性问题
>
> 相反 pdnsd 运行的时候甚至不需要进程守护。。。我都没看过这个程序crash 的 [...]

lean大今天的新版本能否解决udp的问题，要不要我去刷一下以前的版本，看看是从哪天开始就开始udp无法通过博联智慧星控制开关了

Bnosora, 2020-03-24:
> 是啊，chinadns-ng的原理听起来很美，和大陆白名单模式应该是完美契合，但是实际用下来就是不稳定，不知道是配置问题还是代码没有完善。。。

我是局域网里单独开个vm跑chinadns-ng，很稳定，如果非要用的话你可以试试这方法。有个问题是`--timeout-sec`得根据实际情况改改，不然会有很多超时错误。

ypx5, 2020-03-24:
> 应该认识到，并不会有一种方法是完美适合所有人的所有情况的。
>
> 就例如理论上 smartdns 和 chinadns-ng 理论上对 A 记录分流更智能，但这两个就是更容易不正常工作。我这边一天里面观察 smartdns 会 crash 几十次以上，chinadns-ng 也有稳定性问题
>
> 相反 pdnsd 运行的时候甚至不需要进程守护。。。我都没看过这个程序crash 的 [...]

感谢大神的付出，可以自定义gfwlist，基本就没啥问题了

rampageX, 2020-03-24:
> 怎么说呢， 用 chinadns-ng 很难保证返回 DNS IP 的可预知性。
> 
> 例如 Netflix.com ，用 chinadns-ng 返回的 IP 是根据你从“路由器源地址” 发出而决定的。而这种情况下，你是基本不可能逃过 Netflix 的反代理检测的了。。
> 
> 而且只要从“路由器源地址” 发出返回的A记录，并不能该服务器保证离你的 VPS 最近，而是离“你的路由最近”
> 
> 目前根据列表来决定保证了分流节点的 DNS 一定是从分流节点发出，且 IPSET 的获取唯一性（同一个域名只能命中一次 IPSET)，只需要及时更新 GFWLIST 列表即可

chinadns-ng 只做分流，具体查询还看上游服务器。还是以 Netflix.com 为例，如果可信服务器为 dns2socks ，那么返回的 A 记录应该就是离 VPS 最近，而且最新的 dns2socks 可以自定义 ECS 地址，可以完美解决那些国内中转服务器的问题。

norikohxb, 2020-03-24:
> chinadns-ng 只做分流，具体查询还看上游服务器。还是以 Netflix.com 为例，如果可信服务器为 dns2socks ，那么返回的 A 记录应该就是离 VPS 最近，而且最新的 dns2socks 可以自定义 ECS 地址，可以完美解决那些国内中转服务器的问题。

我在编译ssrplus，并没有打钩那个dns2sock，请问有没有问题，会不会影响到我的trojan的运行？虽然编译可以通过。

hxjs123, 2020-03-24:
我来说下我的情况，和使用愿景好了。就目前的情况来说，感觉在ssrp或者passwall难免会有一些节点服务器会有明文数据匹配拦截一些地址，那么加密dns比较符合我的平常使用环境。比如我强制下载机不走代理，但是域名解析依然希望由路由解析，并且能无污染，但是p2p之类的可能就会被拦截。目前只能由smartdns接手暂时解决这问题。不过不可避免的是，真的不稳定，经常smartdns会crashed。现在国内外都有加密dns服务了，总体上服务效果尚可，可以是时候做个计划，路由层面上都上加密dns，而不是现在chrome和Firefox程序内使用加密dns。

xhqpp, 2020-03-25:
> 听大雕的没错，毕竟专业人士，就像n1单臂，我就是学了hanwckf方法才永久安定，恩山五花八门的方法都是有隐患的

N1 hanwckf的方法有链接吗 我也去学学

denver9527, 2020-03-25:
对于现在的递归式DNS，理论上来说DNS返回的是无法劫持的。应该都是在发出的方向上被墙吧。利用快速返回错误的地址欺骗客户端

nftbty, 2020-03-25:
说到DNS污染的问题，遇到有段时间了。sourceforge.net网页和下载服务器域名都被污染。
`sourceforge.net  45.116.145.122`
`master.dl.sourceforge.net 220.250.64.26`
`nchc.dl.sourceforge.net 220.250.64.26`
`jaist.dl.sourceforge.net  220.250.64.26`
还可以把 .dl 前面镜像站名随便换其他不存在的比如abc,cn,us,wuran等，都是同样的IP`220.250.64.26`，运营商联通。不知道其他地区、运营商有没有污染这个站的域名，可以用这个网站的域名测试。

norikohxb, 2020-03-25:
sourceforge.net打开正常没有你说的那个问题。

nftbty, 2020-03-25:
@norikohxb  用宽带和流量都可以上吗？哪家运营商？我这边宽带和手机都是联通，上不去。手机流量是开关飞行后试的。而且我ping 不存在的(字母).dl.sourceforge.net都会有IP，出现好久了。

ypx5, 2020-03-25:
> 我是局域网里单独开个vm跑chinadns-ng，很稳定，如果非要用的话你可以试试这方法。有个问题是`--timeout-sec`得根据实际情况改改，不然会有很多超时错误。
感谢提醒，我弄了个树莓派跑了一下chinadns-ng，修改超时为10秒后确实很稳定，但不知道为什么。文档说默认超时是5秒，但默认的参数就是不稳，5秒其实也远远够了啊，感觉解析都挺快的，真要10秒那不卡死了啊。

norikohxb, 2020-03-25:
> @norikohxb 用宽带和流量都可以上吗？哪家运营商？我这边宽带和手机都是联通，上不去。手机流量是开关飞行后试的。而且我ping 不存在的(字母).dl.sourceforge.net都会有IP，出现好久了。是不梯子出去很正常

厦门电信，lede原生，没有去分析

kenyloveg, 2020-03-26:
所以smartdns全部服务器都用DOH，DOT去连接国外公共DNS，而这些DNS的IP直连，强制不走机场线路

norikohxb, 2020-03-28:
请教一个问题，我发现我的三星的平板，note12.2寸，一个比较老的平板，我给他放在镜柜那边刷牙的时候可以看看youtube。然后每天早上发现起来看youtube都需要关闭wifi，再开启一次才可以正常看youtube视频，否则就打不开，总是在那边转圈圈，app是youtube官方的。我在想这个是不是因为我的trojan不可能是一个晚上都是稳定，谷歌云抽风是可以理解，在他抽风的时候，youtube的ip变成了被污染的，然后早上起来刷牙的时候还必须关闭再开启有点是清理这个。
咱们的ssr，能否在所用的trojan服务器或者是v2ray的服务器抽风的时候（这个时候肯定上不了youtube可以理解）抽风的时候或者是trojan服务器断开的时候。仍然强制让youtube的域名不会给污染。一直持续到trojan缓过神来，恢复正常的时候。能否这样实现呢？

zatenchan, 2020-04-03:
> 请教一个问题，我发现我的三星的平板，note12.2寸，一个比较老的平板，我给他放在镜柜那边刷牙的时候可以看看youtube。然后每天早上发现起来看youtube都需要关闭wifi，再开启一次才可以正常看youtube视频，否则就打不开，总是在那边转圈圈，app是youtube官方的。我在想这个是不是因为我的trojan不可能是一个晚上都是稳定，谷歌云抽风是可以理解，在他抽风的时候，youtube [...]

把service.googleapis.cn加到强制代理里试试

closed 2020-07-04 by coolsnowwolf