coolsnowwolf/lede #3967: 长期困扰我的DNS_PROBE_FINISHED_NXDOMAIN

BOBINIUNIU, 2020-03-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：这个问题从很早就有。在运行一段时间后，大约1到2天，访问经过代理的网站chrome会报错DNS_PROBE_FINISHED_NXDOMAIN，刷新一下又能连上。访问不被代理的国内网站则不会有这个问题。
我的设置是ssr，大陆白名单模式，PDNSD （换过dns2sock，一样），1.1.1.1 （ssr设置是aes256，aes128sha1，tls1.2）
DNS 查询缓存的大小 1000

我觉得这个问题可能是由于dns缓存引起的，因为重启一下问题会消失，但是第二天或第三天又会随机出现。

### (2) 路由器型号和固件版本

A：x86, OpenWrt R20.3.19

### (3) 详细日志

A：

Aspiriniii, 2020-03-23:
偶尔遇到过，刷新下再进就好了

hxlls, 2020-03-23:
吧去广告关了就行了

coolsnowwolf, 2020-03-23:
关闭 IPv6 或其他 DNS 类软件（去广告，SMARTDNS 等立马好了

BOBINIUNIU, 2020-03-24:
感谢各位回复，实际上我并未安装去广告和smartdns之类的服务。作为一个极简主义爱好者，什么 BBR，硬件加速，FullCone-NAT能关的都给关了，adbyby和smartdns都没有编译进去，luci里只选了ssr和samba。

IPV6是指的 DNS设置里禁止解析 IPv6 DNS 记录吗，那个也是关掉的。
这个问题出现得也很随机，难道是墙在作怪？

coolsnowwolf, 2020-03-24:
禁止解析 IPv6 DNS 记录 开启

BOBINIUNIU, 2020-03-24:
> 禁止解析 IPv6 DNS 记录 开启

我说错了，那个选项是开启的。WAN和LAN里面的‘使用内置的 IPv6 管理’选项需要关闭吗？

closed 2020-03-25 by coolsnowwolf

sdandroid, 2022-01-15:
@BOBINIUNIU 问题解决没

netcalf, 2022-02-25:
openwrt软路由，开启了smartDNS，ssr使用本机smartDNS解析，我的一个内网地址的解析出现DNS_PROBE_FINISHED_NXDOMAIN，后来改openwrt-网络-DHCP/DNS，增加了一个自定义劫持域名，问题解决。另：注意刷新路由ipconfig /flushdns

hcym, 2022-02-25:
mosdns，这个完美解决了大雕和ipv6的问题😂

wangyubow, 2022-06-20:
求助： 登陆 gmail （https://mail.google.com/mail/u/0/）出现DNS_PROBE_FINISHED_NXDOMAIN。

登陆 gmail （https://mail.google.com/mail/u/0/）出现DNS_PROBE_FINISHED_NXDOMAIN。
登陆youtube失败，无法同步gmail账号。Can't sync to  ****.gmail.com

以上办法，都测试过（除了mosdns外），无效。

但是，在Ubuntu，使用Qv2ray2.7.0， 使用Xray，和OpenWrt一起使用， 一切正常，可以登陆GMAIL。

配置情况， 以下配置，均出现一样的问题 DNS_PROBE_FINISHED_NXDOMAIN：
OpenWrt + passwall
OpenWrt + passwall + smartDNS
OoenWrt + Bypass
OpenWrt + HelloWorld

软件版本：
固件版本	OpenWrt R22.6.16 / LuCI Master (git-22.168.62806-d59f9ce)
内核版本	5.15.48
luci-app-passwall	4.54-3
luci-i18n-passwall-zh-cn	4.54-3