coolsnowwolf/lede #3971: SOCKS5 代理服务端（全局）：显示运行中，但实际并没有开

R3pl4c3r, 2020-03-23:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：SOCKS5 代理服务端（全局）：显示运行中，但实际并没有开
[image]
[image]
[image]

### (2) 路由器型号和固件版本

A：x86_64 OpenWrt R20.3.19 / LuCI Master (git-20.079.47035-c242d65)

### (3) 详细日志

A：
```
Mon Mar 23 21:15:11 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Mar 23 21:15:12 2020 kern.info /usr/bin/ssr-server[1587]: UDP relay enabled
Mon Mar 23 21:15:12 2020 kern.info /usr/bin/ssr-server[1587]: initializing ciphers... rc4-md5
Mon Mar 23 21:15:12 2020 kern.info /usr/bin/ssr-server[1587]: tcp port reuse enabled
Mon Mar 23 21:15:12 2020 kern.info /usr/bin/ssr-server[1587]: udp port reuse enabled
Mon Mar 23 21:15:12 2020 kern.info /usr/bin/ssr-server[1587]: listening at 0.0.0.0:65500
Mon Mar 23 21:15:12 2020 kern.info /usr/bin/ssr-server[1587]: running from root user
Mon Mar 23 21:15:22 2020 kern.err /usr/bin/ssr-redir[1272]: server recv: Connection reset by peer
Mon Mar 23 21:15:22 2020 kern.err /usr/bin/ssr-redir[1258]: server recv: Connection reset by peer
Mon Mar 23 21:15:22 2020 kern.err /usr/bin/ssr-redir[1256]: server recv: Connection reset by peer
Mon Mar 23 21:15:22 2020 kern.err /usr/bin/ssr-redir[1258]: server recv: Connection reset by peer
Mon Mar 23 21:15:31 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Mar 23 21:15:31 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Mar 23 21:15:31 2020 cron.info crond[2442]: crond (busybox 1.30.0) started, log level 8
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2706]: initializing ciphers... rc4-md5
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2706]: tcp port reuse enabled
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2706]: listening at 0.0.0.0:1234
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2706]: running from root user
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2708]: initializing ciphers... rc4-md5
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2708]: tcp port reuse enabled
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2708]: listening at 0.0.0.0:1234
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2708]: running from root user
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2710]: initializing ciphers... rc4-md5
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2710]: tcp port reuse enabled
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2710]: listening at 0.0.0.0:1234
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2710]: running from root user
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2712]: initializing ciphers... rc4-md5
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2712]: tcp port reuse enabled
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2712]: listening at 0.0.0.0:1234
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2712]: running from root user
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2714]: initializing ciphers... rc4-md5
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2714]: tcp port reuse enabled
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2714]: listening at 0.0.0.0:1234
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2714]: running from root user
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2716]: initializing ciphers... rc4-md5
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2716]: tcp port reuse enabled
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2716]: listening at 0.0.0.0:1234
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2716]: running from root user
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2718]: initializing ciphers... rc4-md5
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2718]: tcp port reuse enabled
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2718]: listening at 0.0.0.0:1234
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2718]: running from root user
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2720]: initializing ciphers... rc4-md5
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2720]: tcp port reuse enabled
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2720]: listening at 0.0.0.0:1234
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2720]: running from root user
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2722]: initializing ciphers... rc4-md5
Mon Mar 23 21:15:32 2020 kern.info /usr/bin/ssr-redir[2722]: tcp port reuse enabled
[... 615 lines cut ...]
Mon Mar 23 21:25:45 2020 kern.err /usr/bin/ssr-redir[22047]: client_decode
Mon Mar 23 21:25:45 2020 kern.err /usr/bin/ssr-redir[22049]: client_decode
Mon Mar 23 21:25:48 2020 kern.err /usr/bin/ssr-redir[22045]: client_decode
Mon Mar 23 21:25:48 2020 kern.err /usr/bin/ssr-redir[22055]: client_decode
Mon Mar 23 21:25:49 2020 kern.err /usr/bin/ssr-redir[22045]: client_decode
Mon Mar 23 21:25:50 2020 kern.err /usr/bin/ssr-redir[22043]: client_decode
Mon Mar 23 21:25:50 2020 kern.err /usr/bin/ssr-redir[22035]: client_decode
Mon Mar 23 21:25:50 2020 kern.err /usr/bin/ssr-redir[22039]: client_decode
Mon Mar 23 21:25:50 2020 kern.err /usr/bin/ssr-redir[22045]: client_decode
Mon Mar 23 21:25:50 2020 kern.err /usr/bin/ssr-redir[22037]: client_decode
Mon Mar 23 21:25:50 2020 kern.err /usr/bin/ssr-redir[22051]: client_decode
Mon Mar 23 21:25:50 2020 kern.err /usr/bin/ssr-redir[22053]: client_decode
Mon Mar 23 21:25:50 2020 kern.err /usr/bin/ssr-redir[22035]: client_decode
Mon Mar 23 21:25:55 2020 kern.err /usr/bin/ssr-redir[22043]: client_decode
Mon Mar 23 21:26:26 2020 kern.err /usr/bin/ssr-redir[22053]: server recv: Connection reset by peer
Mon Mar 23 21:27:09 2020 kern.err /usr/bin/ssr-redir[22051]: server recv: Connection reset by peer
Mon Mar 23 21:27:09 2020 kern.err /usr/bin/ssr-redir[22041]: server recv: Connection reset by peer
Mon Mar 23 21:31:57 2020 kern.err /usr/bin/ssr-redir[22033]: server recv: Connection reset by peer
Mon Mar 23 21:33:03 2020 user.info mwan3track[6981]: Check (ping) failed for target "114.114.114.114" on interface wan (pppoe-wan)
Mon Mar 23 21:33:08 2020 user.info mwan3track[6981]: Lost 1 ping(s) on interface wan (pppoe-wan)
Mon Mar 23 21:34:22 2020 kern.err /usr/bin/ssr-redir[22041]: server recv: Connection reset by peer
Mon Mar 23 21:34:50 2020 kern.err /usr/bin/ssr-redir[22037]: server recv: Connection reset by peer
Mon Mar 23 21:34:50 2020 kern.err /usr/bin/ssr-redir[22045]: server recv: Connection reset by peer
Mon Mar 23 21:34:50 2020 kern.err /usr/bin/ssr-redir[22031]: server recv: Connection reset by peer
```

tone88, 2020-03-23:
我也遇到这情况 
SOCKS5 代理服务端（全局）未开启，服务端中开启socks5的情况下，状态中显示如下图：
[image]
SOCKS5 代理服务端（全局）显示为“运行中”

ESXI下使用的软路由，今天的源码编译全新刷的机。 刷完后重置一次再配置各项设置。

coolsnowwolf, 2020-03-23:
已修复，错误显示

closed 2020-03-23 by coolsnowwolf

tone88, 2020-03-24:
> 已修复，错误显示

今天试了下，还是有问题，默认停用情况情下状态显示仍然是运行中，这时需要手动在高级设置里点一下保存&应用后状态里才会显示为 未运行

yihongyuelan, 2020-03-26:
@tone88 
请教下，socks5全局设置和服务端都开启后，如果要从外网访问，那么需要在防火墙里面添加端口转发，你有试过这样操作么？