coolsnowwolf/lede #3974: 编译固件时出现Found invalid GPT and valid MBR; converting MBR to GPT format in memory. 警告

minfengyk, 2020-03-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
编译固件出现Found invalid GPT and valid MBR; converting MBR to GPT format in memory. 警告
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
固件编译成功，但查看编译详情，发现有Found invalid GPT and valid MBR; converting MBR to GPT format in memory. 警告。编译出来的combined固件完美运行，但同时编译出的uefi固件启动后总是报错：F2FS-fs (sda3)：Can't find valid F2FS filesystem in superblock（曾在issues里请教过这个问题，一度认为是文件系统不同引起的），Google和搜索issues没有获得答案，现在考虑这两个问题是同一原因，不知道是否bug，还是编译时选项冲突引起的。

### (1) 具体问题
固件编译成功，但查看编译详情，发现有Found invalid GPT and valid MBR; converting MBR to GPT format in memory. 警告。编译出来的combined固件完美运行，但同时编译出的uefi固件启动后总是报错：F2FS-fs (sda3)：Can't find valid F2FS filesystem in superblock（曾在issues里请教过这个问题，一度认为是文件系统不同引起的），Google和搜索issues没有获得答案，现在考虑这两个问题是同一原因，不知道是否bug，还是编译时选项冲突引起的。

### (2) 路由器型号和固件版本
型号：Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz : 1 Core 2 Thread (CpuMark : 37155.441337 Scores)
固件：OpenWrt R20.3.19 / LuCI Master (git-20.062.50551-a6189c5)

### (3) 详细日志
sgdisk -g "/home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.img"

***************************************************************
Found invalid GPT and valid MBR; converting MBR to GPT format
in memory. 
***************************************************************

Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
The operation has completed successfully.
sgdisk -t 2:EF00 "/home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.img"
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
The operation has completed successfully.
sgdisk -t 3:EF02 "/home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.img"
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
The operation has completed successfully.
sgdisk -u 4:8A860D54-9A57-45EE-8A2F-6A4480360002 "/home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.img"
Warning: The kernel is still using the old partition table.
The new table will be used at the next reboot or after you
run partprobe(8) or kpartx(8)
The operation has completed successfully.
sgdisk -h "/home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.img"
# Setup EFI grub
grub-bios-setup-efi --device-map="/home/minfeng/lede/build_dir/target-x86_64_musl/linux-x86_64/grub2/device.map" -d "/home/minfeng/lede/build_dir/target-x86_64_musl/linux-x86_64/grub2" -r "hd0,gpt1" "/home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.img"
cp -fpR /home/minfeng/lede/build_dir/target-x86_64_musl/linux-x86_64/root.squashfs /home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-rootfs-squashfs.img
cp -fpR /home/minfeng/lede/build_dir/target-x86_64_musl/linux-x86_64/bzImage /home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-vmlinuz
gzip -f9n /home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-rootfs-squashfs.img
gzip -f9n /home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-combined-squashfs.img
gzip -f9n /home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.img
IPKG_NO_SCRIPT=1 IPKG_INSTROOT=/home/minfeng/lede/build_dir/target-x86_64_musl/root.orig-x86 TMPDIR=/home/minfeng/lede/build_dir/target-x86_64_musl/root.orig-x86/tmp /home/minfeng/lede/staging_dir/host/bin/opkg --offline-root /home/minfeng/lede/build_dir/target-x86_64_musl/root.orig-x86 --force-postinstall --add-dest root:/ --add-arch all:100 --add-arch x86_64:200 list-installed > /home/minfeng/lede/bin/targets/x86/64/openwrt-x86-64-generic.manifest
make[5]: Leaving directory '/home/minfeng/lede/target/linux/x86/image'
make[4]: Leaving directory '/home/minfeng/lede/target/linux/x86'
make[3]: Leaving directory '/home/minfeng/lede/target/linux'
time: target/linux/install#69.93#9.67#90.91
make[2]: Leaving directory '/home/minfeng/lede'
export MAKEFLAGS= ;make -w -r package/index
make[2]: Entering directory '/home/minfeng/lede'

dongjuanyong, 2020-03-24:
屁事没有，不当用，请无视：

> Found invalid GPT and valid MBR; converting MBR to GPT format in memory.

这是因为磁盘镜像还在合成，GPT分区表配置不对，这个是提示你正在修复。

> F2FS-fs (sda3)：Can't find valid F2FS filesystem in superblock

生成的uefi固件为了保证兼容性（在bios系统上能启动），建立了第三个分区（sda3）pbr记录来保证能引导，为了节省空间，此分区只有1mb大小，没有任何文件系统。赶巧lean的固件默认加入了f2fs支持，自动mount的时候扫到这个分区，发现认不出来，就冒了句提示

minfengyk, 2020-03-24:
> 屁事没有，不当用，请无视：
>
> Found invalid GPT and valid MBR; converting MBR to GPT format in memory.
>
> 这是因为磁盘镜像还在合成，GPT分区表配置不对，这个是提示你正在修复。
>
> F2FS-fs (sda3)：Can't find valid F2FS filesystem in superblock [...]
谢谢！被这个问题困扰好久，解答了我的疑惑。不懂编程，出问题了也自己分析不了。Google和论坛也没有得到准确的答案，还是这里靠谱！再请教一下，如果不选择f2fs文件系统支持，是不是就不会有这个提示了？不支持f2fs文件系统，对系统性能没什么影响吧。

dongjuanyong, 2020-03-24:
可以去了，没影响，没人在pc上用f2fs，ssd主控是专门干这个的

minfengyk, 2020-03-25:
感谢！以后编译uefi固件时去掉这个选项。

closed 2020-07-04 by coolsnowwolf