coolsnowwolf/lede #3982: 阿里云日志异常

tone88, 2020-03-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
没有类似的issue

## 2. 详细叙述

### (1) 具体问题
最新源码编译，DDNS里阿里云日志异常。
全新刷机后手动设置各项配置。

### (2) 路由器型号和固件版本
X86  OpenWrt R20.3.19 / LuCI Master (git-20.068.29105-b2cf61d)

### (3) 详细日志
系统日志截图：
[image]
动态DDNS日志查看器截图：
[image]

详细日志稍后私发大雕

zfh135072, 2020-03-24:
同问题，好像泛解析也不行了

tone88, 2020-03-24:
> 同问题，好像泛解析也不行了

今天早上编译的才遇到这问题，之前的版本没感觉到这问题

EnnawYang, 2020-03-24:
我的没问题 阿里云、dnspodcn均正常解析，早上编译的

tone88, 2020-03-24:
> 我的没问题 阿里云、dnspodcn均正常解析，早上编译的

你是物理机还是虚拟机里跑的OP?    

我是在EXSI下跑的OP

EnnawYang, 2020-03-24:
> > 我的没问题 阿里云、dnspodcn均正常解析，早上编译的
> 
> 你是物理机还是虚拟机里跑的OP?
> 
> 我是在EXSI下跑的OP

物理

coolsnowwolf, 2020-03-24:
看看是否泛域名，需要用 * 这个符号的

tone88, 2020-03-24:
> 看看是否泛域名，需要用 * 这个符号的

没有泛域名，刚刚登陆阿里云看解析设置里显示是正常状态
[image]

 路由里的日志一直还在报错！

Tue Mar 24 09:44:41 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 1/0
Tue Mar 24 09:54:41 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 2/0
Tue Mar 24 10:04:42 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 3/0
Tue Mar 24 10:14:42 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 4/0
Tue Mar 24 10:24:43 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 5/0
Tue Mar 24 10:34:44 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 6/0
Tue Mar 24 10:44:44 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 7/0
Tue Mar 24 10:44:44 2020 user.err ddns-scripts[5687]: Aliyun: wget 错误代码: '8'
Tue Mar 24 10:44:44 2020 user.warn ddns-scripts[5687]: Aliyun: 传输失败 - 1/0 在 60 秒后重试
Tue Mar 24 10:55:45 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 8/0
Tue Mar 24 11:05:45 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 9/0
Tue Mar 24 11:15:46 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 10/0
Tue Mar 24 11:25:46 2020 user.warn ddns-scripts[5687]: Aliyun: Updating IP at DDNS provider failed - starting retry 11/0

EnnawYang, 2020-03-24:
建议再检查下设置

tone88, 2020-03-24:
> 建议再检查下设置

设置和之前的一样，没有变化的
设置如图：
[image]

EnnawYang, 2020-03-24:
> > 建议再检查下设置
> 
> 设置和之前的一样，没有变化的
> 设置如图：
> [image]

那我也不太清楚了

coolsnowwolf, 2020-03-24:
删除ddns配置再添加一次吧

tone88, 2020-03-24:
> 删除ddns配置再添加一次吧

删除了重新添加也不行，刚刚又重新编译了一次刷机，日志还是报错，搞不懂了。

closed 2020-03-25 by coolsnowwolf

free521521521, 2022-03-09:
域名解析实际成功，但是log里显示failed。奇怪