coolsnowwolf/lede #3983: 水星d12g有合适的openwrt吗？

canaanlin, 2020-03-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：之前家里买的水星d12g，买回来发现需要ttl刷，没办法直接刷，所以一直放在家里吃灰。现在登上看到lean发起征求说要不要停止对8+64的更新，所以在这之前特地的问一下。

### (2) 路由器型号和固件版本

A：水星D12G的CPU是MT7620DA，比常见的MT7620A多了一个字母“D”，应该是集成了DRAM，集成了64MB内存的CPU，其它参数跟MT7620A一样。5G芯片也是最常见的MT7612EN，闪存型号是25Q64CSIG，容量8MB。千兆交换机芯片的型号是**RTL8367S**

### (3) 补充

A：我又查了一下大体配置和极路由3差不多，芯片是一样的，只是极路由3是16+128的，而水星D12G是8+64，最主要的区别是极路由是Realtek的**RTL8211E**千兆交换芯片，而水星D12G千兆交换机芯片的型号是**RTL8367S**。
我怕直接开刷极路由3 的op容易端口用不了，所以来问一下L大的意见。

Wilson-coder, 2020-03-24:
肯定是不行的了，就好像都是骁龙965，你也不可能把小米的ROM刷到OPPO手机上啊。再说了，8M ROM的路由器，基本上啥也干不了，直接用厂家固件做个AP好了。需要用到OP的功能，直接上个软路由就好了。

Aes64X, 2020-03-24:
全集成一站式芯片。。。。这个就不要想了，原厂固件用吧。

hxlls, 2020-03-24:
16M大部分人投票都不需要支持了，8M 洗洗睡吧，芯片还不是大问题，问题是ROM太小

haxc, 2020-03-25:
建议有修砖能力再搞，刷个breed就可以刷固件了

closed 2020-03-25 by coolsnowwolf