coolsnowwolf/lede #3989: 编译一直卡在这个地方，也不报错

pixellimited, 2020-03-24:
x86_64-openwrt-linux-musl-g++ -Os -pipe -o /home/sss/lede/build_dir/target-x86_64_musl/node-v12.15.0/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/runtime/runtime-compiler.o ../deps/v8/src/runtime/runtime-compiler.cc '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-DV8_TARGET_ARCH_X64' '-DV8_EMBEDDER_STRING="-node.16"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DV8_USE_SNAPSHOT' '-DV8_CONCURRENT_MARKING' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' -I/home/sss/lede/staging_dir/target-x86_64_musl/usr/include -I../deps/v8 -I../deps/v8/include -I/home/sss/lede/build_dir/target-x86_64_musl/node-v12.15.0/out/Release/obj/gen/inspector-generated-output-root -I../deps/v8/third_party/inspector_protocol -I/home/sss/lede/build_dir/target-x86_64_musl/node-v12.15.0/out/Release/obj/gen/torque-output-root -I/home/sss/lede/build_dir/target-x86_64_musl/node-v12.15.0/out/Release/obj/gen/generate-bytecode-output-root  -pthread -Wno-unused-parameter -m64 -Wno-return-type -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/sss/lede/build_dir/target-x86_64_musl/node-v12.15.0/out/Release/.deps//home/sss/lede/build_dir/target-x86_64_musl/node-v12.15.0/out/Release/obj.target/v8_base_without_compiler/deps/v8/src/runtime/runtime-compiler.o.d.raw  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/sss/lede/build_dir/target-x86_64_musl/node-v12.15.0:node-v12.15.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/sss/lede/staging_dir/target-x86_64_musl/usr/include -I/home/sss/lede/staging_dir/target-x86_64_musl/include -I/home/sss/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/sss/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/sss/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include  -c

一直重复出现这些，也不报错，求怎样解决，谢谢了，新手

hxlls, 2020-03-24:
加钱换编译机就完事了

guochan2019, 2020-03-24:
toolchain-x86_64_gcc-7.5.0_musl拉不回来，开始编译前执行make download，SCP到dl目录确认下载完整性，再开始编译

closed 2020-03-25 by coolsnowwolf