coolsnowwolf/lede #3995: 如何取消强制选择的默认值？

Tuskio, 2020-03-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我想取消一些强制的默认值。翻看了源码，没有找到在哪（是我太菜了）。
请大佬告知一些强制的默认值在什么位置取消。

### (2) 路由器型号和固件版本

路由器型号：TP-Link Archer C7 v2
固件版本：lede-master（openwrt19.07）

### (3) 详细日志
 如图中bootstrp主题就默认不能取消，我想删除一些默认值。但是在翻看了源码以后，没有定位到位置，求各位大佬指点一下。感谢L大（@coolsnowwolf ）
[image]

hxlls, 2020-03-24:
好像是所有的主题都是建立在这个之上的，所以你取消不了

lphgor, 2020-03-24:
`luci-theme-bootstrap`是`luci`的依赖包，所以取消不掉。
如果真的想改的话，可以在执行完`./scripts/feeds install -a`之后，编辑`feeds/luci/collections/luci/Makefile`这个文件，把`luci-theme-bootstrap`改成你要的主题。

closed 2020-03-24 by Tuskio