coolsnowwolf/lede #3999: 2020.3.24日版本更新后无法在编译的时候选择vhd格式。否则编译失败

jh2921291, 2020-03-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
2020.3.24日版本更新后无法在编译的时候选择vhd格式。否则编译失败
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
，尝试使用未更新之前的源码 可以正常编译。然后在相同环境下更新最新源码后，就编译失败了。
### (1) 具体问题
编译失败，具体请看下面的详细日志
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
sgdisk -h "/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.img"
# Setup EFI grub
grub-bios-setup-efi --device-map="/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub2/device.map" -d "/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub2" -r "hd0,gpt1" "/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.img"
rm /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.vhdx || true
rm: cannot remove '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.vhdx': No such file or directory
/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/host/bin/qemu-img convert -O vhdx /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.img /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/bin/targets/x86/64/openwrt-x86-64-uefi-gpt-squashfs.vhdx
qemu-img: Unknown file format 'vhdx'
make[5]: *** [install-images] Error 1
make[4]: *** [install] Error 2
Makefile:350: recipe for target 'install-images' failed
make[5]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/target/linux/x86/image'
make[3]: *** [install] Error 2
Makefile:29: recipe for target 'install' failed
make[4]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/target/linux/x86'
Makefile:13: recipe for target 'install' failed
make[3]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/target/linux'
time: target/linux/install#71.57#8.22#74.26
make[2]: *** [target/linux/install] Error 2
target/Makefile:23: recipe for target 'target/linux/install' failed

jh2921291, 2020-03-24:
在本地机器编译和github云编译均尝试过。添加vhd格式后编译失败。不添加时能成功编译

R3pl4c3r, 2020-03-24:
确实有同样情况

jliveusa, 2020-03-24:
回退x86: fix path for qemu-img (#3856)之前的设置可解决

closed 2020-03-25 by coolsnowwolf