coolsnowwolf/lede #4003: 3.25编译的最版本打开迅雷upnpd不能自动映射端口

helianling, 2020-03-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：3.25编译的最版本打开迅雷upnpd不能自动映射端口

### (2) 路由器型号和固件版本

A：X86_64

### (3) 详细日志

A：Wed Mar 25 03:26:12 2020 daemon.notice netifd: Interface 'wan_6' is now up
Wed Mar 25 03:26:12 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Wed Mar 25 03:27:32 2020 daemon.err miniupnpd[3122]: Failed to get ip address for interface
Wed Mar 25 03:28:18 2020 daemon.err uhttpd[2730]: sh: out of range
Wed Mar 25 03:28:18 2020 daemon.err uhttpd[2730]: sh: out of range
Wed Mar 25 03:28:39 2020 daemon.err miniupnpd[3122]: Failed to get ip address for interface
Wed Mar 25 03:30:07 2020 daemon.err uhttpd[2730]: sh: out of range
Wed Mar 25 03:30:07 2020 daemon.err uhttpd[2730]: sh: out of range
Wed Mar 25 03:30:33 2020 daemon.err miniupnpd[3122]: Failed to get ip address for interface

alepennyq, 2020-03-25:
同样的问题，upnp更新后失效了，不止迅雷用不了，是都用不了。

jwf8280031, 2020-03-25:
同样都用不了，4.14和4.19都编译了，都不能用。

coolsnowwolf, 2020-03-25:
不再兼容迅雷，只兼容其他

coolsnowwolf, 2020-03-25:
[image]

alepennyq, 2020-03-25:
我2.0.20170421-2是正常的，前两天升级到2.0.20170421-2，不论勾不勾选IGDv1 模式都不生效。

coolsnowwolf, 2020-03-25:
ssh 里面， cp /rom/etc/config/upnpd /etc/config/upnpd， 然后重启即可

coolsnowwolf, 2020-03-25:
界面重新应用下

Aspiriniii, 2020-03-25:
实测迅雷X无问题
[image]

alepennyq, 2020-03-25:
> ssh 里面， cp /rom/etc/config/upnpd /etc/config/upnpd， 然后重启即可

还是不行：
root@OpenWrt:~# cp /rom/etc/config/upnpd /etc/config/upnpd
root@OpenWrt:~# /etc/init.d/miniupnpd restart
sh: out of range

[image]

qianxu2001, 2020-03-25:
用个迅雷要什么 upnp，开会员加速就行了

alepennyq, 2020-03-25:
> ssh 里面， cp /rom/etc/config/upnpd /etc/config/upnpd， 然后重启即可

日志：
Wed Mar 25 11:47:13 2020 daemon.err miniupnpd[3416]: Failed to add NAT-PMP 20489 tcp->192.168.1.101:32400 'NAT-PMP 20489 tcp'
Wed Mar 25 11:50:13 2020 daemon.err uhttpd[2781]: sh: out of range
Wed Mar 25 11:50:13 2020 daemon.err uhttpd[2781]: sh: out of range
Wed Mar 25 11:51:41 2020 daemon.err uhttpd[2781]: sh: out of range
Wed Mar 25 11:51:41 2020 daemon.err uhttpd[2781]: sh: out of range
Wed Mar 25 12:30:00 2020 cron.info crond[2719]: USER root pid 18145 cmd /etc/ssrplusupdate.sh
Wed Mar 25 12:30:16 2020 daemon.warn pdnsd[6118]: Caught signal 15. Exiting.
Wed Mar 25 12:30:16 2020 cron.info crond[18489]: crond (busybox 1.30.0) started, log level 8
Wed Mar 25 12:30:17 2020 daemon.info pdnsd[18747]: pdnsd-1.2.9b-par starting.
Wed Mar 25 12:40:03 2020 daemon.err miniupnpd[3416]: Failed to add NAT-PMP 20489 tcp->192.168.1.101:32400 'NAT-PMP 20489 tcp'
Wed Mar 25 13:35:30 2020 daemon.err uhttpd[2781]: sh: out of range
Wed Mar 25 13:35:30 2020 daemon.err uhttpd[2781]: sh: out of range
Wed Mar 25 13:36:01 2020 daemon.err uhttpd[2781]: sh: out of range
Wed Mar 25 13:36:01 2020 daemon.err uhttpd[2781]: sh: out of range
Wed Mar 25 13:37:04 2020 authpriv.info dropbear[22630]: Child connection from 192.168.1.102:58183
Wed Mar 25 13:37:04 2020 authpriv.notice dropbear[22630]: Password auth succeeded for 'root' from 192.168.1.102:58183
Wed Mar 25 13:37:32 2020 daemon.notice miniupnpd[3416]: shutting down MiniUPnPd
Wed Mar 25 13:37:32 2020 daemon.notice miniupnpd[23657]: HTTP listening on port 5000
Wed Mar 25 13:37:32 2020 daemon.notice miniupnpd[23657]: Listening for NAT-PMP/PCP traffic on port 5351
Wed Mar 25 13:40:03 2020 daemon.err miniupnpd[23657]: Failed to add NAT-PMP 20489 tcp->192.168.1.101:32400 'NAT-PMP 20489 tcp'
Wed Mar 25 13:40:22 2020 authpriv.info dropbear[28397]: Child connection from 192.168.1.102:58322
Wed Mar 25 13:40:22 2020 authpriv.notice dropbear[28397]: Password auth succeeded for 'root' from 192.168.1.102:58322
Wed Mar 25 13:43:44 2020 daemon.notice miniupnpd[23657]: shutting down MiniUPnPd
Wed Mar 25 13:43:44 2020 daemon.notice miniupnpd[343]: HTTP listening on port 5000
Wed Mar 25 13:43:44 2020 daemon.notice miniupnpd[343]: Listening for NAT-PMP/PCP traffic on port 5351
Wed Mar 25 13:43:48 2020 daemon.err uhttpd[2781]: sh: out of range
Wed Mar 25 13:43:48 2020 daemon.err uhttpd[2781]: sh: out of range

huanghuangge, 2020-03-25:
> 
> 
> ssh 里面， cp /rom/etc/config/upnpd /etc/config/upnpd， 然后重启即可

还是无效，迅雷U享版3.2.1.498、BitComet1.65都无法自动端口映射
迅雷U享版在upnp2.0.20170421-2版正常

路由器是K2P，日志如下：
Wed Mar 25 14:13:20 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Wed Mar 25 14:13:21 2020 daemon.notice procd: /etc/rc.d/S60flowoffload: DNSMASQ change
Wed Mar 25 14:13:21 2020 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Wed Mar 25 14:13:21 2020 user.notice ddns-scripts[1797]: myddns_ipv4: PID '1797' started at 2020-03-25 14:13
Wed Mar 25 14:13:22 2020 user.warn ddns-scripts[1797]: myddns_ipv4: Service section disabled! - TERMINATE
Wed Mar 25 14:13:22 2020 user.warn ddns-scripts[1797]: myddns_ipv4: PID '1797' exit WITH ERROR '1' at 2020-03-25 14:13
Wed Mar 25 14:13:22 2020 daemon.notice procd: /etc/rc.d/S94miniupnpd: sh: out of range
Wed Mar 25 14:13:22 2020 daemon.notice miniupnpd[2048]: HTTP listening on port 5000
Wed Mar 25 14:13:22 2020 daemon.notice miniupnpd[2048]: Listening for NAT-PMP/PCP traffic on port 5351
Wed Mar 25 14:13:23 2020 daemon.info procd: - init complete -
Wed Mar 25 14:14:28 2020 daemon.notice netifd: Interface 'wan_6' is now up
Wed Mar 25 14:14:28 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Wed Mar 25 14:14:29 2020 daemon.warn odhcpd[1027]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Wed Mar 25 14:14:59 2020 daemon.info pppd[1271]: System time change detected.
Wed Mar 25 14:16:39 2020 daemon.err uhttpd[1299]: sh: out of range
Wed Mar 25 14:16:40 2020 daemon.err uhttpd[1299]: sh: out of range
Wed Mar 25 14:19:57 2020 daemon.err uhttpd[1299]: sh: out of range
Wed Mar 25 14:19:57 2020 daemon.err uhttpd[1299]: sh: out of range

swxk521, 2020-03-25:
ed Mar 25 16:08:55 2020 daemon.err miniupnpd[2934]: upnp_event_recv: recv(): Connection reset by peer
Wed Mar 25 16:08:55 2020 daemon.err miniupnpd[2934]: upnpevents_processfds: 0x7f5aee626c40, remove subscriber uuid:18ccaa05-f268-435e-bc20-02482abbe21a after an ERROR cb: http://192.168.5.101:2869/upnp/eventing/orjmirwlwx
Wed Mar 25 16:08:55 2020 daemon.err miniupnpd[2934]: upnp_event_recv: recv(): Connection reset by peer
Wed Mar 25 16:08:55 2020 daemon.err miniupnpd[2934]: upnpevents_processfds: 0x7f5aee626c40, remove subscriber uuid:2d7a6d28-d3a4-4921-b3ef-9527e870daf2 after an ERROR cb: http://192.168.5.101:2869/upnp/eventing/kwfjssiquc

最新版 UPNP有问题，任何端口都不行

jwf8280031, 2020-03-25:
Wed Mar 25 16:02:40 2020 daemon.notice procd: /etc/rc.d/S94miniupnpd: sh: out of range
Wed Mar 25 16:02:40 2020 user.notice upnp daemon: external interface not found, not starting
Wed Mar 25 16:02:40 2020 daemon.warn miniupnpd[4205]: Cannot get IP address for ext interface . Network is down
Wed Mar 25 16:02:40 2020 daemon.notice miniupnpd[4205]: HTTP listening on port 5000
Wed Mar 25 16:02:40 2020 daemon.notice miniupnpd[4205]: Listening for NAT-PMP/PCP traffic on port 5351
最新UPNP还是不行

Aes64X, 2020-03-25:
实测无问题，应该说我就没见出过问题。。。
[image]

helianling, 2020-03-25:
我还是换回老版本了。
[image]

alepennyq, 2020-03-25:
> 我还是换回老版本了。
> [image]

怎么回？退回去再编译吗？

alepennyq, 2020-03-25:
> 我还是换回老版本了。
> [image]
> 这才是迅雷正常显示的样子
> [image]

我怎么看到你这个commits？奇怪

helianling, 2020-03-26:
> > 我还是换回老版本了。
> > [image]
> > 这才是迅雷正常显示的样子
> > [image]
> 
> 我怎么看到你这个commits？奇怪

https://github.com/coolsnowwolf/packages/commits/master

ricksuzade-maker, 2020-03-26:
为了专门支持迅雷而无法更新最新的UPNP版本，反而影响最新的各种其他应用。这应该算是迅雷的bug而不是op的了吧？

closed 2020-03-28 by coolsnowwolf

helianling, 2020-03-28:
> 为了专门支持迅雷而无法更新最新的UPNP版本，反而影响最新的各种其他应用。这应该算是迅雷的bug而不是op的了吧？

不只是迅雷用1.0版本的，看一下我下图的列表

[image]