coolsnowwolf/lede #4013: 最新代码编译出错

hzmax0416, 2020-03-25:
代码如下，缺少libubox20170601这个依赖
请问这个是我自己的问题还是代码的问题？
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for base-files:
 * 	libubox20170601
 * opkg_install_cmd: Cannot install package base-files.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mtd:
 * 	libubox20170601
 * opkg_install_cmd: Cannot install package mtd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for relayd:
 * 	libubox20170601
 * opkg_install_cmd: Cannot install package relayd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for urngd:
 * 	libubox20170601
 * opkg_install_cmd: Cannot install package urngd.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/hzmax/lede'
package/Makefile:108: recipe for target '/home/hzmax/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/hzmax/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/hzmax/lede'
/home/hzmax/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

yasum2006, 2020-03-25:
遇到相同问题

kevinlo0909, 2020-03-25:
遇到相同问题

coolsnowwolf, 2020-03-25:
make clean

ppdragon16, 2020-03-25:
make clean 后编译过了。👍

closed 2020-03-25 by coolsnowwolf

alkalixin, 2020-03-28:
`make clean` 以后重新编译还是报错。

```../libtool: line 1100: /home/lx/lede/staging_dir/target-mips_24kc_musl/../host/bin/grep: No such file or directory
/home/lx/lede/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/lib/gcc/mips-openwrt-linux-musl/7.5.0/../../../../mips-openwrt-linux-musl/bin/ld:.libs/libcurl.ver:2: syntax error in VERSION script
collect2: error: ld returned 1 exit status
Makefile:1227: recipe for target 'libcurl.la' failed
make[5]: *** [libcurl.la] Error 1
make[5]: Leaving directory '/home/lx/lede/build_dir/target-mips_24kc_musl/curl-7.66.0/lib'
Makefile:946: recipe for target 'install-recursive' failed
make[4]: *** [install-recursive] Error 1
make[4]: Leaving directory '/home/lx/lede/build_dir/target-mips_24kc_musl/curl-7.66.0'
Makefile:183: recipe for target '/home/lx/lede/build_dir/target-mips_24kc_musl/curl-7.66.0/.built' failed
make[3]: *** [/home/lx/lede/build_dir/target-mips_24kc_musl/curl-7.66.0/.built] Error 2
make[3]: Leaving directory '/home/lx/lede/package/network/utils/curl'
time: package/network/utils/curl/compile#28.73#4.56#32.09
package/Makefile:111: recipe for target 'package/network/utils/curl/compile' failed
make[2]: *** [package/network/utils/curl/compile] Error 2
make[2]: Leaving directory '/home/lx/lede'
package/Makefile:107: recipe for target '/home/lx/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/lx/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lx/lede'
/home/lx/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
```