coolsnowwolf/lede #4014: 移植SSR PLUS 出现问题，

king-NB, 2020-03-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：本人小白，稍微懂点linux，一点点，只需要SSRPLUS，其余不太需要，就想往官方固件上移植
移植比较顺利，只是不能使用，最后安装luci-ssr提示，
awk: /etc/config/black.list: No such file or directory
awk: /etc/config/black.list: No such file or directory
awk: /etc/config/white.list: No such file or directory
uci: Entry not found
uci: Entry not found
将black.list，white.list，复制到对应文件夹后还是不能使用
具体表现在添加节点出现问题，不能订阅，可以通过链接形式添加节点，但是节点测试有问题

期待大大们指导！
谢谢

norikohxb, 2020-03-25:
能问一下为什么不直接用lede的编译呢。自己编译看得到。为何去整一个openwrt，然后再来花费更大的力气移植ssrplus，你要知道ssrplus本身就是lean大开发的，在lede上是最无缝结合的啊

dongjuanyong, 2020-03-25:
Kernel Version	5.4.24
这是openwrt snapshot现在的内核版本，还有很多其他的东西，喜欢折腾的人就是想要去追这些

对于issue，不知道你移植的是哪个commit，新版black.list和white.list俩文件刚刚被删掉……
不能订阅的是不是依赖的问题，猜的，没log谁也搞不了

king-NB, 2020-03-25:
> 为什么要整一个openwrt，然后再来花费一点的力气移植ssplus，你要知道ssplus本身就是瘦大开发的，在lede上是最无缝结合的啊

可能是闲的蛋疼吧

king-NB, 2020-03-25:
> 内核版本5.4.24 
> 这是openwrt快照现在的内核版本，还有很多其他的东西，喜欢折腾的人就是想要去追这些
> 
> 对于问题，不知道你移植的是哪个commit，新版black.list和white.list俩文件刚刚被删除掉…… 
> 不能订阅的是不是依赖的问题，猜的，没log谁也搞不了

我前两天编译的，现在改了吗？……ssr需要的依赖，列出来的都安装了……还是太白，不适合干这活……

hcym, 2020-03-25:
现在的版本应该可以，就是麻烦，要自己排查一下，我是经常用官方加lean的ssr，以前的搞不定

king-NB, 2020-03-25:
> 现在的版本应该可以，就是麻烦，要自己排查一下，我是经常用官方加lean的ssr，以前的搞不定

按说我应该是前天git的啊……

dongjuanyong, 2020-03-25:
你看看每天多少个commit，又有多少个是ssr的……
中间有一两个版本有问题也很正常

king-NB, 2020-03-25:
> 你看看每天多少个commit，又有多少个是ssr的……
> 中间有一两个版本有问题也很正常

年龄大了，学东西也慢了，我一会重新全部弄一遍

cmheia, 2020-03-25:
`black.list`/`white.list`这两个`touch`空文件即可。
不能订阅是因为 #3965 改坏了`update.lua`，把路径还原即可。

king-NB, 2020-03-25:
> `black.list`/`white.list`这两个`touch`空文件即可。
> 不能订阅是因为 #3965 改坏了`update.lua`，把路径还原即可。
第一个没问题，第二个还原路径我得研究下……不会……谢谢

coolsnowwolf, 2020-03-25:
openwrt trunk 的 uci 和 之前版本的不同，删了 getfirst 函数，需要你们自己改写了

iyuangang, 2020-03-25:
> > `black.list`/`white.list`这两个`touch`空文件即可。
> > 不能订阅是因为 #3965 改坏了`update.lua`，把路径还原即可。
> > 第一个没问题，第二个还原路径我得研究下……不会……谢谢

可以根据这个下面的步骤来。[移植步骤](https://github.com/iyuangang/openwrt/issues/2)

king-NB, 2020-03-25:
> openwrt trunk 的 uci 和 之前版本的不同，删了 getfirst 函数，需要你们自己改写了

谢谢大大，这个需要学习一段时间了……

king-NB, 2020-03-25:
> > > `black.list`/`white.list`这两个`touch`空文件即可。
> > > 不能订阅是因为 #3965 改坏了`update.lua`，把路径还原即可。
> > > 第一个没问题，第二个还原路径我得研究下……不会……谢谢
> 
> 可以根据这个下面的步骤来。[移植步骤](https://github.com/iyuangang/openwrt/issues/2)

谢谢大大，慢慢学习下

iyuangang, 2020-03-25:
> > > > `black.list`/`white.list`这两个`touch`空文件即可。
> > > > 不能订阅是因为 #3965 改坏了`update.lua`，把路径还原即可。
> > > > 第一个没问题，第二个还原路径我得研究下……不会……谢谢
> > 
> > 
> > 可以根据这个下面的步骤来。[移植步骤](https://github.com/iyuangang/openwrt/issues/2)
> 
> 谢谢大大，慢慢学习下

跟L大比我就是个小白。L大和这个项目的贡献者才是真的大牛。

king-NB, 2020-03-25:
> > > > > `black.list`/`white.list`这两个`touch`空文件即可。
> > > > > 不能订阅是因为 #3965 改坏了`update.lua`，把路径还原即可。
> > > > > 第一个没问题，第二个还原路径我得研究下……不会……谢谢
> > > 
> > > 
> > > 可以根据这个下面的步骤来。[移植步骤](https://github.com/iyuangang/openwrt/issues/2)
> > 
> > 
> > 谢谢大大，慢慢学习下
> 
> 跟L大比我就是个小白。L大和这个项目的贡献者才是真的大牛。

我是真小白，你还有比较，😄

hcym, 2020-03-27:
昨天用主分支编译n1过了，机场订阅不能导入，v2不能用压缩，其它都不错

waterun, 2020-03-27:
> > 为什么要整一个openwrt，然后再来花费一点的力气移植ssplus，你要知道ssplus本身就是瘦大开发的，在lede上是最无缝结合的啊
> 
> 可能是闲的蛋疼吧

大雕的版本做了很多魔改神优化，我工作项目开发用的版本最初用的chunk版，性能低下，各种不爽，后来还是用了大雕的版本，真是不比不知道，一比就知道大雕版本对内核补丁的优化有多么神奇，就是同一个程序在chunk版本跑站cpu20%多，大雕版本只占5%

norikohxb, 2020-03-27:
我一入手软路由，再整完爱快和那个kooshare以后，跟着视频就直接到lede了。后面因为passwall出来开了点小差，现在又用回来了lede和ssrplus，毕竟比较一套。

closed 2020-07-04 by coolsnowwolf