coolsnowwolf/lede #4017: iOS网易云音乐解锁成功但无法播放

Kisoul, 2020-03-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题

A：iOS  的网易云音乐解锁成功，但无法播放，提示“网络不给力，无法继续播放”证书已安装。屏蔽223.252.199.10的方式也试过了。
安卓，Windows都没问题

### (2) 路由器型号和固件版本

A：25号最新编译的x86-64

### (3) 详细日志

A：日志里没有任何反应

coolsnowwolf, 2020-03-26:
信任证书

Kisoul, 2020-03-26:
> 信任证书

信任并安装了

Kisoul, 2020-03-26:
> 信任证书

谢谢，解决了，辛苦

closed 2020-03-26 by Kisoul

michealhansun, 2021-02-02:
IOS 14.3  本地wifi下添加证书信任后即可播放歌曲，公网手机流量使用小火箭设置代理好代理，查日记是解锁成功，已经获取了歌曲地址，但是感觉好像没有推送到公网手机上

[image]
[image]
[image]
[image]